betaflight/betaflight #13102: BETF_BETAFPV411 incompatible with  Serial ELRS FC v2.1

hyp0dermik-code, 2023-09-30:
### Describe the bug

Calib arm flag not cleared when reflashing from configurator - no output observed from gyro

### To Reproduce

Reflash 4.4.0+ from configurator
Load custom defaults for board (unless 4.5z)
Enable bidir
Observe no output from gyro 

Run `set dshot bitbang = on`
Observe gyro working 

### Expected behavior

Default config does not force bitbang off
https://github.com/betaflight/unified-targets/blob/db65066064791ce8c7d381666a7a9f67ce4efd1d/configs/default/BEFH-BETAFPVF411.config#L113

### Support ID

```txt
``
```

### Flight controller

BETF_BETAFPV411

### Other components

BetaFPV Serial ELRS FC v2.1  (Meteor75 Pro)

hyp0dermik-code, 2023-09-30:
Not working:
[image]
`
status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=30degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3318, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: a70fcd089938ae62e83004e3c9939c08 (4.5.0-zulu)
System Uptime: 13 seconds, Current Time: 2023-09-30T03:08:01.099+00:00
CPU:20%, cycle time: 308, GYRO rate: 3246, RX rate: 15, System rate: 9
Voltage: 75 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS ANGLE CALIB CLI MSP RPMFILTER
`
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

hyp0dermik-code, 2023-09-30:
Working:
`
dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE 
`

Noctaro, 2023-09-30:
Output from V2.0
BetaFPV Serial ELRS FC v2.0 (Meteor75 Pro)

[image]
[status_dma.txt](https://github.com/betaflight/betaflight/files/12774409/status_dma.txt)

SteveCEvans, 2023-09-30:
`timer` and `timer show` in the non-working case would be useful. Is bidirectional DSHOT on?

SteveCEvans, 2023-09-30:
It's working for me.

```
# version
# Betaflight / STM32F411 (S411) 4.5.0 Sep 30 2023 / 11:08:59 (26522a7a8) MSP API: 1.46
# config rev: 999c9ef
# board: manufacturer_id: BEFH, board_name: BETAFPVF411

# get dshot_bitbang
dshot_bitbang = OFF
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=40degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3715, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MSP (53 x 20)
BUILD KEY: 7faaf08f271b032953050181573e5481 (4.5.0-zulu)
System Uptime: 117 seconds, Current Time: 2023-09-30T17:19:39.537+00:00
CPU:39%, cycle time: 313, GYRO rate: 3194, RX rate: 15, System rate: 9
Voltage: 204 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_SDI 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_SDO 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```

[image]

Noctaro, 2023-09-30:
@SteveCEvans is this rev 2.0 of the board?

hyp0dermik-code, 2023-10-01:
> `timer` and `timer show` in the non-working case would be useful. Is bidirectional DSHOT on?
```
# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```
with dshot_bitbang = off as per default config
```
# dshot_telemetry_info
Dshot reads: 1166073
Dshot invalid pkts: 4
Dshot directionChange cycles: 316, micros: 2

Motor    Type   eRPM    RPM     Hz Invalid   TEMP    VCC   CURR  ST/EV   DBG1   DBG2   DBG3
=====  ====== ====== ====== ====== ======= ====== ====== ====== ====== ====== ====== ======
    1   R----      0      0      0   0.00%      0   0.00      0      0      0      0      0
    2   R----      0      0      0   0.00%      0   0.00      0      0      0      0      0
    3   R----      0      0      0   0.00%      0   0.00      0      0      0      0      0
    4   R----      0      0      0   0.00%      0   0.00      0      0      0      0      0

7 7 7 7 7 7 7 14 14 7 14 14 7 14 7 14 0 0 0 0 0 0 
0 0 0 0 0 0 7 0 4294967289 7 0 4294967289 7 4294967289 7 4294967282 0 0 0 0 0 
```

[image]

hyp0dermik-code, 2023-10-01:
OSD and blackbox work with working config but blackbox sometimes distorts a line so may be affected by the DMA change, but current sensor constantly reports ~12a with bitbang on or off

SteveCEvans, 2023-10-01:
Flash and OSD are both on SPI bus 2 which isn’t good for higher logging rates.

SteveCEvans, 2023-10-01:
> @SteveCEvans is this rev 2.0 of the board?
> Rev 2.0 also works for me. 2.1 seemst to have trouble.

Not sure to be honest. Mine is in an early Meteor75 HD (WS). I’ll look to see if a rev is visible on the silk-screen.

hyp0dermik-code, 2023-10-01:
> > @SteveCEvans is this rev 2.0 of the board?
> > Rev 2.0 also works for me. 2.1 seemst to have trouble.
> 
> Not sure to be honest. Mine is in an early Meteor75 HD (WS). I’ll look to see if a rev is visible on the silk-screen.

The rev on mine is in between motor ports

SteveCEvans, 2023-10-01:
Yup, found it. V2.0.

hyp0dermik-code, 2023-11-02:
Still needs further investigation, simple solution would be set bitbang on/auto in config.h, but these issues suggest that this will cause problems for other users with the older version?
https://github.com/betaflight/betaflight/issues/10819#issuecomment-1038014371
https://github.com/betaflight/betaflight/issues/12173

gizlonk, 2023-12-07:
I also have the V2.1, and my acellerometer went wild after upgrading to 4.4.3. 

Using the instructions above, I was able to get it working:

1. Flash 4.4.3
2. Enable BiDir and "save and reboot"
3. Reconnect and run "set dshot_bitbang = on" in the cli tab
4. "save" in the cli tab
5. Calibrate acellerometer
6. Configure gyro alignment (mine was 90 degrees CW)

haslinghuis, 2023-12-07:
@hyp0dermik-code should we change the default for `dshot_bitbang` for this target?

ot0tot, 2023-12-07:
Seems to be an issue with losing DMA on SPI2 (used for blackbox and OSD) when DShot bitbang is disabled. Just a poor choice of motor resource pins complicated by having multiple peripherals on the same SPI bus. 

@hyp0dermik-code If you disable blackbox, does everything else work properly with dshot_bitbang off?

The downside of defaulting to dshot_bitbang on is you lose DMA on the gyro. This doesn't seem to be a huge issue with the BMI270 gyro due to it's low sample rate, but would cause issues with FCs that use this target that have the MPU6000 gyro.

haslinghuis, 2023-12-08:
Had no issues with Meteor 85 either (ICM20689 using a development version with added blackbox from betafpv).

Prajwal-Koirala, 2023-12-26:
The issue reported in GitHub issue #13102 is related to the BETAFPV411 flight controller being incompatible with the BetaFPV Serial ELRS FC v2.1 (Meteor75 Pro). The specific problem described is that when the flight controller is reflashed from the configurator with firmware 4.4.0+ (up to 4.5z), the calibration (Calib) arm flag does not clear, and no output is observed from the gyro. This issue affects the normal functioning of the flight controller, rendering it unable to correctly process gyroscopic data【22†source】.

The flight controller in question is the BETF_BETAFPV411, and the other main component involved is the BetaFPV Serial ELRS FC v2.1. Unfortunately, there is no detailed information on how these components are wired together, which might be crucial for diagnosing and resolving the issue【23†source】.

A user facing a similar problem with the V2.1 version reported that their accelerometer behaved erratically after upgrading to 4.4.3. However, they managed to resolve the issue by following a series of steps: flashing the 4.4.3 firmware, enabling BiDir and saving/rebooting, running "set dshot_bitbang = on" in the CLI tab, saving the settings, calibrating the accelerometer, and configuring the gyro alignment (in their case, by 90 degrees CW)【24†source】.

This solution suggests that specific settings related to dshot_bitbang and gyro alignment may be crucial in resolving compatibility or functionality issues that arise after firmware updates.

alexberdnik, 2024-01-08:
Same issue for me, ELRS FC with BMI270, no ceramic antenna. dshot_bitbang is required and gyro aligment should be set to 90CW.

haslinghuis, 2024-01-18:
If using the SPI ELRS FC you would need BEFH-BETAFPVF4SX1280 instead of BEFH-BETAFPVF411.

BEFH-BETAFPVF4SX1280 does not have blackbox as BEFH-BETAFPVF411 does.

Please submit support ID here for more information. You can create one in the CLI tab.

demvlad, 2024-03-29:
Hello!
I have this issue on BetaFPV F4 5A AIO intergrated serial ELRS.
No arm with CALBRATING reasone. There are no Gyro output on sensor tab.
Betaflight 4.5RC3.
I've checked, the BF 4.4 does not work too.
Is exists any way to do this FC work?

demvlad, 2024-03-29:
> Hello! I have this issue on BetaFPV F4 5A AIO intergrated serial ELRS. No arm with CALBRATING reasone. There are no Gyro output on sensor tab. Betaflight 4.5RC3. I've checked, the BF 4.4 does not work too. Is exists any way to do this FC work?

It seems, the dshot_bitbang=on is realy helpfull!

MWTaken, 2024-03-29:
Led outputs won’t work with bitbang on, don’t bother trying to solder on addressable leds.

haslinghuis, 2024-03-29:
Try disabling blackbox for led outputs to work.

demvlad, 2024-03-29:
Ok. Thank you.
I've disabled LED when flash BF firmware.
But i need blackbox very much:).
Will the blackbox work without LED?

demvlad, 2024-03-29:
There are no other limits by using bit_bang=On ?

demvlad, 2024-04-02:
I've got an answer from BetaFPV support. They written that this is some firmware issue.

Justicefpv, 2024-05-01:
Below is from my Discord support forum post. I've run into this issue recently, as it seems many others have.

> I recently replaced two BETAFPVF411 1s AIO’s both are of the same 2.1 revision. However, I have encountered an issue with the gyro calibration/initialization process.

Upon setting up and testing the flight controllers, I found that the FCs fail to calibrate properly during the initialization process after plugging in the USB cable. (Setting the gyro_calib_noise_limit has no effect) The arming flag within the configurator is stuck at 'calibrating', the model does not move on the setup tab and the sensors tab also shows no movement.

I have attempted various troubleshooting steps and the only way in which I can get both FCs to work correctly is to set both dshot_bitbang to ON and acc_calibration to a value I found in the dump file on the betafpv website within the CLI. By doing this the configurator will then report the gyro correctly.

I found that the two FCs also behave differently when it comes to the dshot_bitbang setting. One unit works correctly with this setting turned OFF, while the other unit only functions properly when dshot_bitbang is set to ON. This behaviour is surprising since both units are supposed to be identical (2.1 revision).

>

krisdover, 2024-05-10:
> Same issue for me, ELRS FC with BMI270, no ceramic antenna. dshot_bitbang is required and gyro aligment should be set to 90CW.

Same here running BF v4.4.3. The FC is a v2.1 betafpv411 AIO board with the serial ELRS receiver. For some strange reason the Gyro update frequency is also stuck on 3.20 KHz. Any ideas as to how to get 8 KHz?

closed 2026-09-26 by blckmn