betaflight/betaflight #13106: GEP RC GEP-F722-35A AIO lost Gyro and possibly other ports after flashing BF fireware 4.40 and 4.4.2

upyopyo, 2023-10-06:
### Describe the bug

Hi, I tried to flash BF fireware from the original version to 4.4.0 and 4.4.2  in order to use newer generation's GPS. And after flashing with sucessful update, Gyro is not detected anymore. (Sorry, I forgot the original BF version, possibly 4.2.x). Before flashing fireware, my drone is working fine.

At the beginning, the firmeware uploading was frozen at initializing step. Then I ran  ImpulseRC_Driver_Fixer.exe. Althoguh it finished with error message, now I can upload the BF firewares. Then I tried to flash with 4.4.0 and 4.4.2 and the both don't  detect Gyro. 

Next I found this link and a magic line for custom defines:
https://github.com/betaflight/betaflight/issues/12138
GYRO_SPI_MPU6000 ACC_SPI_MPU6000 

However, it still did not fix my Gyro detection, and now I have no idea how to fix this issue. 
Actually I found other issues. Airunit does not output video, and receiver does not receive signal although UARTs set properly as same as before flashing. Thus, I am thinking the problem is not only Gyro, but also other components, too. 

Maybe, I am missing very critical process. Please let me know how to fix all, thanks. 
These attached screenshots are after flashing frimewares. 

[image]
[image]
[image]
[image]

### To Reproduce

rerun to flash fireware updatings.

### Expected behavior

I am expecting to see detect Gyro and 

### Support ID

```txt
*** Support data submitted *** 
Id: "2bf4e59c-4bdd-44d7-b9cc-cd98510d3235"
```

### Flight controller

GEP-F722-35A AIO

### Other components

receiver Futaba
VTX DJI airunit
GPS HGLRC M100_MINI GPS
buzzer gepRC superBuzzer

haslinghuis, 2023-10-06:
Please flash core only and submit support ID so we can add hardware defines to configuration.

upyopyo, 2023-10-06:
Thanks for your reply, and I see what you mean!

*** Support data submitted *** 
Id: "176ae1e2-0205-4c22-b670-2a706d2d6dbf"

haslinghuis, 2023-10-06:
There are I2C errors in status, but still no gyro or baro. For gyro all drivers are included in CORE build. Are you sure you have the correct target (there is also a HD version)? Auto-detect only reads what firmware has been flashed previously.

According to https://geprc.com/wp-content/uploads/2022/05/F722-35A-AIO-USER-MANUAL2.pdf you will need firmware `GEPRC_F722_AIO`