betaflight/betaflight #13110: NeutronRC F435SE causes subtle twitches (4.5 nightly)

LordGG, 2023-10-08:
### Describe the bug

I get subtle twitches with NeutronRC AT32F435 FC, using NEUTRONRCF435SE target.

### To Reproduce

Just flying with that FC, with NEUTRONRCF435SE target and 4.5 firmware, it occures randomly. It is very hard to notice on video but much more obvious in blackbox or in the goggles. It is a bit more noticeable at constant mid throttle speed.

Bidir Dshot enabled or not makes no difference.

### Expected behavior

No twitches, smooth flying.

### Support ID

```txt
*** Support data submitted *** 
Id: "038cf688-b4cf-4642-9aaa-4c2e536cdc4a"
```

### Flight controller

NeutronRC AT32F435 (20x20 / 30x30 board)

### Other components

ESCs T-Motor F55Pro II F3 55A (32.9), Velox V3 2306 1950KV, Vista, ELRS 3.2

### How are the different components wired up (including port information)

Vista wires are soldered (powered from lipo, not BEC)
FC/ESCs with wiring harness
RX soldered

### Add any other context about the problem that you think might be relevant here

At the begining, I used Skystars AM60 ESCs (AM32) and I thought they were the culprit. 
Then I switched the AM32 ESCs with T-Motor BLHeli32... and the twitches were still there.
So I switched the AT32 FC with an STM32F405 FC (same gyro) and... twitches are gone !

STM32 FC was flashed with 4.5 nighlty for this test, to be as close as possible of the AT32 FC.

I have uploaded a blackbox and the video here : 
https://filetransfer.io/data-package/ZJRMViMB#link

Video is also available on YT (with index of the twitches) : 
https://www.youtube.com/watch?v=M5lrgP2i49s

Another video  with twitches (when using AM32 ESCs) (twitches also indexed) : 
https://www.youtube.com/watch?v=oEFw5AC_0XA

I know those twitches are not easy to see on the video, listening to the motors help to spot them. And they are easy to see in the blackbox.

Alka (AM32 dev) told me that an other guy also add the exact same issue with the same FC.

Except for this problem, Blackbox is fine, gyro signal is clean.

Thanks for your help :)

LordGG, 2023-10-12:
I first noticed this issue with the NeutronRC F435SE.

I have built a 3" with the NeutronRC AiO AT32/AM32 45A. I couldn't see the random twiches on the recordings,  but they are noticeable in the Blackbox : 
https://filetransfer.io/data-package/1wIqxPFS#link
(12s, 55s, 57s, 1min08 etc.).

This is the second AT32 FC that gives me these twitches. On the 5", switching the FC with a STM32 F405 base FC made the twitches go away.

LordGG, 2023-10-19:
Hi,

I did more tests.

**3" (NeutronRC AiO AT32/AM32) :** 
I got rid of the twitches by changing only one thing : setting PWM Frequency to fixed 48khz instead of variable (24khz-byRPM).
It seems, the ESCs were the culprit, not the FC or the AT32. So let's forget it, I will rename the issue like it was before.

**5" (NeutronRC AT32F435 + ESCs T-Motor F55Pro II F3 55A) :**
Here are the tests I did and the results : 

- Setting PWM Frequency to fixed 48Khz : ❌ KO, twitches still there
- Rolling back ESCs to BLHeli32 32.7 : ❌ KO, twitches still there
- Disabling Dynamic Idle : ❌ KO, twitches still there
- Switching ESCs to Skystars AM32 (AM60) : ❌ KO, twitches still there
- Switching FC to Skystars F405HD2 (T-Motor ESCs) : ✅ OK, no more twitches

All tests done with Betaflight Nightly 4.5. So it seems like the issue is related to the FC or its target.

There are 2 potential targets for the [NeutronRC AT32F435](https://fr.aliexpress.com/item/1005005612227664.html) : 

- NEUTRONRCF435 : initial one
- NEUTRONRCF435**SE** : added later, with more defines, so this is the one I use.

But when comparing the 2 configurations, there is a difference concerning the gyro : 
[image]

I am reaching my limits here : do you think, the differences concerning the 2 last lines could be a potential explanation ? Is it worth giving a try to NEUTRONRCF435 target, dispite the missing defines ?

SupaflyFPV, 2023-12-05:
> Hi,
> 
> I did more tests.
> 
> **3" (NeutronRC AiO AT32/AM32) :** I got rid of the twitches by changing only one thing : setting PWM Frequency to fixed 48khz instead of variable (24khz-byRPM). It seems, the ESCs were the culprit, not the FC or the AT32. So let's forget it, I will rename the issue like it was before.
> 
> **5" (NeutronRC AT32F435 + ESCs T-Motor F55Pro II F3 55A) :** Here are the tests I did and the results :
> 
> * Setting PWM Frequency to fixed 48Khz : ❌ KO, twitches still there
> * Rolling back ESCs to BLHeli32 32.7 : ❌ KO, twitches still there
> * Disabling Dynamic Idle : ❌ KO, twitches still there
> * Switching ESCs to Skystars AM32 (AM60) : ❌ KO, twitches still there
> * Switching FC to Skystars F405HD2 (T-Motor ESCs) : ✅ OK, no more twitches
> 
> All tests done with Betaflight Nightly 4.5. So it seems like the issue is related to the FC or its target.
> 
> There are 2 potential targets for the [NeutronRC AT32F435](https://fr.aliexpress.com/item/1005005612227664.html) :
> 
> * NEUTRONRCF435 : initial one
> * NEUTRONRCF435**SE** : added later, with more defines, so this is the one I use.
> 
> But when comparing the 2 configurations, there is a difference concerning the gyro : ![diff_targetsF435](https://private-user-images.githubusercontent.com/6628377/276778308-8bdebffe-8099-4bc1-9ab7-23817458ccdc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3Nzc5NDYsIm5iZiI6MTcwMTc3NzY0NiwicGF0aCI6Ii82NjI4Mzc3LzI3Njc3ODMwOC04YmRlYmZmZS04MDk5LTRiYzEtOWFiNy0yMzgxNzQ1OGNjZGMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDVUMTIwMDQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzM1YTVjOTljMjE5Y2I2ODVhZWU4N2NiYzhlOWY4ZDQzNTM2ZGY1YjZmYmUwNWRjYzMxZjk5ZTg2MzgxM2MyYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.QknPtmCrr4LytNFHH2OCGbCtcKX40ffJipxpNhiRWwI)
> 
> I am reaching my limits here : do you think, the differences concerning the 2 last lines could be a potential explanation ? Is it worth giving a try to NEUTRONRCF435 target, dispite the missing defines ?

thanks for making the effort to do a comprehensive test.

The issue you have is usually a esc related one, although it can be the sign of a faulty gyro as well. The 'twitches' you hear are either minor commutation errors (due to minor esc setup or some esc hardware failure) or occasional bad data outputs from the gyro.

The flight controller firmware is otherwise running as it should, and has to try to combat the hardware errors as they arrive on a random basis.

I would not download the log unfortunately because it was required that I make an account. Please just zip them next time and upload directly. You can add files at the bottom of the text window.

You have identified the issue to esc setting, and the FC. It will not be related to the target-I have seen a lot of these issues over the last couple of years. Gyro quality control, and gyro positioning and power supply is tricky with FC manufacturing and small problems cause these types of error. I recommend to stick to brands on our 'Approved Partner' list - Skystars and T motor are on there. 

https://betaflight.com/partner

SupaflyFPV, 2023-12-05:
@haslinghuis  I think you can close this

closed 2023-12-05 by haslinghuis