betaflight/betaflight #13116: PLEASE HELP! Motor values are increasing by themselves in Betaflight!

101000june, 2023-10-12:
### Describe the bug

I have FC: "JHEMCU GHF13AIO", Betaflight Configurator does detect it: "COM3 - Betaflight STM32F411". 
When I arm it in Motors tab, it arms corectly, all values raise from 1000 to 1051-1052 **and the value remains unchanged**. 
**_But_** when I raise Throttle stick to ~20% (**_and leave it there at 20%_**), value increases to ~1200, and one or two motor values **continue increasing all the time**, reaching 1500 and more during 60 seconds. This works without battery, and with battery, making the drone fly crazy. In Receiver tab it remains correctly: value is always 1200! 

**_Please help to fix and provide any helpful thoughts on this issue! Thank you in advance!_**
Please see my attached video that demonstrates the problem:
https://github.com/betaflight/betaflight/assets/39669835/75b7bff1-1fcd-4647-954b-67e5b6b52aa1

### To Reproduce

Here is what I've done:
1) Use "JHEMCU GHF13AIO", seen in Configurator as "STM32F411"
2) Setup TBS Crossfire Nano, calibrate accelerometer, calibrate ESC, ensure motors spin with correct direction and power
3) In Motors tab, allow to spin motors, I caught this issue both with **_and without battery, you can unplug it_**
3) On bound Remote Control, Arm and raise Throttle stick near 20%
4) Go to Receivers tab, see Throttle is near **_1200 and remains unchanged! It does NOT raise to 1500 etc_**
5) Return to Motors tab, see that **_one of motors value is increasing all the time from 1200 upwards! 1300, 1500, 1700 etc_**
6) Move Throttle to 0%, values are correct now again

### Expected behavior

Expected: In Motors tab, when I raise Throttle near 20%, motors value increase to near 1200 **_and remain!_**
Actual: In Motors tab, when I raise Throttle near 20%, motors value increase to near 1200 **_and is raising slowly, to 1500 during 60 seconds!_**

### Support ID

```txt
df982fd1-ba35-4bb8-bd08-f066f950a821
```

### Flight controller

JHEMCU GHF13AIO

### Other components

RX: TBS Crossfire Nano
Motors: HGLRC FD1106-2400kv
All motors spin correctly when tested separately in Betaflight

101000june, 2023-10-12:
**UPD:** I caught the same behavior with another FC "Diatone Mamba MK4 F740 AIO". And without physically connected motors the same too. Two motor values increase to 1800 (and one is reducing to 1050) continuously

MJ666, 2023-10-14:
This is completely normal behavior on the bench. The FC is trying to control the position. Since no motors are in your case there is no reaction to the command and the FC is trying harder to fix the position.

closed 2023-10-14 by 101000june