betaflight/betaflight #13128: Add a password to being able to see and/or change the quad settings

AlexandreBonneau, 2023-10-19:
### Is your feature request related to a problem? Please describe

In order to prevent unwanted changes to an existing quad by a third party, for instance if you do not want somebody to change your settings without you knowing during a competition, it would be great to be able to 'protect' the quad using a simple password.

### Describe the solution you'd like

You could imagine, when a password field is setup beforehand, that everytime you plug your quad, betaflight-configurator will ask for such password.
If you fail to enter the correct one, then you keep having betaflight-configurator asks for a correct password.
If you enter the correct password, then you can access betaflight-configurator as before.

Since there is no need for very high security, you could retrieve the password in plain text used using `diff all` and `dump`, facilitating copying your configuration between your quads, or just setting up a personal preset.

### Describe alternatives you've considered

No other alternative is possible to secure a BF quad like this.

ElectricRCAircraftGuy, 2023-10-25:
Security should be considered. Perhaps store a SHA256sum of the password only, not the password itself. Then when the password is entered, sha256sum it, and compare the two hashes. Don't store the password.

haslinghuis, 2023-10-26:
Password won't help here. Physical access enables tampering.

Some STM32 MCU's allows for secure boot: https://www.st.com/resource/en/product_presentation/x-cube-sbsfu_marketing-pres.pdf

See also https://blog.st.com/stm32trust/

AlexandreBonneau, 2023-10-26:
The point of this feature request is not to have a super security, this is why I said that keeping the password in plain text was ok. Think 'bind phrase' like with ELRS.
The point is to have a quick-and-dirty _simple_ security measure so that somebody that could handle the quad quickly couldn't just plug it to its phone, and change its settings easily.

The use case here is the following:
- You are at a race event, and need to wait for your next race
- You most likely won't keep your quad in your hands while waiting
- A bad actor could theoretically get your quad, plug it in the speedybee app, then quickly change a few settings like rates, PIDs, aux, modes or anything else really. Those changes could not be immediately apparent when arming your quad, but could somehow still makes you lose the race once in the air.

So in this particular case, having a super high security password option is moot, but even a simple "Enter a 4 digit"-type of password would thwart that kind of attack.

Now, being able to _completely_ secure your quad as @haslinghuis suggested could indeed be useful is some other cases, but I think that would require opening another feature requests.

haslinghuis, 2023-10-26:
Spoofing, jamming, hijacking, In-flight over the air reconfiguration - lots of ways to attack a drone.

ElectricRCAircraftGuy, 2023-10-26:
> Spoofing, jamming, hijacking, In-flight over the air reconfiguration - lots of ways to attack a drone.
> 
> https://www.youtube.com/watch?v=D3Z8_bvrKbg

I've never even used BetaFlight (I'm about to), so I don't have user experience with this particular thing yet, but I'd like to point out that this clearly wouldn't be about *stopping* tampering, it would be about *deterring* it. Why else would we have glass windows that lock, screen doors that lock? 

I have a tiny little lock on my saddle bag on my bicycle. If someone wants to get through it?--they just have to *bend open* the zipper a little and the pouch comes open. But, from 10 feet away if two bikes are visible, they see a lock on mine and not on someone else's. It's a deterrent. No lock can stop a determined actor. 

So, the fact that *other* ways exist to cause problems doesn't mean we should ignore *this* way. 

Now I'm going to go find the tinniest Flight Controller possible that can run both BetaFlight and Ardupilot for my 550mm flying wing.

SteveCEvans, 2023-10-26:
This would break existing use of MSP and would not be backwards compatible. Not viable.

AlexandreBonneau, 2023-10-26:
So, let's say this feature is introduced in 4.5.
Now, if you plug your quad and check it with Betaflight configurator, can't you detect the BF version used on the quad, and then check if the `connection_pin` value is set. If yes, ask for a pin number?
Better yet, do not check the BF version, and just check if the `connection_pin` variable is set.
Perhaps also detect if a usb cable has been plugged, so that MSP isn't affected when not using a usb cable?
Maybe this can only affect Betaflight-configurator and not BF?

I'm not sure to follow how this would affect MSP, or need to be backwards compatible.

As @ElectricRCAircraftGuy mentioned, this feature is a simple one to deter basic (and quick) attempts to tamper with a quad.

haslinghuis, 2023-10-28:
Simple flashing board (backup - restore) and tampering with configuration anyway will make this feature useless - with physical access security barriers will always be overcome unless we use specific hardware like TPM

closed 2023-10-28 by haslinghuis

AlexandreBonneau, 2023-11-07:
I fail to understand why this is closed, since it's still a valid use case.
I also fail to see how TPM or physical tampering is relevant in the use case described in the issue, i.e. having a very simple way to prevent a quick non-technical access to BF settings.

Since other have expressed interest in such feature, does it need to be closed even if not planned yet?

ledvinap, 2023-11-07:
@AlexandreBonneau : Just put 'Do Not Reconfigure Without Authorization' sticker on your FC. It will provide about the same level of security and no hard to maintain code is necessary.