betaflight/betaflight #13129: Multiple FCs FREEZE midflight and drop

michael-blue2, 2023-10-20:
### Describe the bug

So the fc freezes in the air and just disarms/drops. All my osd elements getting removed from the goggles 2. (running vista) and I only have voltage which gets it from its own power and the link bars of the video.

The fc leds are all light up and you cannot do anything to it. I do still have connection with my tango 2 (green light) and signal bar on the controller. 

When plugin the fc into the computer its just dead. restarting the tango doesnt change anything. I must remove the battery from the fc and then everything works back to normal

3 Different builds. 
2x 5" and one 2.5" 

I had this several times in the past months. First on a Kakute f7 HDV. and replaced it with a matek f405. no probelm so far..
Second quad with a Matek f722HD

And third quad which is a fresh build 2.5" betafpv aio 35 f722 and while tuning it, was almost default. it happened again... first battery 
I have no idea what causes this

On the video I was going for disarm to get some satellites and it just dropped a few moments before I hit the disarm button.  And you can see that i am loosing the osd elements. And after you can see when I connect it to the computer on what it does. 

Its vary rare to happen but it still does happen. 

https://youtu.be/2V1hGZwrdrg

### To Reproduce

Just fly and the fc freezes 

### Expected behavior

To not shut down...

### Support ID

```txt
1523d9d7-a60c-45f0-a1b6-05914db1b1ec
```

### Flight controller

Kakute f7 HDV, betafpv aio 35 f722,  Matek f722HD

### Other components

TBS TANGO 2
tbs nano rx
dji vista 
goggles 2

michael-blue2, 2023-10-20:
These are the boards with firmware that got me the issue. 
5" 4.2.9 Kakute F7 HDV
5" 4.4.2 Matek F722 HD
2.5" 4.4.2 BetaFPV aio 35a f722

klutvott123, 2023-10-21:
which crsf firmware version are you using?

michael-blue2, 2023-10-21:
> which crsf firmware version are you using?

Latest that is available on tbs agent web. 
Which is 6.19.
I was thinking at the beginning it might be a failsafe. But restarting the tango 2 changes nothing. And if you check the video i uploaded , the fc its frozen and osd elements go away and connecting to betaflight doesnt do anything.

SteveCEvans, 2023-10-21:
Have you checked off this still happens with the latest dev build? I fly F7s and have had no such issues.

michael-blue2, 2023-10-22:
> Have you checked off this still happens with the latest dev build? I fly F7s and have had no such issues.

it seems to be exactly the same problem with this #13130 tha you just merged. 

So you just merged it into the latest dev master correct. On 4.3 zulu or 4.5?

fabi9163, 2023-10-22:
Hi together,
I had the same issue. Maybe it's related to #12924. But I had no JETIEXBUS-Protocol selected

SteveCEvans, 2023-10-22:
That was very specific to JETIEXBUS.

haslinghuis, 2023-10-23:
@michael-blue2 please check again with 4.4.3-zulu build

haslinghuis, 2023-10-24:
Closing as duplicate of #12924 to consolidate issues

closed 2023-10-24 by haslinghuis