betaflight/betaflight #13141: 3d mode disables return to home

chimera15, 2023-10-25:
### Describe the bug

I just built a 3d mode, and ticked the 3d mode and noticed that it disabled my return to home....  I have 3d mode set on a switch and only will activate it occasionally, for like surfing mountains....  Will return to home not work with 3d, it can't figure out the throttle or something like that I guess?  But if I have it set to a switch and the default is disabled and may not even fly any given flight in 3d mode then it should be accessible in that case? 

### To Reproduce

tick the 3d mode in the motor tab, rth unable to be chosen in the modes tab or the failsafe tab.

### Expected behavior

rth should be able to be chosen if 3d mode is an option or not, especially if 3d mode is set to be disabled by default and on a switch. 

### Support ID

```txt
no not missing
```

### Flight controller

any

atomgomba, 2023-10-25:
Duplicate of #9233 and #12739

chimera15, 2023-10-25:
Also this wasn't explicit when it occurred it's something I had to track down and figure out over 2 days to realize this was what was disabling it. I rebuilt my quad multiple times cause I thought I had something wrong.   There should be a warning that pops up at least...that said we will disable your rth or something if you won't fix it so they can be configured at the same time.

haslinghuis, 2023-10-25:
Duplicate of https://github.com/betaflight/betaflight/issues/9233 and https://github.com/betaflight/betaflight/issues/12739

closed 2023-10-25 by haslinghuis