betaflight/betaflight #13142: Single dynamic notch with 500 Q factor (default) increasing mechanical vibrations

yaanfpv, 2023-10-26:
### Describe the bug

This issue arose from I intermittently seeing heavy mechanical vibrations (gyro_scaled roll) that would randomly disappear when PID and filter tuning. I was finally able to narrow it down to the dynamic notch. 
Running RPM filtering with 3 harmonics and minHz of 100 (default)

**Bug:**
The default value of :
1 Notch Count
500 Q factor
100 minHz
600 maxHz
causes immense vibrations in the frame. Bench testing without props confirms the **motors shaking violently**.
This effect is **only seen if armed via radio (PID loop active)** and not when the motors are spun from the motors tab.
Attached below are the log and screenshots:
[1Notch_Log.zip](https://github.com/betaflight/betaflight/files/13182290/1Notch_Log.zip)
[image]
[image]

When the dynamic notch values are changed to :
2 Notch Count
400 Q factor
100 minHz
600 maxHz
, the vibrations all but disappear. The motors run cool and clean when armed and there are no ill effects.
Attached below are the log and screenshots:
[2Notch_Log.zip](https://github.com/betaflight/betaflight/files/13182328/2Notch_Log.zip)
[image]
[image]

**Idea**

I have a sense that a single notch and its correction is probably getting applied out of phase (due to delay?) and that is causing the vibrations to be amplified.

### To Reproduce

Set default Dynamic notch values on a custom frame
1 Notch Count
500 Q factor
100 minHz
600 maxHz

### Expected behavior

The default value should not amplify the vibrations (mechanically as seen in gyro_scaled)

### Support ID

```txt
d0c3683d-57c7-4bf6-a677-42edc6683d45
```

### Flight controller

Foxeer F722 V3 (BMI270 gyro)

### Other components

ESC: Foxeer Reaper F4 128K 65A BL32
Motors: iFlight Xing 2208 1800kV
RX: Happymodel ES900RX
VTX: DJI Air Unit with Micro camera
Battery: GNB 1550mAH 6S
Buzzer: Vifly Finder 2

### How are the different components wired up (including port information)

RX on UART 3
Air Unit on UART6

### Add any other context about the problem that you think might be relevant here

Running BF 4.4.2

haslinghuis, 2023-10-28:
@KarateBrot please check

yaanfpv, 2023-11-03:
@KarateBrot do you need any more specific info or further logs? Let me know, so I can capture the same.

mituritsyn, 2023-11-24:
I think the problem is that one dynamic notch is not enough to filter out this noise

yaanfpv, 2023-11-24:
@mituritsyn but that should show up in gyro logs (post filtering) if its not filtering enough right? 
It shouldnt change vibrations in gyro_scaled (pre filtering) which is true vibrations correct?

yaanfpv, 2023-11-24:
Also i tested the same frame with an AT32 FC with BMI gyro (BF 4.3). The problem does not crop up. 

I have tested now on 3 F7 FCs with BMI gyros and seeing the same problem. 

Testing with a F4 FC over the weekend. Will also do some tests with other gyros that ill pull from old builds.

mituritsyn, 2023-11-25:
> Also i tested the same frame with an AT32 FC with BMI gyro (BF 4.3). The problem does not crop up.
> 
> I have tested now on 3 F7 FCs with BMI gyros and seeing the same problem.
> 
> Testing with a F4 FC over the weekend. Will also do some tests with other gyros that ill pull from old builds.

could you please provide gyro_scaled log from another FCs?

mituritsyn, 2023-11-25:
> @mituritsyn but that should show up in gyro logs (post filtering) if its not filtering enough right? It shouldnt change vibrations in gyro_scaled (pre filtering) which is true vibrations correct?

Not exactly, unfiltered noise would be amplified by the PIDs.

KarateBrot, 2024-01-02:
Sorry, this went over my head somehow.

@yaanfpv Can you try this again using the latest BF 4.5 (zulu), using debug mode `FFT_FREQ` and then send me the log? I can see if the dynamic notch is not working correctly with a special script.
[image]

closed 2024-02-10 by github-actions[bot]