betaflight/betaflight #13143: Incorrect variometer readings GEP M10-DQ

dereksynkopa, 2023-10-27:
### Describe the bug

I have a Geprc Mark5 drone and I am trying unsuccessfully to get the correct variometer readings from the DPS310 barometer. Unfortunately, even in this version of betaflight 4.5.0 Zulu, the problem still remains unresolved. The indications are completely irrational. Values reach up to 60-70 m/s during a sudden upward flight. I noticed this problem already in BF 4.4.2 version. In the earlier version BF4.3.2, the readings were also incorrect and ended around 15 m/s during ascent and descent. Previously, I had a Nazgul Evoque drone also on FC F722 and there were no such problems there. Where is the solution to this situation?

### To Reproduce

Accurate parameterization of the variometer readings

### Expected behavior

Correct reading of the drone's descent and ascent in m/s (GEP M10-DQ)

### Configurator version

BF10.10.0-RC1

### Flight controller configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Oct  8 2023 / 18:49:37 (63f506530) MSP API: 1.46
# config rev: c04e477

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name GEPRCF722_BT_HD
manufacturer_id GEPR
mcu_id 003c002a3532510639323431
signature 

# name: MARK5

# resources
resource SERIAL_TX 6 NONE
resource SERIAL_RX 6 NONE

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 4 2 115200 115200 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1300 2100 1 0
aux 1 0 3 1700 2100 1 0
aux 2 1 0 1300 1700 1 0
aux 3 1 3 1700 2100 1 0
aux 4 27 1 1700 2100 0 0
aux 5 46 2 1700 2100 0 0
aux 6 13 4 1700 2100 0 0
aux 7 28 0 1300 2100 1 0
aux 8 28 1 1300 2100 1 0
aux 9 52 1 1700 2100 0 0

# master
set gyro_to_use = FIRST
set acc_calibration = 0,1,25,1
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_calibration = 640,-185,-164
set fpv_mix_degrees = 25
set blackbox_sample_rate = 1/16
set dshot_bidir = ON
set motor_output_reordering = 2,3,0,1,4,5,6,7
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set failsafe_recovery_delay = 10
set bat_capacity = 1550
set current_meter = ADC
set vbat_scale = 111
set ibata_scale = 96
set small_angle = 180
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 30
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_ground_speed = 1500
set gps_rescue_descent_dist = 75
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set osd_cap_alarm = 1500
set osd_alt_alarm = 2000
set osd_tim1 = 2561
set osd_tim_1_pos = 2567
set osd_remaining_time_estimate_pos = 2599
set osd_flymode_pos = 2617
set osd_current_pos = 3592
set osd_mah_drawn_pos = 2472
set osd_craft_name_pos = 386
set osd_gps_speed_pos = 3625
set osd_gps_lon_pos = 2061
set osd_gps_lat_pos = 2076
set osd_gps_sats_pos = 2646
set osd_home_dir_pos = 2650
set osd_home_dist_pos = 2651
set osd_flight_dist_pos = 3561
set osd_altitude_pos = 19881
set osd_warnings_pos = 2582
set osd_avg_cell_voltage_pos = 2503
set osd_battery_usage_pos = 35304
set osd_disarmed_pos = 2391
set osd_nvario_pos = 3529
set osd_esc_rpm_pos = 32
set osd_stat_bitmask = 4542106
set osd_displayport_device = MSP
set debug_mode = GYRO_SAMPLE
set vcd_video_system = HD
set displayport_msp_fonts = 0,0,0,0
set gyro_1_sensor_align = DEFAULT
set craft_name = MARK5

profile 0

# profile 0
set i_pitch = 96
set d_pitch = 57
set f_pitch = 99
set i_roll = 92
set d_roll = 50
set f_roll = 95
set i_yaw = 92
set f_yaw = 95
set horizon_level_strength = 50
set d_min_roll = 37
set d_min_pitch = 42
set motor_output_limit = 90
set feedforward_smooth_factor = 60
set simplified_i_gain = 115
set simplified_d_gain = 125
set simplified_feedforward_gain = 80

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0
```

YarosMallorca, 2023-10-28:
Try to change the altimeter source to baro only from CLI

```set position_alt_source = BARO_ONLY```

dereksynkopa, 2023-10-28:
In my FC there is a command: set altitude source, but it is the same. I have it set to default. So I will set: only baro and check. Thanks.

dereksynkopa, 2023-11-19:
After many tests, I can say that this did not provide the correct solution. The values are still irrational, at the level of 40-50 m/s when flying up and down.

haslinghuis, 2023-11-21:
> Previously, I had a Nazgul Evoque drone also on FC F722 and there were no such problems there.

Did it have another BARO (I2C or SPI)?

dereksynkopa, 2023-11-21:
Nazgul had the same BARO DPS310 (I2C), but it was located on the PCB FC. In the case of Marc5 BARO DPS310 (I2C) is located on PCP from GPS. I replaced the GPS with GEP-M8Q, but the problem is the same even though there is BARO MS5611. I wrote to GEPRC support about this and received a reply that there is an error in the fc firmware that needs to be corrected. They are supposed to fix it, but they haven't given a date.

egnerfl, 2023-11-29:
Hi there, I face the exact same issue (as soon as the GPS is plugged in via I2C no baro is detected) and tried everything mentioned which includes

- Adding the `BARO` build flags to the cloud build
- Making sure the I2C device is the same as the mag (which works just fine)
- Setting the baro type manually (to DPS310/MS5611 and the internal one BMP280)

Unfortunately none of this helped.

BTW I found these hints in this document by geprc and this GH issue #12993
https://geprc.com/wp-content/uploads/2023/08/The-solution-for-GEP-M8Q-M10-DQ-and-other-GPS-module-magnetometers-and-barometers-not-working.pdf

Please let me know if I can provide any logs / information to resolve this issue. I'm new to betaflight so in case I should provide anything please tell me how to gather the information. ✌️

Prajwal-Koirala, 2023-12-26:
Issue #13143 on the Betaflight GitHub is related to incorrect variometer readings on a Geprc Mark5 drone with a DPS310 barometer. The user reports experiencing irrational variometer readings, reaching 60-70 m/s during sudden upward flights, on Betaflight versions 4.5.0 Zulu and 4.4.2. They observed similar issues with lower readings (around 15 m/s) in version 4.3.2. Previously, they used a Nazgul Evoque drone with the same DPS310 barometer (I2C) but did not encounter such problems【44†source】.

The issue persists even after replacing the GPS unit with a GEP-M8Q, which also has a BARO MS5611. The user contacted GEPRC support, who acknowledged an error in the firmware that needs correction, but no fix date was provided【45†source】.

Another observation suggests that the internal DPS310 works correctly, but similar issues are observed with the GEP-M10-DQ and GEP-M8Q, which share the same I2C interface with the magnetometer. This indicates that the problem might not be with the individual components but with how they interact on the I2C line【46†source】.

closed 2024-02-08 by github-actions[bot]

haslinghuis, 2024-02-10:
Tested GEPRC-M1025 with MS5611 baro and IST8310 mag (needs `set mag_i2c_address = 14`) on SPEEDYBEEF7V3.

[image]