betaflight/betaflight #13147: OSD for Battery Voltage and Warning Disappears when Battery Critical - 4.5 Zulu

MacDaddyFPV, 2023-10-29:
### Describe the bug

On Zulu 4.5 when the battery level goes into the battery critical range the warning and voltage disappears from osd. When the level returns to warning or normal levels the osd warning and battery level returns.

### To Reproduce

On a new flash of Zulu 4.5, apply hdzero preset to enable msp osd then add average cell voltage to osd.
Plug into power supply and reduce voltage until it gets into battery critical stage. Increase the voltage back up to see the warning and battery voltage return to osd.

### Expected behavior

Expect to see the battery voltage and the "Land Now" warning in the OSD.

### Support ID

```txt
4395eaa2-81c7-42c4-bb95-2f4bf4d96b83
```

### Flight controller

Foxeer Mini F722 V3 Pro

### Other components

HDZERO RaceV3 and Race V2 both tested. HDZERO Goggles and Nano90 Camera.

### How are the different components wired up (including port information)

VTX is wired to UART 5

### Add any other context about the problem that you think might be relevant here

Using the same method of reproduction, the issue does not exist in Betaflight 4.4.2

SteveCEvans, 2023-10-30:
If using a video system which doesn’t support colour fonts simply `set displayport_msp_fonts = 0,0,0,0` and you’ll get just the monochrome warnings etc.

i suggest you check for Google firmware updates first though.

MacDaddyFPV, 2023-10-30:
> If using a video system which doesn’t support colour fonts simply `set displayport_msp_fonts = 0,0,0,0` and you’ll get just the monochrome warnings etc.
> 
> i suggest you check for goggle firmware updates first though.

Thanks for the reply Steve,

I am quite sure HDZERO supports colour fonts as all of the other warnings and OSD elements are there. It is only when the Battery Voltage goes critical that the issue occurs. It is also functioning without issue on BF 4.4.2.

I have used the current goggle firmware along with the RC goggle firmware. I have also tried the current VTX firmware along with the RC VTX firmware.

Cheers

SteveCEvans, 2023-10-30:
Do the RX related OSD elements display red when your tx is turned off, or not at all?

Perhaps try 0.1.1.1 and see if that’s any better.

MacDaddyFPV, 2023-10-30:
Hi Steve,

My apologies. I tried your suggestion of set displayport_msp_fonts = 0,0,0,0 and it fixed the problem.
I was confused as I do see coloured fonts in the goggles osd, even with the above setting.
This will likely be an issue with a lot of users. Is it perhaps something that should be added to the 4.5 preset for HDZERO?

MacDaddyFPV, 2023-10-30:
I have created an update to the preset for 4.5 and submitted a pull request. 

Cheers

closed 2023-10-30 by MacDaddyFPV

SteveCEvans, 2023-10-30:
Whilst the individual characters do contain colour there is the concept of four fonts for normal/info/warning/critical text. This enables normal text, the GPS symbol etc. to be displayed in white/green/amber/red respectively. CRASH FLIP SWITCH is displayed in green, LOW BATTERY in amber, and LAND NOW in red, for example. Anything RX related is displayed in red if there’s an issue. The GPS symbol/count is red with no lock, amber when there are insufficient satellites for GPS rescue, and green when fully locked. I understand that HDZero intend supporting all for colours in a forthcoming release.