betaflight/betaflight #13148: EDT not working in BF 4.5-zulu (master) on IFLIGHT_BLITZ_F722 board

neoxic, 2023-10-30:
### Describe the bug

Extended DSHOT telemetry is not working when the iFlight BLITZ F722 board is running latest BF 4.5-zulu. The settings are all default, i.e. the issue manifests itself right after flashing and enabling bidirectional DSHOT in the Motors tab. I also tried to build against earlier commits from the drop-down list with the same result.
[image]
[image]
Not sure what they mean, but the following numbers look kinda weird:
[image]

The same board running BF 4.4.2 with all default settings has no problems:
[image]

### To Reproduce

- Flash the IFLIGHT_BLITZ_F722  board with latest BF 4.5-zulu with the "Erase Chip" turned on
- Go to the Motors tab in BF Configurator
- Enable "DSHOT 600" and "Bidirectional DSHOT"
- Enable Test Mode (observe no temperature readings)
- Go to CLI and enter `dshot_telemetry_info`

### Expected behavior

- Non-zero temperature in the Motors tab
- RTVCS type in the `dshot_telemetry_info` command's output

### Support ID

```txt
*** Support data submitted *** 
Id: "0a9f06cb-d829-4fa3-a3cf-75a70079da0b"

# copy ID and provide to the betaflight team.
*** Support data submitted *** 
Id: "645480e7-f7d9-489e-a1b6-c53a3b7e51f6"
```

### Flight controller

IFLIGHT_BLITZ_F722

neoxic, 2023-10-30:
The ESCs are running ESCape32 firmware.

neoxic, 2023-10-30:
Additional info. A different AT32F435-based AIO board running BF 4.5-zulu (the ESCs are running the exact same ESCape32 firmware revision) has no problems:
[image]
[image]

hyp0dermik-code, 2023-11-01:
@neoxic Can you confirm whether it was working on a previous version i.e. 4.4.0? Do you have another board with an F722 to rule out whether it is board or chip specific?

neoxic, 2023-11-01:
> @neoxic Can you confirm whether it was working on a previous version i.e. 4.4.0?

As mentioned above, the same board running BF 4.4.2 with all default settings has no problems. Does BF 4.4.2 count a previous version?

> Do you have another board with an F722 to rule out whether it is board or chip specific?

No, I don't. But how can you rule out if it's chip specific with another F722 board? I mentioned above that a different board with a different chip works fine, so yes, it's chip specific.
 
> Can you also post the output of `dma show` and `timer show` for the board mentioned in the OP?

Is this info not included in the Support ID output (I posted it above)? What is the Support ID output for then? Here you go:
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_SDI 3
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 1
DMA1 Stream 3: SPI_SDI 2
DMA1 Stream 4: SPI_SDO 2
DMA1 Stream 5: SPI_SDO 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_SDI 1
DMA2 Stream 3: SPI_SDO 1
DMA2 Stream 4: MOTOR 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 4

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

hyp0dermik-code, 2023-11-01:
> > @neoxic Can you confirm whether it was working on a previous version i.e. 4.4.0?
> 
> As mentioned above, the same board running BF 4.4.2 with all default settings has no problems. Does BF 4.4.2 count a previous version?

Thanks, sorry missed that. Yes it does. When you get a chance, could you post the results from ```dma show``` and ```timers show``` with 4.4.2 installed, but please see the bottom comment before reflashing
> 
> > Do you have another board with an F722 to rule out whether it is board or chip specific?
> 
> No, I don't. But how can you rule out if it's chip specific with another F722 board? I mentioned above that a different board with a different chip works fine, so yes, it's chip specific.

This would isolate a problem with the DMA/timers layout with that particular board. I have an F722 that works fine in bitbang mode but I have not been able to test in DMA.
> 
> > Can you also post the output of `dma show` and `timer show` for the board mentioned in the OP?
> 
> What do you mean by the OP? I can get the output for the F722 board that has the issue:

Thats what I was looking for, thanks. This shows the bidirectional is in DMA mode.
Can you run the following commands please and retest:
```
set dshot_burst = AUTO
set dshot_bitbang = ON
save
```

neoxic, 2023-11-02:
Enabling dshot bitbang makes the issue disappear:
```
# dshot_telemetry_info
Dshot reads: 241429
Dshot invalid pkts: 3
Dshot directionChange cycles: 0, micros: 0

Motor    Type   eRPM    RPM     Hz Invalid   TEMP    VCC   CURR  ST/EV   DBG1   DBG2   DBG3
=====  ====== ====== ====== ====== ======= ====== ====== ====== ====== ====== ====== ======
    1   RTVCS      0      0      0   0.00%     27  11.25      0      0      0      0      0
    2   RTVCS      0      0      0   0.00%     28  11.25      0      0      0      0      0
    3   RTVCS      0      0      0   0.00%     27  11.25      0      0      0      0      0
    4   RTVCS      0      0      0   0.00%     29  11.25      0      0      0      0      0

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

I will flash VF 4.4.2 to get you the `timer/dma show` output.

> I have an F722 that works fine in bitbang mode but I have not been able to test in DMA.

What is the reason why you haven't been able to test your F722 board in DMA?

neoxic, 2023-11-02:
Ok, flashed BF 4.4.2. The requested output:

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 1
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: MOTOR 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 4

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

hyp0dermik-code, 2023-11-02:
Thanks for your results, it seems like we have narrowed the problem down to a change in DMA bidirectional between 4.4.2 and 4.5z

I believe the the Betaflight team was focusing on bitbang going forwards, but somebody else like @SteveCEvans or @haslinghuis would be the people to take it from here.

> > I have an F722 that works fine in bitbang mode but I have not been able to test in DMA.
> 
> What is the reason why you haven't been able to test your F722 board in DMA?

Board layout seems to be incompatible with DMA, see the outputs from my timer/DMA show here:
https://github.com/betaflight/betaflight/issues/12994#issuecomment-1713768374

Don't quote me, another user seemed to think it should work, but never did even with the suggested changes to the ADC DMA channel. AFAIK only certain pins can be mapped to certain channels, this is one of the advantages of AT/G4/H7 as the DMA mux eradicates this problem.

That AiO has now been removed so I am unable to provide any further data from it for the near future until it finds another home.

neoxic, 2023-11-02:
> Thanks for your results, it seems like we have narrowed the problem down to a change in DMA between 4.4.2 and 4.5z

Umm.... Don't get me wrong, but personally, I fail to see how a "change in DMA" (whatever that might be) can be the root cause of EDT malfunction. The thing is that regular bidirectional DSHOT RPM telemetry seems to work without issues. And since EDT is just a logical abstraction over the same physical signalling, I don't see how RPM telemetry can continue to work where EDT can't.

haslinghuis, 2023-11-02:
https://github.com/betaflight/betaflight/pull/11851#issuecomment-1258628297

hyp0dermik-code, 2023-11-02:
> [#11851 (comment)](https://github.com/betaflight/betaflight/pull/11851#issuecomment-1258628297)

Not sure how that is relevant? Motors tab sets edt on when enabled, which is why it works when bitbang is enabled

AlkaMotors, 2023-11-05:
[image]
[image]

This is what the Blitz F7 is sending out for dshot commands when dshot bit-bang is OFF. It sends a partial packet.

damosvil, 2023-12-04:
> Umm.... Don't get me wrong, but personally, I fail to see how a "change in DMA" (whatever that might be) can be the root cause of EDT malfunction. The thing is that regular bidirectional DSHOT RPM telemetry seems to work without issues. And since EDT is just a logical abstraction over the same physical signalling, I don't see how RPM telemetry can continue to work where EDT can't.
> 
> EDIT: The only valid cause that might break EDT at the physical level is a bug that prevents EDT from being enabled in the first place. In other words, due to some bug, EDT is never properly enabled via corresponding DSHOT commands.

To enable EDT in the FC side Betaflight has first to:
- Send dshot command 13 for 10 times
- Receive and process a zero status frame

[image]

neoxic, 2023-12-04:
@damosvil 
I'm very well aware of that. That was exactly my point that EDT is never enabled due to some bug, i.e. DSHOT EDT enable commands are never transmitted properly at the physical level as you can clearly see on screenshots @AlkaMotors  posted above.

Prajwal-Koirala, 2023-12-26:
This is an issue about Extended DSHOT telemetry not working on the IFLIGHT_BLITZ_F722 board running Betaflight 4.5-zulu. The user tried flashing the board with earlier versions of Betaflight, but the problem persists. They also ruled out a hardware problem by testing a different board with the same ESCape32 firmware and a different AIO board with an AT32F435 chip. The issue seems to be specific to the IFLIGHT_BLITZ_F722 board and Betaflight 4.5-zulu. Enabling DSHOT bitbang mode resolves the problem, but the user would like to use DMA mode. The cause of the problem is likely a change in DMA between Betaflight 4.4.2 and 4.5-zulu.

I hope this is helpful!

howels, 2024-02-22:
Suggest we move this board to use `DSHOT_BITBANG` by default.

Please let us know if there are any issues with that.

neoxic, 2024-02-24:
Wow. Nice work, eh!

haslinghuis, 2024-02-24:
@neoxic Feel free providing further info on the issue. It only got closed due to the bitbang configuration change. If not resolved it should be re-opened - but this case seems board specific - we just need to find out why.

neoxic, 2024-02-25:
@haslinghuis With bitbang enabled by default, the issue doesn't go anywhere. It simply gets hidden whether it's board specific or not because no code change has been made to address that. What is the point of having the `DSHOT_BITBANG` option after that if disabling it will lead to broken DSHOT? Should you need more info, feel free to let me know what exactly you need, and I'll do my best to provide that.

JBKingdon, 2024-02-25:
Hi @neoxic  
We know we have more work to do on pwm based dshot, so just having this issue closed doesn't mean that nothing more will be done. I assume you're joking with the question about why we would want to keep the option as obviously it's still useful on many other boards.

neoxic, 2024-02-25:
@JBKingdon Yes I mean this particular board only. I know dshot bitbang is unavoidable on many other boards. I just don't get it why a broken feature (DMA-based DSHOT) is simply plastered over with a workaround just to get it out of the view despite the fact that it worked in the previous version without issues. It's nothing short of a regression. If this is what you need, well, feel free to do whatever you think is right. I'm no one to educate anyone here what to do with this issue.

JBKingdon, 2024-02-25:
I see what you mean. I don't have a strong opinion either way about whether the issue is open or closed, but your opening the issue was appreciated. Please don't feel put off from reporting future problems - we need that feedback. We don't have a whole lot of people that are able to look deeper into the problem right now, so this work-around should minimise the number of people who are inconvenienced in the short term, and hopefully longer term we can take another look at what's going on.

haslinghuis, 2024-02-25:
> Another problem with that is that I don't see any point in reporting issues in the future anymore because they get closed just because they are standing in the way of an upcoming release. If it can't be fixed right now, it's fine! But leave it open - why close it?

Closing of the issue was explained here: https://github.com/betaflight/betaflight/issues/13148#issuecomment-1962761220

neoxic, 2024-02-25:
@haslinghuis 
> Closing of the issue was explained here: https://github.com/betaflight/betaflight/issues/13148#issuecomment-1962761220

_"It only got closed due to the bitbang configuration change."_ -- Yes, and I didn't get that to be honest. As mentioned here (https://github.com/betaflight/betaflight/issues/13148#issuecomment-1962763107), the issue doesn't go anywhere with that configuration change because if someone disables bitbang, it will re-surface again. If DMA-based DSHOT is not officially supported on this board anymore due to whatever reason, then this particular issue ticket has no meaning and can be closed as I see it.

closed 2026-09-26 by blckmn