betaflight/betaflight #13149: Receiver not detected on BF 4.4.2 for Nazgul V2 (succex E F4)

bimasakti1985, 2023-10-30:
### Describe the bug

Updated to BF 4.4.2 from BF 4.2.4 for nazgul V2 and suddenly receiver not detected on BF. Downgrade to BF 4.4.0 and receiver now  become detected.

### To Reproduce

Update nazgul v2 (succex E F4) to 4.4.2

### Expected behavior

Receiver should be automatically detected upon upgrading if using default uart for the receiver. 

### Support ID

```txt
d8440690-dc9d-4728-83b7-554a8a6a97d9
```

### Flight controller

Succex E F4

### Other components

VTX : Vista
RX : Crossfire
GPS : Beitian

### How are the different components wired up (including port information)

Using iFlight default setup

haslinghuis, 2023-10-30:
This issue is fixed in 4.4.3-zulu (development build)

mituritsyn, 2023-11-24:
@bimasakti1985 could you please check on latest release and confirm it is fixed?

closed 2024-01-01 by github-actions[bot]