betaflight/betaflight #13151: Bidirectional DShot not working

hyp0dermik-code, 2023-11-01:
### Describe the bug

From @crteensy:
Upgraded BF on an old Mamba F405 FC (FURYF4OSD). It showed ESC telemetry before, but now I get 100% error rate in the motors tab. This ESC reported back via DShot300 when the FC was running BF 4.2.8

### To Reproduce

Flash 4.5z, enable bidirectional DShot, observe 100% errors

### Expected behavior

Bidirectional Dshot works

### Support ID

```txt
##
```

### Flight controller

Mamba F405 FC (FURYF4OSD)

hyp0dermik-code, 2023-11-01:
4.5z:
[image]
```# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE```

After ```set dshot_burst = off
Set dshot_bitbang = off
Save```:
![image](https://github.com/betaflight/betaflight/assets/177669/a03047fb-8b40-4d7f-9a37-17275778de76)
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 1
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 3
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

hyp0dermik-code, 2023-11-01:
4.4.0:
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 1
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 3
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```
<not working>

After enabling bitbang on 4.4.0 it works
```
set dshot_burst = AUTO
set dshot_bitbang = ON
save
```
[image]
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH2 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

SteveCEvans, 2023-11-01:
Upgrade your ESC FW.

hyp0dermik-code, 2023-11-01:
> Upgrade your ESC FW.

Could be that, @crteensy reports AM32 1.95

I thought the auto margin changes mitigated this?

hyp0dermik-code, 2023-11-03:
Confirmed issue in AM32 version

closed 2023-11-03 by hyp0dermik-code