betaflight/betaflight #13152: RTC on won't show on OSD changing timezone_offset_minutes from 0 to -240

TTGG4400, 2023-11-02:
### Describe the bug

RTC won't show on my OSD after I changed timezone_offset_minutes from 0 to -240 - UNLESS - I hook up USB and startup BFC - then it shows up in my OSD.  And if I remove the USB, it will stay there.  But if I power toggle my LR4 drone, the RTC is gone.  My OSD is on DJI 03 Goggles 
2.  I'm running BF 4.4.2.  I've tried this several times - all with the same result - the RTC won't show on my OSD unless I have BFC connected.  And it happened after I changed timezone_offset_minutes from 0 to -240 earlier today.  I changed it back to 0, but it still won't show on the OSD without BFC hooked up to it.  This was fine, until today, when I changed timezone_offset_minutes from 0 to -240.  Makes no sense.

### To Reproduce

See bug description.

### Expected behavior

Timezone should show on the OSD.

### Support ID

```txt
Id: "a09e03f6-0dbd-4bb5-8994-271fa677c8e7"
```

### Flight controller

Goku Versatile F405 Pro Mini Stack ( F405 FC+ 40A ESC ) 8Bit 2-6S 20 x20 Plug&Play plug O3 Air unit

### Other components

Craft is the Flywoo Explorer LR4 03.

### How are the different components wired up (including port information)

Wiring is stock.

TTGG4400, 2023-11-03:
NEVERMIND - when I went out to fly today, the RTC was back on the OSD - no idea what happened.  Also, it's correct - it's using the timezone_offset_minutes value of -240 I'd set it to yesterday.

closed 2023-11-03 by TTGG4400

jonathanoeijoeng, 2023-12-20:
how to show rtc in osd? I have enabled in OSD and connected to remote but rtc was no showed up. I use analog vtx

TTGG4400, 2023-12-20:
> how to show rtc in osd? I have enabled in OSD and connected to remote but rtc was no showed up. I use analog vtx

Dunno.  But I can tell you it never works indoors hooked up to my laptop.  IOW, I only see the RTC on the OSD when I'm outside actually flying it.

TTGG4400, 2024-03-21:
I've since come to realize, that the BF OSD RTC requires a GPS that has sats, in order to display the time.  As none of our FCs have internal batts to support RTCs on em.  Would be helpful if the BFC OSD support for RTC indicated this fact of FPV life.

TTGG4400, 2025-08-09:
Actually, it seems to be the F405 FC at fault.  It cannot handle the load, supposedly.  

And I have since learned that a GPS is not needed, for the RTC to show on the OSD.  I’ve seen RTCs displayed on OSDs on quads without GPS.  

But in those cases, I do not know how the FC gets RTC.  Perhaps from the DJI 04, or the Tx?