betaflight/betaflight #13156: OSD does not work

Asket2000, 2023-11-06:
### Describe the bug

Betaflight - ver. 4.4.2
Flight controller - SpeedyBee F405 v3
VTX - Rush Tank Solo 5.8GH
Helmet -
[image]
 SkyZone Cobra X v2

With this configuration, OSD does not work correctly

If we install Betaflight - ver. 4.4.0
Everything is working fine.

### To Reproduce

With this configuration, OSD does not work correctly

If we install Betaflight - ver. 4.4.0
Everything is working fine.

### Expected behavior

Correct mistake

### Support ID

```txt
# ###

# # Problem description

# # OSD not work

# ###

# version
# Betaflight / STM32F405 (S405) 4.4.2 Jun  9 2023 / 12:46:31 (23d066d08) MSP API: 1.45

# config: YES
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF405V3

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.15V, Core temp=37degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3652, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MSP (30 x 13)
BUILD KEY: 0ca0088baa325289bf95a3ca1c7c01bc (4.4.2)
System Uptime: 77 seconds, Current Time: 2023-11-06T21:21:35.881+00:00
CPU:34%, cycle time: 312, GYRO rate: 3205, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x3, 15558144kB, 09/2022, v8.0, 'SS16G'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.2 Jun  9 2023 / 12:46:31 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF405V3
manufacturer_id SPBE

# resources
resource BEEPER 1 C05
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B05
resource MOTOR 8 B04
resource LED_STRIP 1 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource LED 1 C08
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource ESCSERIAL 1 NONE
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource SDCARD_CS 1 A15
resource SDCARD_DETECT 1 NONE
resource PINIO 1 C03
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer B09 AF2
# pin B09: TIM4 CH4 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 0
# SPI_MOSI 2: DMA1 Stream 4 Channel 0
dma SPI_MOSI 3 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 0
# SPI_TX 2: DMA1 Stream 4 Channel 0
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B09 NONE
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature OSD
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 2
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SDCARD
set dshot_burst = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 386
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set sdcard_detect_inverted = OFF
set sdcard_mode = SPI
set sdcard_spi_bus = 2
set sdio_clk_bypass = OFF
set sdio_use_cache = OFF
set sdio_use_4bit_width = OFF
set system_hse_mhz = 8
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set usb_msc_pin_pullup = ON
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "0f2d5e8a-563a-40f6-98ec-62201d926ea6"

# copy ID and provide to the betaflight team.
```

### Flight controller

SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack (Aliexpress)

nerdCopter, 2023-11-10:
Probably due to shared SPI on SpeedyBeeF405V3. please reference https://github.com/betaflight/betaflight/issues/12583 .
Please disable blackbox logging and test.

tuvinez, 2023-11-29:
I have same Issue with SD card installed. 4.4.0 works well.

haslinghuis, 2023-11-29:
> Betaflight - ver. 4.4.2

Flash 4.4.3

closed 2023-11-29 by haslinghuis