betaflight/betaflight #13167: GPS Sat Count Not Showing Up in OSD 4.5 Zulu

herokme, 2023-11-13:
### Describe the bug

I currently have my gps set up perfectly in BF 4.4.2. If I flash 4.5 zulu, all my osd elements are working properly, EXCEPT my gps sat count. I can see on my radio telemetry that I have 12+ sats, but in my osd, I don't even get the element showing up. Every once in a while, my sat count would flash on the osd for a second, and then disappear. I've had other people reprt that their sat count works for 4.5 zulu, not sure why mine isn't working

HDZero, BF 4.5 zulu, Speedybee F405 FC (also tried on Span F722 BT HD V2)

### To Reproduce

Power on quad, get gps sat count (can see in radio telemetry), but none in osd (it works perfectly in 4.4.2)

### Expected behavior

Sat count should appear in OSD

### Support ID

```txt
0b705b0a-0710-49b7-bad0-2327b61b8f6f
```

### Flight controller

Speedybee F405 V3 (also tried on Span F722 BT HD V2)

### Other components

Hdzero freestyle V1 VTX, Span 50a ESC

### How are the different components wired up (including port information)

GPS on UART 2, properly configured (working perfectly in 4.4.2, just not 4.5 zulu)

Loatan, 2023-11-13:
Сonfirm. RSSi dbm missing too.

SteveCEvans, 2023-11-13:
See https://github.com/betaflight/betaflight/issues/13147#issuecomment-1784353072

closed 2023-11-13 by SteveCEvans