betaflight/betaflight #13173: DJI O3 OSD overlay does not work with 4.4.2

SJurick, 2023-11-19:
### Describe the bug

I upgraded from 4.3.2 to 4.4.2 on my SpeedyBee, and now I do not get the OSD overlay on my DJI goggles V2 as I did before.  I tried both ways:

### To Reproduce

Setup Peripherals for USB VCP to (VTX (MSP+DisplayPort), AUTO in BF ports section

Manually through BF CLI
set osd_displayport_device = MSP
set displayport_device_serial = 0

BF says the above CLI commands are invalid.  So something changed.  Is there a new set of commands or something?  I don't want to upgrade any of my other quads until I can be sure the display will work.

Steve

### Expected behavior

I expect the OSD overlay from BF to appear in my goggles.

### Support ID

```txt
Id: "1cb03a3e-d3da-4c62-85a2-29b0f7cd53ca"
```

### Flight controller

SPBE/SPEEDYBEEF7V3(STM32F7X2)

haslinghuis, 2023-11-19:
Did you check the [release notes](https://betaflight.com/docs/release/Betaflight-4.4-Release-Notes#2-hd-osd)?

SJurick, 2023-12-01:
Thank you Mark.  I checked them and the CLI was using in prior versions had
changed.  All working now!

closed 2023-12-01 by haslinghuis