betaflight/betaflight #13175: Betaflight - ESP32 Firmware

ArshaanTech, 2023-11-20:
### Is your feature request related to a problem? Please describe

Hello, I just want to request to make a request for making a firmware for ESP32.
The ESP32 has four SPI peripheral devices.
The ESP32 chip has 3 UART controllers.
ESP32 has 2 I2C controllers.

I think this configuration is enough to make a stable flight controller.

### Describe the solution you'd like

I actually think it would be nice to try and make a betaflight firmware for ESP32 boards.
Sensors that can be used:
Gyro - MPU6050 i2c OR MPU6500 SPI etc
Baro - BMP280, i2c
magnetometer - HMC5883 i2c, QMC5883 i2c
GPS - neo 6m UART
Receiver - PPM / SBUS or CRSF UART

### Describe alternatives you've considered

I have seen a firmware for betaflight on ESP32, but it needs some improvements which I think should be made like to add GPS support, Failsafe, Return to launch etc.

ASDosjani, 2024-03-12:
I asked it earlier from the devs and they said it would require a ton of work to port because the architecture differs a lot from STM32 MCUs. So it's very unlikely to happen.

ArshaanTech, 2024-04-17:
i see, No luck then i guess, although it would be really nice if we would get a ESP32 specific firmware.

closed 2026-09-26 by blckmn