betaflight/betaflight #13179: Mamba mk4 h743 v2 blackbox logging rate higher than 1/4 2khz causing flyaways

NspaceCITY713, 2023-11-23:
### Describe the bug

When I set my logging rate higher than 1/4 2khz  the quad gets a low frequency spike at 55hz. Thought the issue was somewhere else till I started playing with the logging rate. Today I tested the quad at 1/1 8khz, noise spikes at 55hz and causes flyaway. If I lower my blackbox logging rate, down to 1/4 2khz you can still see the noise in blackbox but it’s very faint and not noticeable in flight. This is in bf 4.3.0 as when I tried 4.4 and above the quad immediately flies away, with no changes besides flashing the newer firmware. It’s like the gyros hardware filtering has been changed on 4.4 vs 4.3. Blackbox also doesn’t always record. Will only log first flight only. If you do multiple flights, it’s only recording first one, not making multiple parts if you do multiple flights on same battery. It picks and chooses when it don’t wanna record, when it doesn’t record it will have a very small file instead. As far as the logging rate creating the flyaways, I’m not sure if this is a filtering issue on the gyro level or a bad design, never had logging rate create noise on a quad like this. 

### To Reproduce

Set blackbox logging rate to 1:1 8khz 

### Expected behavior

Not cause flyaways and log appropriately  

### Support ID

```txt
5614bae1-2480-488a-b420-b112a84b0f8a
```

### Flight controller

Mamba mk4 h743 v2 - pyrodrone 

### Other components

Bought as stack with 65amp esc, o3 air unit, happy model elrs ep1 

### How are the different components wired up (including port information)

Flight controller to esc is soldered on both. Wires ran underneath esc. Esc is rotated 180° with 35v 1000uf Panasonic fm cap directly on esc and one on the vbat of fc. 

### Add any other context about the problem that you think might be relevant here

Flight controller has always had flyaway issue in this build. Originally tried bf 4.4. Had wiring between esc and fc and was causing gyro noise. After rerouting cables, flyaways stopped but oscillations were to bad. Went and flashed 4.3.0 which is what the fc came with, and was flyable. Noticed noise at 50hz still but not much. Thought it was the wiring still. After upping my logging rate the quad flew away again, so I dropped it and it stopped. Tested multiple times without making any changes, flyaways at any higher rate than 2khz. Not sure if this is a hardware issue or software but deff seems like the gyros internal filtering is less on 4.4 and above. 
[image]
[image]

NspaceCITY713, 2023-11-23:
[image]
[image]

SteveCEvans, 2023-12-23:
@NspaceCITY713 Could you please try the latest development build as https://github.com/betaflight/betaflight/pull/13218 fixed a rather nasty issue with LED_STRIP which would have badly impacted timing, and thus filtering. I'm keen to hear if this fixes your issue.

NspaceCITY713, 2023-12-23:
@SteveCEvans I will give it a shot. I recently went to 4.5zulu and same issues occur, it is 100% an issue with blackbox logging. Not sure if it's a bad placement of the flash drive, W25N01GV, it's mounted bottom side board directly underneath the gyro, dual icm 42688p. Noise is very apparent on bf 4.4 and above. It is there on 4.3 but I can lower logging rate enough to get by. On 4.5 if I turn on blackbox logging it makes quad vibrate like crazy. Almost causing flyaways. Logs show noise at 45-50hz. If i turn off blackbox logging, no more vibrations and it flies perfectly. I have other issues with blackbox as well. It will only record the first flight. If you disarm and rearm the second log is very small file with nothing in it. Only work around is to unplug battery. I will try and see about the above fix and see what happens.

SteveCEvans, 2023-12-24:
The W25N01GV is unfortunately not well supported. With other FLASH devices the data is transferred in the background using DMA whilst the processor can get on with more flight critical tasks, however the driver for this one device has to switch banks etc. and blocks execution during the write. Quite frankly if you have an FC with one of these devices I’d invest in an openlager. I considered rewriting the driver to be non blocking using a sequence of interrupt driven DMAs as per the SPI ELRS driver, but quite frankly it would be a lot of effort for minimal benefit.

NspaceCITY713, 2023-12-24:
I've got a Openlager on order in the mail it'll be here just after Xmas. I think that was the most practical solution as they run $20. At first I couldn't figure it out but now I'm positive the flash drive is the issue. With logging off, it flies great. Mamba discontinued these boards the h743 v2 and they were fairly new. Claimed it was bc of a chip shortage.... I tuned it by ear till I can get an Openlager device and take it further. Btw Thanks for the info!!

Prajwal-Koirala, 2023-12-26:
Issue #13179 on the Betaflight GitHub is about a problem encountered with the Mamba mk4 h743 v2 flight controller when setting the blackbox logging rate higher than 1/4 2kHz. The user reports that at higher logging rates, a low-frequency spike at 55Hz occurs, leading to flyaways of the quadcopter. This issue was specifically observed in Betaflight firmware version 4.3.0, and when trying version 4.4 or higher, the quadcopter immediately flew away without any changes besides the firmware update. The user suspects that the gyros' hardware filtering might have changed between versions 4.3 and 4.4, contributing to this issue. Additionally, the blackbox does not always record as expected, often only logging the first flight and failing to record subsequent flights on the same battery【52†source】.

The flight controller is part of a setup that includes a 65amp ESC, an O3 air unit, and a Happy Model ELRS EP1【53†source】. The user also added context that the flight controller always had a flyaway issue in this build. Initially, they suspected wiring between the ESC and the flight controller was causing gyro noise. After rerouting the cables, the flyaways stopped, but there were still oscillations. Testing with different logging rates revealed that the quad flew away at rates higher than 2kHz, suggesting a possible issue with hardware or firmware, particularly concerning the gyros' internal filtering【54†source】.

closed 2024-02-08 by github-actions[bot]