betaflight/betaflight #13180: BMP280 Barometer Unstable in 4.4.X

vincentpoont2, 2023-11-23:
### Describe the bug

In 4.4.X, compare to 4.3.2, BMP280 barometer waveform shows appear to be unstable and has much higher variance. The flight control test environment is all tested indoors under static conditions. The firmware is upgraded within Betaflight Configurator

[image]
[image]

[image]

### To Reproduce

Tested with Kakute H7 V1.3 & Kakute H7 V2.

### Expected behavior

4.4.X should show similar value as 4.3.2. 

### Support ID

```txt
N/A
```

### Flight controller

Kakute H7 V1.3 & Kakute H7 V2

haslinghuis, 2023-11-24:
After bringing back `taskAltitudeCheck` in tasks.c (changed in https://github.com/betaflight/betaflight/commit/f17a2af8c3aba115f071ccb2385d418ca15c35e3) getting less variance:

[image]

on master:

[image]

haslinghuis, 2023-11-25:
@SteveCEvans @ctzsnooze

jsk2084, 2023-11-27:
The altitude trace is smoothed when armed, whereas before it was smoothed all the time. To test while connected to config, enable motor test mode and arm (with props off). I think default filtering is less now, renamed as 'altitude_lpf' in CLI. A lower value while armed should show similar traces as 4.3.2 and earlier.

ctzsnooze, 2023-12-12:
This is not a bug.  We are using different Baro sampling and filtering in 4.4 than 4.3.  There is nothing wrong with the BMP380 data.  It works better in  4.4. than 4.3.

closed 2023-12-12 by ctzsnooze