betaflight/betaflight #13181: Mag does not work and softserial port missing in 4.5zulu

aowi7280, 2023-11-24:
### Describe the bug

Everything works in 4.4.3, but the same config does not work in 4.5.  Need the config for 4.5.

### To Reproduce

Flash 4.5zulu, paste in Diff from 4.4.3

### Expected behavior

GPS, mag, LEDs, softserial all work. 

### Support ID

```txt
Id: "4ac2588e-587c-4692-8b92-f34d9c03a638"
```

### Flight controller

Airbot OMNIBUSF4SD

### Other components

Tx: Frsky QX7 w_HappyModel 915MHz ES900TX ExpressLRS V3.3.1
Rx: R9MM w/ExpressLRS V3.3.1
Vtx: RushFPV Rush Tank SOLO
Lumenier Multi-functional RGB LED
Holybro Micro M8N GPS with Mag

### How are the different components wired up (including port information)

Configuration/MSP on USB VCP
Serial Rx on UART1
GPS on UART6
VTX TBS Smartaudio on SOFTSERIAL1

### Add any other context about the problem that you think might be relevant here

Just need the new way to configure features and resources for 4.5. From BF4.4.3
resource MOTOR 5 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A01
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature SOFTSERIAL
feature GPS
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

haslinghuis, 2023-11-24:
To configure mag look at which I2C port is defined in resources. For example with I2C port 2 used set:

    set mag_i2c_device = 2
    save
    
Soft serial resource pin has been moved [to 30 and 31](http://betaflight.com/docs/development/Serial#1-serial-port-identifier)

closed 2023-11-26 by haslinghuis

aowi7280, 2023-11-26:
The mag config worked.

But I tried both of these and still no soft serial port that I can use for Smartaudio VTx control. Would like to have the ability to run low power most of the time, then increase it for long range.

serial 30 2048 115200 57600 0 115200
serial 31 2048 115200 57600 0 115200

Andy