betaflight/betaflight #13188: Bf 4.5. Elrs stopped working.

Nikguy321, 2023-12-03:
### Describe the bug

Fc: iflight f7 blitz 30x30
Rx: radiomaster RP1 ELRS (uart 2)
Radio: Tx16s mkii 

The radio will connect to the quad (says telemetry recovered) but betafl8ght doesn't see my stick inputs. Downgrading the firmware back to 4.4 solves the issue 

### To Reproduce

Try upgrading iflight blitz f7 30x30 to BF4.5 with a radiomaster RP1 on uart2 

### Expected behavior

It should read the incoming stick and button inputs like normal. But it doesn't 

### Configurator version

10.10.0-RC1

### Flight controller configuration

```txt
#

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.3 Nov 14 2023 / 18:40:26 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name IFLIGHT_BLITZ_F722
manufacturer_id IFRC

# name: FlyWize

# feature
feature GPS
feature ESC_SENSOR

# serial
serial 0 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1550 2100 0 0
aux 2 13 3 1550 2100 0 0
aux 3 35 1 1525 2100 0 0
aux 4 36 7 1475 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 525
set dyn_notch_count = 1
set dyn_notch_q = 550
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 10,16,-4,1
set rc_smoothing_auto_factor = 250
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set fpv_mix_degrees = 25
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 280
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = WAAS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 105
set osd_vbat_pos = 2433
set osd_rssi_pos = 2402
set osd_tim_2_pos = 2208
set osd_flymode_pos = 2145
set osd_craft_name_pos = 2176
set osd_gps_speed_pos = 2426
set osd_gps_sats_pos = 2394
set osd_home_dir_pos = 2447
set osd_altitude_pos = 31129
set osd_warnings_pos = 14731
set osd_disarmed_pos = 2368
set osd_canvas_width = 30
set osd_canvas_height = 13
set debug_mode = GYRO_SCALED
set craft_name = FlyWize

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_static_hz = 78
set dterm_lpf2_static_hz = 157
set p_pitch = 54
set i_pitch = 96
set d_pitch = 61
set f_pitch = 143
set p_roll = 51
set i_roll = 92
set d_roll = 50
set f_roll = 138
set p_yaw = 40
set i_yaw = 72
set f_yaw = 108
set d_min_roll = 37
set d_min_pitch = 45
set thrust_linear = 30
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 9
set dyn_idle_min_rpm = 30
set simplified_pids_mode = RP
set simplified_master_multiplier = 115
set simplified_d_gain = 110
set simplified_pitch_d_gain = 105
set simplified_dterm_filter_multiplier = 105

rateprofile 0

# rateprofile 0
set rateprofile_name = freestyl
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 95
set pitch_srate = 95
set yaw_srate = 95

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

I am having trouble updating the RX for some reason on elrs configurator. It is saying it can't find it, though it has worked in the recent past.

nerdCopter, 2023-12-04:
what mode/rate is ELRS set? (in case this is indeed a bug)

Nikguy321, 2023-12-04:
> use Configurator 10.10-rc. make sure you built with `CRSF`. ![image](https://private-user-images.githubusercontent.com/56646290/287727130-de81e0f0-6d84-481c-85bb-4753182683d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3MDQ0NjAsIm5iZiI6MTcwMTcwNDE2MCwicGF0aCI6Ii81NjY0NjI5MC8yODc3MjcxMzAtZGU4MWUwZjAtNmQ4NC00ODFjLTg1YmItNDc1MzE4MjY4M2QzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA0VDE1MzYwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwMTg1YTg1NWZlMDRjNWI3ZjM1MTdjYjY5NTY3YmZmOTNhZjlkNmQ4MDcxYWM4NTA0ZjQwNzNmNTlkZDY2YjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.s5UCYRBkbzAhJ69QBbtY3mqVmIEZw1OJmg8ubNOwlJc)
> 
> CLI:
> 
> ```
> feature RX_SERIAL
> set serialrx_provider = CRSF
> set serialrx_inverted = OFF
> set serialrx_halfduplex = OFF
> save
> ```
> 
> FAQ: https://hackmd.io/@nerdCopter/S1N9D9rQh
I'm going to confirm this is all correct tonight if I can. 

> what mode/rate is ELRS set? (in case this is indeed a bug)

I'm using a hybrid switch mode. Model match off. 500hz. Tried with telemetry both on (multiple ratios) and with it off. 

Thanks

Prajwal-Koirala, 2023-12-26:
Issue #13188 on the Betaflight GitHub describes a problem with an iFlight F7 Blitz 30x30 flight controller and a Radiomaster RP1 ELRS receiver (connected via UART 2) used with a Radio Tx16s mkii. After upgrading to Betaflight firmware 4.5, the user reports that the radio connects to the quadcopter (indicated by the telemetry recovery message), but Betaflight does not register the stick inputs from the controller. Downgrading the firmware to version 4.4 resolves this issue【60†source】.

The expected behavior with the upgraded firmware is for Betaflight to read the incoming stick and button inputs normally, which it fails to do【61†source】. Additionally, the user mentions having trouble updating the receiver (RX) using the ELRS configurator, which indicates that the receiver is not being recognized, even though it worked correctly in the past【62†source】.

These issues suggest a potential compatibility problem between the hardware components and the newer version of the Betaflight firmware, which may require further investigation or a firmware update to resolve.

closed 2024-02-08 by github-actions[bot]

michb90, 2024-02-13:
Hello, i have the same issue, any update how to solve it? Already tried to downgrade to BF 4.4 with no effect. I have jumper t-20 radio, same FC as listed here...

Nikguy321, 2024-02-13:
> Hello, i have the same issue, any update how to solve it? Already tried to downgrade to BF 4.4 with no effect. I have jumper t-20 radio, same FC as listed here...

So I guess backing up with a preset from 4.4 doesn't work.... 

I had to just screenshot each page and then manually recreate my settings in 4.5... then it worked fine.

michb90, 2024-02-13:
> > Hello, i have the same issue, any update how to solve it? Already tried to downgrade to BF 4.4 with no effect. I have jumper t-20 radio, same FC as listed here...
> 
> So I guess backing up with a preset from 4.4 doesn't work....
> 
> I had to just screenshot each page and then manually recreate my settings in 4.5... then it worked fine. Just flash firmware again and don't apply your backup. Then manually remake it

Hi, tried once more. Flash reciever through wifi, it works fine. Then upgrade to bf 4.5, load backup from 4.4. and its working...