betaflight/betaflight #13208: Change LED colour with VTX channel on HDZero

fireduckfpv, 2023-12-08:
### Is your feature request related to a problem? Please describe

Not sure if this is the correct place to post this, but I would like the LED colors to change when I change VTX channel. It works on analog (SmartAudio) but not on HDZero (MSP)

### Describe the solution you'd like

I would like the LED colour to change when I change VTX channel

### Describe alternatives you've considered

You can change the colour manually, but it would be better if it happened automatically, like on analog.

SteveCEvans, 2023-12-08:
This would require the VTX to communicate the channel to the FC which I don’t believe it does at present. An alternate approach could be via the 
LUA script if it could pick up ELRS backpack channel settings and then set the channel number in the FC, but that sounds overly complicated.

fireduckfpv, 2023-12-08:
But if you use Betaflight to set the vtx channel, could not Betaflight also set the colour automatically?

sugaarK, 2023-12-10:
> But if you use Betaflight to set the vtx channel, could not Betaflight also set the colour automatically?

it did work then I got broken and now it works again.. you need to use Betaflight 4.5 Zulu and there is a 4.5 preset for it

sugaarK, 2023-12-10:
see this PR #13096

sugaarK, 2023-12-10:
fixed in #13096

closed 2023-12-10 by sugaarK