betaflight/betaflight #13220: Kiss passthrough is broken

KissUltra, 2023-12-11:
### Describe the bug

Kiss passthrough is broken. 

### To Reproduce

Use kiss esc. go to console and do escprog ki 255

### Expected behavior

Passthrough is working

### Support ID

```txt
on 4.3/4.4 command and its msp equivalent not enabled by default.on 4.5 dev it IS enabled, but doesnt work either.
```

### Flight controller

aikon f7

### Other components

Kiss esc, Ghost RX

haslinghuis, 2023-12-12:
According to comment in https://github.com/betaflight/betaflight/pull/12992 escserial does not work on F7 targets.

Think we need to remove USE_ESCSERIAL define from the F7 target for now.

Quick-Flash, 2023-12-12:
It can't work doing it as you described. 255 will never be a valid motor index. You should instead try:
``` 
 escprog ki 0  # Connect to ESC 1
 escprog ki 1  # Connect to ESC 2
 escprog ki 2  # Connect to ESC 3
 escprog ki 3  # Connect to ESC 4
 ```
 What occurred when you ran that command? Did you get any CLI output after running it?

KissUltra, 2023-12-12:
code wise if u use 255 it should switch to kissescall. thats what i am after. ---Cheers,Alex

KissUltra, 2023-12-12:
f7 should work. its just soft serial isnt it?---Cheers,Alex

Quick-Flash, 2023-12-12:
> code wise if u use 255 it should switch to kissescall. thats what i am after.

Unless I'm mistaken it will not do that. That number is meant to be the esc number (zero indexed) that you are trying to connect to via one way serial.

KissUltra, 2023-12-12:
255 should set all outputs to serial and flash all escs simultaneously. at least it was before like this. google it. 
---
Cheers,
Alex

> On 12 Dec 2023, at 21:03, Kevin Plaizier ***@***.***> wrote:
> 
> will

haslinghuis, 2023-12-12:
https://github.com/betaflight/betaflight/blob/03495f9273427c5203d7e6c9761692fcd275877e/src/main/drivers/serial_escserial.c#L971-L975

KissUltra, 2023-12-12:
yes, i checked. and it does not work. hence bug report.

KissUltra, 2023-12-12:
its not one way blheli serial :) its all the way kiss serial haha :)

KissUltra, 2023-12-13:
Btw. On kiss i just bit-bang it. :) speed is low, no need to fiddle with timers etc :) Just fyi.

howels, 2023-12-29:
Does any of the core team have KISS ESCs to test this? Sounds like only @KissUltra has the hardware to test this.
If you wish to improve support for this KISS ESC type then Betaflight welcomes patches, but if the ask if for the team to support your hardware then the guidelines are laid out in https://betaflight.com/docs/development/manufacturer/approved-partner-program

haslinghuis, 2024-01-01:
@KissUltra please submit a PR to fix the issue as we do not have samples to test with.

haslinghuis, 2024-09-10:
Stale

KissUltra, 2024-09-19:
#13911

commit a6d70be9 (2024-09-21): Fixes #13220: Fix kiss esc passthrough, Fix castle (escprog cc) command. (#13911)

* Fix kiss esc passthrough, Fix castle (escprog cc) command.

* Update src/main/drivers/serial_escserial.c

Thank you.

---------
commit d73d1a07 (2024-09-24): [4.5.2] Fix kiss passthrough (#13922)

Fixes #13220: Fix kiss esc passthrough, Fix castle (escprog cc) command. (#13911)

* Fix kiss esc passthrough, Fix castle (escprog cc) command.

* Update src/main/drivers/serial_escserial.c

Thank you.

---------
closed 2024-09-21 by haslinghuis