betaflight/betaflight #13229: DJI O3 osd not working for BETAFPV405 ver.4.4.3

zlodei2000, 2023-12-16:
### Describe the bug

I have BETAFPV Pavo Pico 20A version with BETAFPV405 flight controller. I flashed it with 4.4.1 firmware and used working dump. HD OSD on O3 is working correctly. If I flash 4.4.3 firmware from Betaflight Configurator and use my diff, HD OSD stop working on O3. Even standard DJI OSD is not visible in right lower corner of the screeen. Seems that O3 does not communicate with flight controller on 4.4.3 firmware.

[BF405 4.4.1_Pavo Pico_O3_20A v1.0_ELRS2.4g_20230725.zip](https://github.com/betaflight/betaflight/files/13693884/BF405.4.4.1_Pavo.Pico_O3_20A.v1.0_ELRS2.4g_20230725.zip)

[denisBTFL_cli_backup_Pavo_Pico_20231216_125449.txt](https://github.com/betaflight/betaflight/files/13693882/denisBTFL_cli_backup_Pavo_Pico_20231216_125449.txt)

### To Reproduce

Take BETAFPV405 , flash it to attached 4.4.1 hex, configure O3 OSD or use diff provided, test it. Than migrate to 4.4.3 and check OSD on O3

### Expected behavior

Fix wrong OSD behavior on 4.4.3 for BETAFPV405.

### Support ID

```txt
3c9fe7c0-129d-4d1c-9367-e76b1694377c
```

### Flight controller

BETAFPV405

### Other components

DJI O3 unit

haslinghuis, 2023-12-16:
Should use `osd_displayport_device = MSP`

zlodei2000, 2023-12-17:
> Should use `osd_displayport_device = MSP`

Check provided dump. It contains "set osd_displayport_device = MSP" (line 145). But OSD stops working after flashing to 4.4.3 (and experimental 4.5). Tried to flash with or without "full chip erase".

Prajwal-Koirala, 2023-12-26:
The issue reported in GitHub issue #13229 involves the HD OSD (On-Screen Display) of a DJI O3 unit not working correctly when used with a BETAFPV405 flight controller after upgrading the firmware to Betaflight 4.4.3. The user describes that the HD OSD functions properly under firmware version 4.4.1, but after upgrading to 4.4.3, the OSD stops working, indicating a potential compatibility or communication issue between the DJI O3 unit and the flight controller on the newer firmware【14†source】.

The setup includes a BETAFPV Pavo Pico 20A version with a BETAFPV405 flight controller and a DJI O3 unit. Unfortunately, there is no detailed information provided about how these components are wired up, which could be crucial for diagnosing the issue【15†source】.

The user has checked the configuration dump and confirmed that the "set osd_displayport_device = MSP" setting is present, which is necessary for the OSD functionality. They tried flashing the firmware with and without the "full chip erase" option, but the problem persisted, suggesting that the issue may be related to the firmware version itself rather than the configuration or flashing process【16†source】.

Given these details, the problem seems to be specific to the interaction between the DJI O3 unit and the BETAFPV405 flight controller when using Betaflight firmware version 4.4.3. This might be a bug in the firmware or a compatibility issue that needs to be addressed by the Betaflight development team.

macflyhub, 2023-12-27:
I have firmware 4.5.0 and using 10.10 rc1 configurator but can't get OSD menu to show up.
If I dump the diff - it has every single OSD setting including the "osd_displayport_device = MSP"
Switch back to configurator 10.9 and its all working fine.

haslinghuis, 2023-12-27:
On 4.5 try - might need a firmware upgrade on the goggles.
```
set displayport_msp_fonts = 0,0,0,0
save
```

macflyhub, 2023-12-28:
Thanks for reply but.... Its actually the OSD option in the Betaflight configurator itself down the left hand side thats missing.

dmyers7, 2024-01-10:
May seem simple, but make sure the following is enabled:
feature OSD

I've run 10.10 rc1 and rc2 and the OSD shows up.  Could not get the OSD to appear in my WalkSnail goggles until I did "osd_displayport_device = MSP" under 4.5 rc1 & rc2.  The HD screen in Configurator was not staying HD and then all of a sudden after that line it started behaving.  Quad working well now.

zlodei2000, 2024-01-10:
> May seem simple, but make sure the following is enabled: feature OSD
>
> I've run 10.10 rc1 and rc2 and the OSD shows up. Could not get the OSD to appear in my WalkSnail goggles until I did [...]

In my case the problem is clearly related to the BF firmware after version 4.4.0.  I made additional tests with iNAV using JHEF7DUAL  flight controller and it works fine. I will cautiously assume that it may be related to the O3 revision

JS36, 2024-01-20:
I have also encountered this. I managed to get it work one time by deactivate then activate MSP & VTX+DisplayPort but now I can't get the OSD started again... Maybe it was 4.4.2 because cannot get 4.4.3 to work now..

zlodei2000, 2024-01-21:
Try to downgrade to 4.4.1, it's working. There is no critical bugs in 4.4.1

asoell, 2024-01-21:
I'm having the same issue. I upgraded from 4.4.0 to 4.5 RC2 and the OSD wouldn't stay on HD in the OSD tab. It also didn't show up on the goggles.

So I downgraded back to 4.4.3, but it still doesn't show up on the goggles. I tried everything. And I'm sure it was set up correctly. In the end, I downgraded to 4.4.0, and the OSD works again. I upgraded to 4.4.2, and it also works. I upgraded again to 4.5 RC2, and it stopped working again.

zlodei2000, 2024-02-21:
Do not close this issue. It has been reported by several users

TheBandoHunter, 2024-02-24:
> Do not close this issue. It has been reported by several users

I'm dealing with this now.. I have 5 air units, but my oldest one is giving me problems showing osd in anything after 4.3.2. Is there a solution? or do I have to get a newer unit?

cupuyc1, 2024-03-15:
I have the same issue on 4.4.3 firmware. how to solve it?

asoell, 2024-04-15:
Don't close this, it has bot been resolved. It's consistent and reproducible, I cannot manage to get OSD on BF 4.4.3 and later! I don't know why nobody seems to care about this, basically BF4.5 is unusable for me.

zlodei2000, 2024-05-16:
Don't close this, it has bot been resolved.

haslinghuis, 2024-06-16:
A lot of users are using O3 without any issue. As I don't have any hardware to comment on this except on configuration not sure how I can debug the issue.

- Be sure the air unit is using latest firmware
- `OSD_HD` flash option in 4.5 defaults to digital FPV
- Setup clean. Do not use previous diffs or dump backups
- Verify using the right UART in ports tab and choose `VTX (MSP + displayport)`

Consider getting support on our [Discord server](https://discord.betaflight.com/invite)

kobynz, 2024-07-30:
This is still an issue, yet the issue keeps getting auto closed without anyone addressing the issue. 

The OSD on my Walksnail HD VTX stopped working after 4.4.2. 
I've just flashed 4.5.1 with the OSD (SD) option removed, and makes no difference.

asoell, 2024-07-30:
Honestly I don't think this is going to get fixed. This probably only affects certain flight controllers and those will just not be able to upgrade past 4.3.

If one of the DEVs wants to take a look at this, tell me what you need. I can give you my exact hardware configuration you tell me which build options to select I will flash different versions, set them up exactly the same way, post the dump and an picture of the OSD working or not working. If that would help troubleshoot the issue I can totally do that.

asoell, 2024-07-31:
Ok so I did proper testing now. I can confirm that the last working version of Betaflight is 4.4.2! 4.4.3 does not show an OSD anymore. Neither does 4.5.1.

Flight controller: T-Motor F7 Pro Full Function
Target: TMOTORF7V2
Video system: DJI O3 Air Unit / DJI Goggles V2

I tested this by flashing 4.5.1 (see image of flasher). Then I applied the following configurations:
-DJI O3 preset on UART1 from the presets tab.
-Set USER2 mode to 1400-1600 on AUX1 (just to activate my PIT mode to turn on the O3).
-Add some random elements to the OSD.
_diff all_ and _dump_ are provided as .txt.

Expected behavior would be seeing the OSD visible on my goggles, but it is missing (see image).

Then I reflashed 4.4.2 (see image) and just to make sure it is set up identically, I applied the 4.5.1 _diff_ from earlier in the CLI.
After this the OSD is visible in the goggles as expected (see image).

Then, to test where the firmware breaks, I flashed 4.4.3 and applied the same _diff_.
After this, the OSD was not shown again.

I also manually verified if port settings were correct, etc.

I hope this information will help to fix the issue. Or someone could point out which setting I'm missing or what has changed from 4.4.2 to 4.4.3. I can also make another standalone bug report if necessary. It would be frustrating if some users cannot upgrade past 4.4.2.

[dum_4.4.3.txt](https://github.com/user-attachments/files/16446785/dum_4.4.3.txt)
[dum_4.4.2.txt](https://github.com/user-attachments/files/16446786/dum_4.4.2.txt)
[dum_4.5.1.txt](https://github.com/user-attachments/files/16446784/dum_4.5.1.txt)
[dif_4.5.1.txt](https://github.com/user-attachments/files/16446787/dif_4.5.1.txt)
[dif_4.4.3.txt](https://github.com/user-attachments/files/16446788/dif_4.4.3.txt)
[dif_4.4.2.txt](https://github.com/user-attachments/files/16446789/dif_4.4.2.txt)
[image]
[image]
[image]
[image]
[image]

rrweather, 2024-08-19:
I'm having the same issue with a T-Motor F7 Pro and DJI O3 with 4.5.1

Flight controller: T-Motor F7 Pro Full Function
Target: TMOTORF7V2
Video system: DJI O3 Air Unit / DJI Goggles Integra

Connected to UART 1 via the stock DJI cable. I installed 4.5.0 and the OSD still didn't work. I tried 4.4.2 but couldn't get it to load properly so I have no idea if that would fix the OSD for me.

closed 2024-09-26 by github-actions[bot]