betaflight/betaflight #13241: M10 GPS not working on Betaflight 4.4.3

jasperlort, 2023-12-23:
### Describe the bug

My foxeer M10Q 250 GPS module does not work on my 5 inch quad running betaflight. The weird thing is that I have the exact same module and that one works on this drone. Is it a bug from betaflight or what could it be?

### To Reproduce

Connect M10Q 250 foxeer module to drone. No sats.

### Expected behavior

3D fix.

### Support ID

```txt
###

# # Problem description

# # M10 GPS problem

# ###

# version
# Betaflight / STM32F405 (S405) 4.4.3 Nov 14 2023 / 16:10:10 (738127e7e) MSP API: 1.45

# config: YES
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF405V3

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=59degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3652, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 3356f501a5cadd829ef4b6538066fb1c (4.4.3)
System Uptime: 301 seconds, Current Time: 2023-12-23T11:38:52.906+00:00
CPU:18%, cycle time: 316, GYRO rate: 3164, RX rate: 15, System rate: 9
Voltage: 2275 * 0.01V (6S battery - OK)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.3 Nov 14 2023 / 16:10:10 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF405V3
manufacturer_id SPBE

# resources
resource BEEPER 1 C05
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B05
resource MOTOR 8 B04
[... 1324 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 85
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#

Prajwal-Koirala, 2023-12-26:
The issue reported on GitHub involves a problem with the Foxeer M10Q 250 GPS module not working on a 5-inch quadcopter running Betaflight 4.4.3. The user mentions having the same GPS module working on another drone, raising questions about whether the issue is related to Betaflight or another factor【6†source】.

The flight controller used is a Speedybee 405V3, and the specific Betaflight version is 4.4.3, released on November 14, 2023. The configuration details of the board include a manufacturer ID of SPBE and the board name SPEEDYBEEF405V3. The system's status report shows no I2C errors, which might be relevant since GPS modules typically communicate over I2C or UART【7†source】.

However, there is no additional information provided about how the GPS module is wired to the flight controller or any other relevant context that might help in diagnosing the problem【8†source】.

Given this information, the problem could stem from various factors, including wiring issues, configuration settings in Betaflight, or even a potential fault with the GPS module itself. It's advisable to check the wiring connections, ensure that the Betaflight configuration is correctly set for GPS usage, and perhaps try a different GPS module to isolate the issue.

ot0tot, 2024-01-18:
Use Betaflight 4.5 for M10 GPS

closed 2024-02-18 by haslinghuis