betaflight/betaflight #13246: (NEW TO FPV) 10.9.0 (67943b) is older than your frimware 4.5.0

UOdrones, 2023-12-27:
### Describe the bug

10.9.0 (67943b) is older than your firmware 4.5.0, is there a place to get this download? New to betaflight and cant get things working due to this.

### To Reproduce

after I connect to drone on betaflight

### Expected behavior

expected to be able to update frimware

### Support ID

```txt
2023-12-26 @20:21:52 -- Flight controller info, identifier: BTFL, version: 4.5.0

2023-12-26 @20:21:52 -- Running firmware released on: Dec 27 2023 01:07:02

2023-12-26 @20:21:52 -- Board: LMNR/LUXHDAIO-G4(STM32G47X), version: 0

2023-12-26 @20:21:52 -- Unique device ID: 0x4000284146501420303734

2023-12-26 @20:21:52 -- Craft name:

2023-12-26 @20:21:52 -- Arming Disabled
```

### Flight controller

DJI

### Other components

https://www.getfpv.com/fpv-quad-kits/lumenier-qav-pro-nano-whoop-2-cinequads-edition-rtf-bundle.html

Bought this bundle and was delivered 3 weeks ago. Cant get anything to work.

### How are the different components wired up (including port information)

Everything is connected from GETFPV.com everything talks to each other and is synced I just cant get the correct settings to load.

### Add any other context about the problem that you think might be relevant here

no idea

Prajwal-Koirala, 2023-12-27:
**I'm unable to directly access or incorporate images, but I'll do my best to provide a comprehensive response based on the information you've provided.**

**Here's a breakdown of the issue and potential solutions:**

**Problem:**

- Betaflight Configurator (version 10.9.0) reports that the flight controller's firmware (version 4.5.0) is newer, preventing firmware updates.
- This is likely due to a date discrepancy between the firmware release date (Dec 27, 2023) and the current date (Dec 26, 2023).

**Troubleshooting Steps:**

1. **Check Date and Time:**
   - Ensure your computer's date and time are correct and synchronized with internet time servers.
   - Reconnect the flight controller to see if the issue persists.

2. **Manual Firmware Download:**
   - If the date fix doesn't work, try downloading firmware 4.5.0 manually from the Betaflight website or the manufacturer's website.
   - Load the firmware using the "Load Firmware (Local)" option in Betaflight Configurator.

3. **Betaflight Configurator Version:**
   - Consider using a different version of Betaflight Configurator, as compatibility issues can sometimes occur.
   - Download older or newer versions from the Betaflight website.

4. **Contact Support:**
   - If none of the above solutions work, reach out to Betaflight support or the flight controller manufacturer for further assistance.
   - Provide them with the Support ID details you've shared.

**Additional Tips:**

- **Double-check Wiring:**  Even though components are connected, carefully examine all wiring for any loose connections or incorrect placements.
- **Community Forums:**  Search for similar issues and potential solutions in online forums dedicated to Betaflight or your flight controller.
- **Clear Instructions:**  When seeking help, provide clear and concise information about your setup, including component details, wiring, firmware versions, and error messages.

**I hope this guidance helps you resolve the firmware issue and get your drone flying!**

sugaarK, 2023-12-27:
you need to use configurator 10.10rc1 10.9 doesn't and will never support 4.5

bigspeedfpv, 2023-12-30:
For reference, the link to the configurator version you need is here: https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC1