betaflight/betaflight #13251: SPEEDYBEEF405V3 Resetting when arming when using  v4.5.0 RC1

liu-root, 2023-12-28:
### Describe the bug

Installed 4.5.0RC and the quad would not arm due to MCU reset of the FC. 
Rolled back to v4.4.3 and same configurations, no issues. 
Could not collect blackbox due to reset. 

See configuration on the support id. 

Please let me know how else I can help. 

### To Reproduce

Flash 4.5.0RC1 on SPEEDYBEEF405V3  with the below dump. 

### Expected behavior

FC  Resets when arming

### Support ID

```txt
00ea0b3f-6391-41b3-9554-05ad3b94f008
```

### Flight controller

SPEEDYBEEF405V3 

### Other components

O3, ELRS Serial,

haslinghuis, 2023-12-28:
Please disable feature GPS if no Ublox GPS is used. Fixed in https://github.com/betaflight/betaflight/pull/13244

Prajwal-Koirala, 2023-12-29:
**I'll provide a comprehensive response to the bug report, incorporating insights from previous responses and addressing potential issues:**

**Summary of the Issue:**

- **Betaflight version:** 4.5.0RC1
- **Flight controller:** SPEEDYBEEF405V3
- **Bug:** FC resets when arming, preventing flight.
- **Reproducibility:** Consistent with 4.5.0RC1, not present in 4.4.3.
- **Blackbox logs:** Not available due to resets.

**Troubleshooting Steps:**

1. **Downgrade Betaflight:**
   - Revert to 4.4.3 for immediate resolution and continued use of the quad.

2. **Check for Known Issues:**
   - Consult Betaflight release notes and community forums for reported bugs in 4.5.0RC1 related to the SPEEDYBEEF405V3 or similar FCs.
   - If a known issue is identified, consider alternative RC versions or await a fix.

3. **Review Configuration:**
   - Carefully examine settings in the 4.4.3 configuration dump for potential conflicts with 4.5.0RC1.
   - Pay close attention to:
     - ESC protocol settings
     - Motor idle speed and timing
     - Voltage and current sensor configurations
     - Any experimental features
   - Adjust settings as needed based on 4.5.0RC1 documentation and community recommendations.

4. **Inspect Wiring and Connections:**
   - Ensure all connections between the FC, ESCs, battery, and other components are secure and free of damage or shorts.
   - Pay particular attention to the power supply circuit and any grounding points.

5. **Update Firmware:**
   - If available, update the SPEEDYBEEF405V3 firmware to the latest version compatible with 4.5.0RC1.

6. **Seek Community Support:**
   - Engage with Betaflight forums or Discord channels for further assistance, providing detailed information about your setup, troubleshooting steps, and any relevant error messages.
   - Share the configuration dump and seek input from experienced users familiar with the SPEEDYBEEF405V3 and 4.5.0RC1.

**Additional Considerations:**

- **Hardware Troubleshooting:**
   - If the issue persists after software troubleshooting, consider potential hardware faults, such as a faulty voltage regulator or capacitor on the FC.
- **Experimental Features:**
   - Disable any experimental features in 4.5.0RC1 that might be causing instability.
- **Alternative RC Versions:**
   - If 4.5.0RC1 remains problematic, explore alternative release candidate versions or wait for a more stable release.

closed 2024-01-28 by haslinghuis