betaflight/betaflight #13252: FC got freezes when ARM

CCRCway, 2023-12-28:
### Describe the bug

4.5.0 RC1,When I have correctly set the various settings of the flight control, including sensor direction, motor sequence and direction, Arm channel, angle mode, horizon mode, air mode, etc., and plug in the battery without disconnecting the PC connection, pressing the Arm switch will correctly prompt "MSP" to prohibit unlocking, and the flight control will also sound an alarm.

However, after disconnecting the PC and replug the battery, before arming, all remote controls are normal, and mode switching can be done normally, allowing the buzzer to sound normally. Once the ARM switch is pressed, the flight control only has a red light left, and then there is no response and it is no longer under control. Only by restarting will it be restored. But it can only be maintained before pressing the arm switch.
I have try CCRC AT32F435AIO and JHEMCU F722DUAL FC.

### To Reproduce

Set flight mode normally and arm flight control.

### Expected behavior

Flight control not working, motor not rotating, uncontrolled

### Support ID

```txt
2023-12-28 @11:09:04 -- 操作系统: Windows

2023-12-28 @11:09:05 -- 配置程序: 10.10.0-debug-8d5533b

2023-12-28 @11:09:22 -- 打开串口 成功 ID: 1

2023-12-28 @11:09:22 -- MultiWii API 版本：1.46.0

2023-12-28 @11:09:22 -- 飞控信息，识别名：BTFL，版本：4.5.0

2023-12-28 @11:09:22 -- 当前固件发布时间： Dec 25 2023 20:33:57

2023-12-28 @11:09:22 -- 飞控：CCRC/CCRCF435AIO(AT32F435G), 版本：0

2023-12-28 @11:09:22 -- 唯一设备 ID: 0x428d34848094c000dcc8720

2023-12-28 @11:09:22 -- Build Key: 4f643b1262b24f8bfa496689db155fd5

2023-12-28 @11:09:22 -- 飞行器名称：

2023-12-28 @11:09:22 -- 禁止解锁

2023-12-28 @11:09:25 -- 关闭串口 成功

2023-12-28 @11:09:32 -- 使用已缓存的 https://build.betaflight.com/api/targets 生成信息。

2023-12-28 @11:09:57 -- 打开 USB 设备 成功 ID: 1

2023-12-28 @11:09:57 -- 检测到闪存空间为 1024 KiB 的芯片

2023-12-28 @11:09:57 -- 读取保护未启用

2023-12-28 @11:10:19 -- 擦除1024 kB 闪存 成功

2023-12-28 @11:10:27 -- 关闭 USB 设备 成功

2023-12-28 @11:10:28 -- 查询飞控板信息以辅助选择正确固件

2023-12-28 @11:10:29 -- 配置程序 已成功 检测并验证了飞控板：CCRCF435AIO

2023-12-28 @11:10:34 -- 打开串口 成功 ID: 3

2023-12-28 @11:10:34 -- MultiWii API 版本：1.44.0

2023-12-28 @11:10:34 -- 飞控信息，识别名：BTFL，版本：4.3.2

2023-12-28 @11:10:34 -- 当前固件发布时间： Aug 6 2023 15:53:11

2023-12-28 @11:10:34 -- 飞控：CCRCF435AIO, 版本：0

2023-12-28 @11:10:34 -- 唯一设备 ID: 0x428d34848094c000dcc8720

[... 629 lines cut ...]
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = DEFAULT
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "e4bc7ad3-ef91-4f7d-b955-07c611268bde"

# copy ID and provide to the betaflight team.
```

### Flight controller

betaflight 10.10 RC1,AT32F435AIO/JHEMCU F722DUAL FC

haslinghuis, 2023-12-28:
Please disable feature GPS if no Ublox GPS is used. Fixed in #13244

haslinghuis, 2023-12-28:
Duplicate of #13251

closed 2023-12-28 by haslinghuis

CCRCway, 2023-12-28:
The problem has been resolved, thank you