betaflight/betaflight #13255: AT32F435 ADC current meter

ellmotion, 2023-12-28:
### Describe the bug

ADC current meter doesn't work. it stuck a 5.0 or 5.1.
maybe a bad pin ?
bataflight 4.5 RC1 and 4.4

### To Reproduce

alway

### Expected behavior

When I connect the battery, the current meter stays locked at 5.0A.
When I arm the motors, it also remains at 5A.
[image]

### Support ID

```txt
e1dd6728-a391-41fe-9392-8c8701d0b815
```

### Flight controller

NeutronRCF435MINI (AT32F435G)

Prajwal-Koirala, 2023-12-29:
**I'm unable to directly diagnose or resolve hardware issues, but I can provide information and guidance based on the details you've shared.**

**Here's a summary of the issue:**

- **Bug:** ADC current meter stuck at 5.0A or 5.1A, not reading actual current.
- **Flight controller:** NeutronRCF435MINI (AT32F435G)
- **Betaflight version:** 4.5 RC1 and 4.4
- **Reproducibility:** Always occurs.
- **Expected behavior:** Current meter should display actual current draw, fluctuating with motor use.

**Possible causes:**

- **Hardware fault:**
    - Faulty current sensor
    - Damaged or loose wiring
    - Faulty ADC pin on the flight controller
- **Software configuration:**
    - Incorrect current sensor settings in Betaflight
    - Conflicting firmware settings

**Troubleshooting steps:**

1. **Check wiring:**
    - Carefully inspect all connections between the current sensor, flight controller, and battery.
    - Ensure they are secure and free of corrosion or damage.
    - Verify that the sensor is connected to the correct ADC pin.
2. **Calibrate current sensor:**
    - If supported by your hardware, follow the manufacturer's instructions to calibrate the sensor in Betaflight.
3. **Test with a different current sensor:**
    - If possible, swap the sensor with a known good one to isolate hardware issues.
4. **Review Betaflight settings:**
    - Double-check the current sensor configuration in the Betaflight Configurator.
    - Compare with recommended settings for your hardware.
    - Consider flashing to a different Betaflight version to rule out firmware bugs.
5. **Seek further assistance:**
    - If the issue persists, provide more details about your setup and troubleshooting attempts on a Betaflight support forum or community.
    - Include photos of your wiring and configuration screenshots.

**Additional considerations:**

- **Firmware updates:** Keep Betaflight updated to the latest stable version for potential bug fixes.
- **Hardware modifications:** If you've made any hardware modifications, ensure they haven't affected the current sensor circuit.
- **Community support:** Utilize Betaflight forums and community resources for troubleshooting tips and expertise.

ellmotion, 2023-12-29:
How can I check or read the free pins to be sure if it is not the correct pin declared in the resources?

lida2003, 2024-01-10:
> How can I check or read the free pins to be sure if it is not the correct pin declared in the resources?

It need sch to verify if the pin is correct or NOT. Maybe we need manufacture support.

Sekilsgs2, 2024-04-05:
FC bayac rc dolphin - 
Voltage sensor working good, but current sensor showing wrong result - very high values if batter plugged.
Im test voltages at curr pin and can say that - without battery (power from usb) pin have 0.04 volts, with battery - 0.24 volts. With battery and without connection curr pin to esc - 1.5v.

lida2003, 2024-04-05:
@Sekilsgs2 @ellmotion  
Have you ever tried to clibrate the current sensor?

[BetaFlight/iNav Current Calibration Guide](https://blog.csdn.net/lida2003/article/details/136000660)

Sekilsgs2, 2024-04-06:
> @Sekilsgs2 @ellmotion Have you ever tried to clibrate the current sensor? Or set scale /offset as ESC manufacture provided.
> 
> [BetaFlight/iNav Current Calibration Guide](https://blog.csdn.net/lida2003/article/details/136000660)

Thanks. this help. But for me need using -30000 offset as start point..

SupaflyFPV, 2024-04-06:
> ### Describe the bug
> ADC current meter doesn't work. it stuck a 5.0 or 5.1. maybe a bad pin ? bataflight 4.5 RC1 and 4.4
> 
> ### To Reproduce
> alway
> 
> ### Expected behavior
> When I connect the battery, the current meter stays locked at 5.0A. When I arm the motors, it also remains at 5A. ![image](https://private-user-images.githubusercontent.com/1707394/293265231-978a2505-01b0-4710-8b33-a95e1e8c04cc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0MTk1OTksIm5iZiI6MTcxMjQxOTI5OSwicGF0aCI6Ii8xNzA3Mzk0LzI5MzI2NTIzMS05NzhhMjUwNS0wMWIwLTQ3MTAtOGIzMy1hOTVlMWU4YzA0Y2MucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDZUMTYwMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGRiOGRmZTRjOGUwMzk5MDU3ZDc4OWM3YWNiZmY4ODg4YmVjNTgzNzcwMWEzNWIzOWRiMzcxNmVjZDA4Y2VkMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.dl6ozyTjKPM5vlni0oVBfltI8iAqFEdm7LF7qu8EnTo)
> 
> ### Support ID
> ```
> e1dd6728-a391-41fe-9392-8c8701d0b815
> ```
> 
> ### Flight controller
> NeutronRCF435MINI (AT32F435G)
> 
> ### Other components
> _No response_
> 
> ### How are the different components wired up (including port information)
> _No response_
> 
> ### Add any other context about the problem that you think might be relevant here
> _No response_

can you confirm if you have the correct target?

thecmac, 2024-04-08:
I am unable to reproduce this with 4.5 RC3.

guowei681, 2024-04-09:
> FC bayac rc dolphin - Voltage sensor working good, but current sensor showing wrong result - very high values if batter plugged. Im test voltages at curr pin and can say that - without battery [...]

HI
   Which PCB board are you using? Can you upload a picture to view? In addition, there are three targets for NEUTRONRC F435 in BF. Which one are you using? The commonly used ones are NEUTRONRCF435MINI and NEUTRONRCF435SE. The target of 435G is not used. Based on the previous description, it is impossible to determine your specific situation. We have not reproduced your current phenomenon in the PCB boards corresponding to NEUTRONRCF435MINI and NEUTRONRCF435SE (betaflightght 4.5-RC3)

Sekilsgs2, 2024-04-09:
> HI Which PCB board are you using? Can you upload a picture to view? In addition, there are three targets for NEUTRONRC F435 in BF. Which one are you using? The commonly used ones are [...]

Im say that im using bayac rc dolphin FC! Please read again my post.

lida2003, 2024-04-09:
> Im say that im using bayac rc dolphin FC! Please read again my post.

@Sekilsgs2 @guowei681 I think @ellmotion is using NeutronRCF435MINI (AT32F435G). 

@Sekilsgs2 BTW, which type of ESC are you using, as you have said that offset is quite large(-30000 ).

thecmac, 2024-04-09:
> > Im say that im using bayac rc dolphin FC! Please read again my post.
> 
> @Sekilsgs2 @guowei681 I think @ellmotion is using NeutronRCF435MINI (AT32F435G). But I'm NOT sure if he is using the correct target.
> 
> @Sekilsgs2 BTW, which type of ESC are you using, as you have said that offset is quite large(-30000 ).

The NeutronRCF435MINI is an AIO with an on-board AM32 4-in-1 ESC.
Possibly ESC firmware issues? I am running version 2.8 of AM32, and my voltage and amperage are fine.

thecmac, 2024-04-09:
> bayac rc dolphin

This is a model of plane, multiple people have read your post.  We are asking you to confirm what Flight Controller you are using.
If you have the PNP version, it comes with this Flight Controller:
https://atomrc.com/products/atomrc-fixed-wing-flight-controller-f405-navi?variant=40563895500933
ATOMRC F405 NAVI Fixed Wing Flight Controller
It runs Ardu and INav, not Betaflight.

Are you sure it is a NeutronRCF435MINI ?

Sekilsgs2, 2024-04-09:
> This is a model of plane, multiple people have read your post. We are asking you to confirm what Flight Controller you are using. If you have the PNP version, it comes with this Flight Controller: [...]

Ughhh...

[image]

Now you understand which fc im using?

thecmac, 2024-04-09:
> > > bayac rc dolphin
> > 
> > 
> > This is a model of plane, multiple people have read your post. We are asking you to confirm what Flight Controller you are using. If you have the PNP version, it comes with this Flight Controller: https://atomrc.com/products/atomrc-fixed-wing-flight-controller-f405-navi?variant=40563895500933 ATOMRC F405 NAVI Fixed Wing Flight Controller It runs Ardu and INav, not Betaflight.
> > Are you sure it is a NeutronRCF435MINI ?
> 
> Ughhh...
> 
> ![S2e87eede11154ac7938103d66f77ecd4X jpg_640x640](https://private-user-images.githubusercontent.com/5908468/320882978-5ef8ee9c-2836-4855-a9cc-3bcc27409909.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NzM0NDEsIm5iZiI6MTcxMjY3MzE0MSwicGF0aCI6Ii81OTA4NDY4LzMyMDg4Mjk3OC01ZWY4ZWU5Yy0yODM2LTQ4NTUtYTljYy0zYmNjMjc0MDk5MDkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDlUMTQzMjIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDEyMjg3YmU2M2ExNGMwZDRjNGU5ZjJiYTU3ODg1Njk3MjZhMDA1ZGE0NGJiNDM2ODY4MTAxYWViNjU4MDQ1ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.olk7DSTn0s1_8j5BphZ6vUXoaS1iPDVG1X7BNiwP558)
> 
> Now you understand which fc im using?
> 
> Config for my FC - https://github.com/betaflight/config/tree/master/configs/BAYCK_DOLPHIN

OK, we got there in the end!
Your Flight Controller is *not* a NeutronRCF435MINI.
It is a BAYCKRC DOLPHIN.

Make sure you flash the correct target:
[image]

Sekilsgs2, 2024-04-09:
:-D and this FC - BAYCK_DOLPHIN - hangs, freeze - stop blinking leds -  after 4 sec if using 2s battery :-D  - very good FC)

thecmac, 2024-04-10:
> :-D and this FC - BAYCK_DOLPHIN - hangs, freeze - stop blinking leds - after 4 sec if using 2s battery :-D - very good FC)

This is to be expected, it is a 3-6s FC.  2s will not provide enough power.
[image]

BAYCKRC, 2024-04-15:
> FC bayac rc dolphin - 电压传感器工作正常，但电流传感器显示错误的结果 - 如果电池插入，则值非常高。我在电流引脚上测试电压，可以说 - 没有电池（来自 USB 的电源）引脚有 0.04 伏，有电池 - 0.24 伏。带电池，不带连接电流引脚至 ESC - 1.5V。

I would like to know which electronic speed controller you are using?
[image]
I connected the 3.3V voltage on the motherboard to the CURR pad and obtained the values shown in the figure from the ground station. I think he is working correctly because 3300 ÷ 40=82.5A. It is very close to the values in the figure.

Sekilsgs2, 2024-04-15:
> I connected the 3.3V voltage on the motherboard to the CURR pad and obtained the values shown in the figure from the ground station. I think he is working correctly because 3300 ÷ 40=82.5A. It is very close to the values in the figure.

I wrote about something else - I wrote that when there is no load, the current sensor shows about 30A - I also wrote that the current sensor works - but with an offset of 30A. and you test and write when connecting 3.3V - this is not a correct comparison - provide a minimum table - what the sensor produces at different voltages starting from 0V.

and the words “I think” scare me - the sensor either works or “I think it doesn’t work”

SteveCEvans, 2024-05-12:
Testing with a `NEUTRONRCF435` and feeding a bench supply into the `Bat` and `Cur` input over the range 0 to 12V and 0 to 3V3 respectively and using the following default settings:

[image]

Varying the `Cur` input with no input voltage had no effect on the current reading. This is normally correct if no battery voltage is detected, but I took out this check thus:

```
diff --git a/src/main/sensors/battery.c b/src/main/sensors/battery.c
index ef8d029..9a6ffec 100644
--- a/src/main/sensors/battery.c
+++ b/src/main/sensors/battery.c
@@ -439,11 +439,12 @@ void batteryInit(void)
 
 void batteryUpdateCurrentMeter(timeUs_t currentTimeUs)
 {
+#if 0
     if (batteryCellCount == 0) {
         currentMeterReset(&currentMeter);
         return;
     }
-
+#endif
     static uint32_t ibatLastServiced = 0;
     const int32_t lastUpdateAt = cmp32(currentTimeUs, ibatLastServiced);
     ibatLastServiced = currentTimeUs;
```
Moving to the `Bat` input, I'm seeing the battery voltage reported ~87% high consistently, so 1.6V gets reported as 3V. A 4V input sees 7.6V and a 12V input sees 22.75V.

Meanwhile with 0V input on the `Bat` line (not the `Cur` input!), zero current is reported, but 1.6V results a current of 6.4A, 4V results in reading of 12.4A, and 12V results in a reading of 29.3A, all whilst there is no voltage input to the `Cur` input.

Further investigation required.

SteveCEvans, 2024-05-13:
Having enabled `DEBUG_ADC_CHANNELS ` I got the following when I briefly applied 1V to the `Bat` input.

[image]

And then to the `Cur` input.

[image]

So this at least shows that the inputs are being sampled, but it also highlights how, with nothing connected to the `Cur` input it is impacted by the level on the `Bat` input.

SteveCEvans, 2024-05-13:
2V on the `Cur` input gives.

[image]

3V gives.

[image]

Note how this also impacts the reading on the `Bat` ADC channel too.

ledvinap, 2024-05-13:
Unconnected ADC input can behave this way - sampling capacitor may not be discharged between measurements, Try grounding the other input .

SteveCEvans, 2024-05-13:
This is most odd. Now I'm seeing the reported current change. Not sure if I could have had a bad connection before, but I don't think so...

[image]

Anyway, applying a voltage to the current input I'm seeing.

| Input Voltage | Reported Current |
| - | - |
| 0.0 | 0 |
|0.5| 12.2 |
|1.0| 24.7 |
|1.5| 37.0 |
|2.0| 49.4 |
|2.5| 61.8 |
|3.0| 74.3 |

Note that the above voltage readings are as indicated on my bench power supply, but as a quick check with my DVM showed 3.0V to be 2.995V I think it's close enough.

SteveCEvans, 2024-05-13:
> Unconnected ADC input can behave this way - sampling capacitor may not be discharged between measurements, Try grounding the other input .

Agreed, they're probably very high impedance.

SteveCEvans, 2024-05-13:
So this doesn't appear to be a generic AT32F435 issue as the `NEUTRONRCF435` is behaving OK.

lida2003, 2024-05-14:
> 3V gives. [...]

- `voltageMeter.unfiltered`  downward trend of voltage
- `voltageMeter.displayFiltered` upward trend

1. ADC BAT pin, suspended?
2. why a downward trend will be filtered as upward trend?

https://github.com/betaflight/betaflight/blob/2d3c8eea334752ca085cdaf3c7bb52996c3ef28c/src/main/sensors/battery.c#L162-L165

ASDosjani, 2024-06-25:
Hi, I also expereienced that current sensor not working on `NEUTRONRCF435SE`. No matter if the motors are spinning or not. I checked, it has DMA assigned.

SteveCEvans, 2024-10-05:
I suggest trying adding a 100nF cap to the current input. Most FC's have a cap on the voltage ADC input (after a potential divider), but not on the current input.

closed 2026-09-26 by blckmn