betaflight/betaflight #13262: dps310 not getting detected on I2C2 SCL=B10 SDA=B11

iottrends, 2024-01-02:
### Describe the bug

5df36847-dddc-4a33-8c7c-fb3eb3acda95

### To Reproduce

Connect the flight controller to Betaflight configurator (10.10)
Load the firmware version BTFL4.4.4.4.
Load the config file shared in above step.
[image]
in top right side Gryo and flash detected but no Baro.

### Expected behavior

I expect Baro(dps310 )to be detected.
status command gives I2C error
Entering CLI Mode, type 'exit' to return, or 'help'

# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.28V, Core temp=46degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MSP (30 x 13)
System Uptime: 181 seconds, Current Time: 2024-01-02T08:39:53.364+00:00
CPU:2%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP

### Support ID

```txt
###

# # Problem description

# # DPs310 not getting detected in FC, this my custom FC. dsp310 present on i2c2_scl=B10 i2c2_sda=B11

# ###

# version
# Betaflight / STM32F405 (S405) 4.4.4 Jan  1 2024 / 15:04:07 () MSP API: 1.45

# config: YES
# board: manufacturer_id: DIYF, board_name: MYDIYFC405

# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.28V, Core temp=46degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MSP (30 x 13)
System Uptime: 272 seconds, Current Time: 2024-01-02T08:23:48.104+00:00
CPU:2%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
[... 320 lines cut ...]
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

stm32f405 Self Built

### Other components

This is self built FC405. this was designed in kicad and pcb + pcbA done at jlcpcb.
1. Flight controller is able to boot and load the firmware.
2. it has 16MB, BMI270, dps310 and at7456e.
3. BMI270, flash and AT7456E are detected.
4. However Baro is not getting detected. I am using BTFL4.4.4 Nover-2023 build. Compiled the firmware on ubuntu platform (make STM32F405) I an see bmi270 and dps310 are getting compiled as part of make.
5. After loading the local firmware I am loading the config file.
[f405.onfig.txt](https://github.com/betaflight/betaflight/files/13808248/f405.onfig.txt)

### How are the different components wired up (including port information)

[image]

aowi7280, 2024-01-03:
One of the devs may be better to answer, but I know I2C bus requires pullup resistors on the SCL/SDA lines. Do you have those?

iottrends, 2024-01-03:
Yes pullups are there...
[image]

iottrends, 2024-01-03:
on I2C2 channel dps310 is there and on i2c1 channel pads are exposed for compass

ot0tot, 2024-01-18:
If the baro is on I2C2, you need:
```set baro_i2c_device = 2
save```

Are you also including the drivers for the DPS310 when building firmware?

haslinghuis, 2024-01-23:
Also see lot of configuration having internal and external I2C shared sensors need external battery to function.

closed 2024-02-29 by github-actions[bot]