betaflight/betaflight #13268: Bf 4.5 osd icon problem

Themightybert95, 2024-01-04:
### Describe the bug

I’ve returned to betaflight after some years with INAV and i’ve flashed BF v4.5 and everyrhing work but in the OSD of my goggles i can’t view the number of satellite icon ( the gps is working and i can see all the other icon like coordinates , homepoint distance, home arrow etc)! Of course i’ve activated it in the OSD menu of my goggles but it’s Not visibile .

### To Reproduce

The problem is persistente even if i plug and unplug the battery

### Expected behavior

I just need to solve this Little problem because i’m afraid to fly long range without knowing the number of satellite that I have 

### Support ID

```txt
34961ac7-98a7-4244-9e9a-1658c5693376
```

### Flight controller

Kakute H7 v2

### Other components

Rx: Tbs diversity RX 
Vtx: Dji air unit 
GPS :matek m8q-5883

### How are the different components wired up (including port information)

Rx to uart 6
Vtx to uart 2
GPS to : uart 4

haslinghuis, 2024-01-04:
See https://github.com/betaflight/betaflight/pull/13005#issuecomment-1676119834

chusky-fpv, 2024-01-12:
Hi, I also have the same problem since I installed 4.5. Have you tried the indicated solution? Are there other tests that can be done?

chusky-fpv, 2024-01-12:
I managed to solve part of the problem, WTFOS can not manage the fonts well and the too many elements inside the osd you have to remove the special fonts and remove some elements to make the gps satellites reappear. Nella vecchia versione 4.4 funzionava meglio questo aspetto.

closed 2024-02-20 by github-actions[bot]

kenmccann, 2024-07-22:
Solution for me was the temporary workaround in this comment: https://github.com/betaflight/betaflight/pull/13005#issuecomment-1676119834

`set ‎displayport_msp_fonts = 0,0,0,0`