betaflight/betaflight #13272: dshot bidirectionnelle don't work if you enable this in the configuration menu betaflight 4.5

Bolbee, 2024-01-05:
### Describe the bug

hi i got this error 
[image]
it basicely say quad x need 4 motor but he detect 0, i was in bluejay 0.19 and i also try 0.19.2
the issue happen when i try to enable dshot bibidirectionnelle, the issue happen if you enable the led strip in the configuration part 
[image]
but if you don't enable it the issue will not happen. 
[image]

### To Reproduce

enable led strip in configuration part
[image]

### Expected behavior

if you activate the led strip the dshot bidirectionnelle won't work and in escn configuration website, the website will not detect the esc

### Support ID

```txt
basic one, idk what is that
```

### Flight controller

f405 v3

haslinghuis, 2024-01-05:
Try `set dshot_bitbang = ON`

Hasselfpv, 2024-01-23:
> Try `set dshot_bitbang = ON`
> 
> Second try to disable `SERVOS`

Nice , I have sam issues but now wit set dshot_bitbang = ON it still works .
FC Iflight Blitz Mini ATF435 with bluejay 0.19.2

closed 2024-03-01 by github-actions[bot]