betaflight/betaflight #13275: SOFTSERIAL not appearing in configurator 10.10.0-RC1 utilizing BF Firmware 4.5.0-RC1 on Flywoo F411

dmyers7, 2024-01-06:
### Describe the bug

Software Serial is available on the Flywoo F411 utilizing FW 4.4.3 and Configurator 10.10.0-RC1.  Upgraded twice (to insure I was not doing something wrong) with a complete wipe to 4.5.0-RC1 and the SoftSerial port 1 is no longer available to configure in Configurator 10.10.0-RC1.

### To Reproduce

Install FW 4.5.0-RC1 on a Flywoo F411 board utilizing the required 10.10.0-RC1 to talk with the board.

### Expected behavior

SoftSerial Port 11 should still be available to configure for accessories like GPS.

### Support ID

```txt
Id: "59db92a4-7072-4cf2-8d86-f9d972858810"
```

### Flight controller

Flywoo F411

### Other components

LEDs, Flywoo GPS, WalkSnail Avatar VTX, CrossFire RX

### How are the different components wired up (including port information)

resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09 (CRSF)
resource SERIAL_TX 2 A02 (GPS)
resource SERIAL_TX 11 B06  (VTX)
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 B07
resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15

haslinghuis, 2024-01-07:
Use `resource SOFTSERIAL_RX 1 B06` instead.

haslinghuis, 2024-01-07:
Use `resource SOFTSERIAL_TX 1 B06` instead.

closed 2024-01-07 by haslinghuis

ctzsnooze, 2024-01-15:
The resource command `SOFTSERIAL_TX` is not used in current Configurator CLI.
Use `resource SerialTx 11 <pin>` to set the Softserial 1 Tx pin to an available MCU pin.

haslinghuis, 2024-01-15:
[image]

cpwilson, 2024-04-29:
Having the same issue as @dmyers7 

Using Configurator 10.10.0 on a Flywoo F411 board there is no softserial for VTX so VTX tab in configurator is blank.

I can  enable softserial in the Configuration tab in Configurator but the port does not then appear in Ports tab

My original BF 4.4.3 resources are as below:
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 11 B06
resource SERIAL_RX 1 A10
resource SERIAL_RX 11 B07
resource ADC_BATT 1 B01
resource GYRO_EXTI 1 B05

Adding these in CLI gives the error below for the softserial:
# resource SERIAL_TX 11 B06
###ERROR IN resource: INDEX NOT BETWEEN 1 AND 10###
# resource SERIAL_RX 11 B07
###ERROR IN resource: INDEX NOT BETWEEN 1 AND 10###

If I then use CLI and add them in as SOFTSERIAL as @haslinghuis suggests as below:
resource SOFTSERIAL_TX 1 B06
resource SOFTSERIAL_RX 1 B07

I get the softserial in the Ports tab but after setting it as VTX (IRC Tramp) I still don't get the VTX tab available.

So for me Configurator 10.10.0 is definately not working with the FLYWOO411

haslinghuis, 2024-04-29:
You could try the UNSUPPORTED define in https://github.com/betaflight/betaflight/pull/13540

cpwilson, 2024-04-29:
ok, firmware built with custom define 'OVERRIDE_SOFTSERIAL_BAUDRATE'

Enabled softserial and still not showing up in ports tab

Tried to add in CLI but get the errors below:
# resource SOFTSERIAL_TX 1 B06
NOTE: B06 already assigned to SERIAL_TX 1.
Resource is set to B06
# resource SOFTSERIAL_RX 1 B07
NOTE: B07 already assigned to SERIAL_RX 1.
Resource is set to B07

Any suggestions?

haslinghuis, 2024-04-29:
> If I then use CLI and add them in as SOFTSERIAL as @haslinghuis suggests as below:
resource SOFTSERIAL_TX 1 B06
resource SOFTSERIAL_RX 1 B07

> I get the softserial in the Ports tab but after setting it as VTX (IRC Tramp) I still don't get the VTX tab available.

Enable feature SOFTSERIAL in configuration tab?

cpwilson, 2024-04-29:
Enable feature SOFTSERIAL in configuration tab?
VTX needs to be included in the build

SOFTSERIAL is enabled and VTX was in the build

this is the resources I now have

resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SOFTSERIAL_TX 1 B06
resource SOFTSERIAL_RX 1 B07

haslinghuis, 2024-04-29:
You need to release the resource

```bash
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 A03
resource SOFTSERIAL_TX 1 B06
resource SOFTSERIAL_RX 1 B07
feature SOFTSERIAL
save
```

cpwilson, 2024-04-29:
The FLYWOO411 only has 2 uarts, hence the softserial is needed for VTX control

[image]

haslinghuis, 2024-04-29:
Yes you are not able to share UART with SOFTSERIAL. If using UART 1 for SOFTSERIAL - softserial will be used instead.

cpwilson, 2024-04-29:
Unfortunately the Flywoo LR4 with the FLYWOOF411 with ELRS and GPS needs the softserial

[image]

haslinghuis, 2024-04-29:
`B04` (motor 5)  or `A15` (led_strip) are better candidates for using SOFTSERIAL_TX 1 instead.

cpwilson, 2024-04-29:
But how would that work for others who may struggle to reassign uarts in cli?

cpwilson, 2024-04-30:
@haslinghuis just wondering.... will this issue be re-opened as I see it was closed as looking like it's an issue for all boards needing softserial?

davidpkgray, 2024-06-08:
For my stock Flywoo LR using Betaflight 4.5
resource SOFTSERIAL_TX 1 A15 
Worked fine to get my VTX working with soft serial.

Denethor85, 2024-10-20:
I also have soft serial errors on 2 totally different FCs but both F411. One is the FC sto k of the BabyApe and the other is the FC sto k of the TwigXL 2023