betaflight/betaflight #13285: BLHeli32 Passthrough broken in 4.5 RC 2

ojwright845, 2024-01-10:
### Describe the bug

BLHeliSuite32.10 fails to read ESC at BF FW 4.5 RC 2.
Error: Found no valid ESC configurations. 

### To Reproduce

Flash FW 4.5 RC 2. Attempt read of ESCs in BLHeliSuite32.

### Expected behavior

Flashing back to FW 4.4.3, and earlier, the problem does not exist 

### Support ID

```txt
e473b82f-fd8c-49b3-9bdc-e46f0e730045
```

### Flight controller

JHEF745

### Other components

None

### How are the different components wired up (including port information)

N/A

### Add any other context about the problem that you think might be relevant here

Tested on the following FC/ESC: TMotor mini F7 / TMotor F45, Jhemcu F745AIO / 40A, Holybro Kakute H7 / Kakute 65A, AxisFlying istack

haslinghuis, 2024-01-10:
Did it work in 4.5RC1?

Rockj71, 2024-01-10:
same problem here

haslinghuis, 2024-01-10:
There is no need for further reporting on this as we established there is an issue and are working on a resolve. Please hold tight :)

SteveCEvans, 2024-01-10:
Please test https://github.com/betaflight/betaflight/pull/13287

ojwright845, 2024-01-11:
Working now

closed 2024-01-11 by ojwright845

FuknFlipper, 2024-02-21:
I'm building my first quad, I have everything setup and appears to be working but I need to change blheli32 settings for the tune I have loaded and am getting the same no esc configuration found notice. It recognizes 4 esc and has four boxes at the bottom but they don't have numbers in them and if you try to change anything it comes back with the same error when flashing. All esc info boxes have ??? In them. BF 4.5RC2. TMotor 4 in 1, f55a f7 stack. I have a feeling it's this timeout loop being talked about as it's doing the exact same thing but I have no clue how to edit it.

haslinghuis, 2024-02-21:
Fixed in latest firmware development version. Join our discord for support questions: https://discord.com/invite/gV4XSq3fus

FuknFlipper, 2024-02-22:
Ok I'll join. I just went backwards to 4.4.3 and it works fine for now.