betaflight/betaflight #13288: Latency issue when moving stick / rigs reacting with delay - PINIO feature from Cloud build

Canaill51, 2024-01-12:
### Describe the bug

FC Speedybee F405 V3, flashed on BF 4.4 & 4.5 (removed PINIO feature from Cloud build)
Caddx Vista (firmware v01.01.0000) and Goggles Integra (firmware v01.04.0000)

When this latency issue occurs, I move sticks to control the rig but I see it reacting with a delay (this with PINIO feature removed from Cloud Build).
When PINIO enabled from Cloud build, I see the rig reacting as soon as I move the sticks. 

### To Reproduce

Flash Speedybee F405 V3 on Cloud build, removing PINIO feature on BF 4.4 and 4.5
VTX-MSP enabled on UART1, using Preset HD VTX .

Rig should have Caddx Vista (firmware v01.01.0000) and Goggle Integra (firmware v01.04.0000)
Random latency issue occuring when arming the rig.

### Expected behavior

I should not feel this latency, this should not occur.
This PINIO feature should not be possible to be removed from Cloud Build, because "pre-defined"
PINIO feature must be kept enabled.

### Support ID

```txt
# resource
[...]
resource PINIO 1 C03
[...]

# get pinio_config
pinio_config = 129,1,1,1
Array length: 4

# get pinio_box
pinio_box = 0,255,255,255
Array length: 4
```

### Flight controller

Speedybee F405 V3

ot0tot, 2024-01-13:
I believe the PINIO setting on the Speedybee boards shuts off power to the BT/WiFi chip (connected via MSP) when armed. 

Trying to use a digital (MSP DisplayPort) VTX with the BT/WiFi chip active may prevent the VTX from receiving the arm signal due to noise/MSP messages from the BT chip, which would cause it to remain in low power mode and operate at low frame rate. That sounds like what the OP is describing.

Canaill51, 2024-01-13:
> I believe the PINIO setting on the Speedybee boards shuts off power to the BT/WiFi chip (connected via MSP) when armed.
>
> Trying to use a digital (MSP DisplayPort) VTX with the BT/WiFi chip active [...]

Would make sense! 
I did check another FC with bluetooth:
Example Mamba MK4 H743 V2 - Diatone (Target https://github.com/betaflight/config/blob/master/configs/MAMBAF722_2022B/config.h). It has same kind of config concerning PINIO.

You're right, BT/WIFI should be turned off when armed --> User must not be able to remove PINIO from Cloud Build Option.
This would avoid any other issue relating to PINIO's impacts.

@haslinghuis ?

Canaill51, 2024-01-19:
Just a quick update:
Regarding to weather condition here, I was not able to test and fly my rig. Cross fingers on tomorrow (should be good... but still very cold).

@haslinghuis ,
Do you have any thoughts about my suggestion above: 
--> User must not be able to remove PINIO from Cloud Build Option.
This would avoid any other issue relating to PINIO's impacts.

Anyway, will keep you informed about my results.

haslinghuis, 2024-01-20:
> User must not be able to remove PINIO from Cloud Build Option.
> This would avoid any other issue relating to PINIO's impacts.

Yeah makes sense. Include in config or common_post if PINIO pin is defined.

Canaill51, 2024-01-21:
Update:
I retried on different rigs yesterday and today. This random issue still occurs.
Occuring on Speedybee F405V3, AIKONF7 v1.1 (F722) and NEUTRONRCF435MINI.
All run on latest 4.5 RC. All correctly configured. All flashed with all features enabled.

I am ouf of idea... Two months testing, investigating, comparing... 

Finally, I am thinking this is a compatibility concern with DJI Integra goggles with Caddx Vista units.

I sent them to DJI --> They confirmed nothing were wrong... because they only tested with O3 (they confirmed they did not have any Caddx Vista units... LOL).

I will sell Integra and get some DJI V1 Goggles back. 

Let me know if you need more details.

Canaill51, 2024-01-22:
@haslinghuis today, I tried something:
I flashed my rig to 4.4.3, all settings defined stock. Reminder; my above issues happened on 4.5.
On 7 packs, no latency issue!
Yesterday, this latency issue did occur again on three different FC, all flashed on BF 4.5 RC2...

Have something related to MSP (Displayport, or else relating to VTX HD, etc...) been changed on 4.5 firmware?

haslinghuis, 2024-01-22:
We have a huge list of changes in https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC1

Can you try both 4.5RC1 and latest zulu to compare behavior?

For 4.5RC2 / zulu please provide support ID (generate using button in CLI tab).

Canaill51, 2024-01-22:
> We have a huge list of changes in https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC1
> 
> Can you try both 4.5RC1 and latest zulu to compare behavior?
> 
> For 4.5RC2 / zulu please provide support ID (generate using button in CLI tab).

Ok, I am going to test 4.5 RC2, test... then 4.5-Zulu and test also. Will report all information you need.

Will try this on tomorrow, cross finger weather will be not rainy.

PR #13315 (2024-01-24): Enable PINIO if PIN defined in config
Fixes #13288
closed 2024-01-24 by haslinghuis