betaflight/betaflight #13290: Delay on disarm

haslinghuis, 2024-01-12:
### Describe the bug

When not using acro mode disarming seems to have a delay. 

### To Reproduce

https://github.com/betaflight/betaflight/assets/8344830/e08ad4dd-74ca-4407-a83d-08f2af5ba4b7

### Expected behavior

Disable arming on switch immediately 

### Support ID

```txt
From discord issue. So now info here.
```

### Flight controller

N/A

### Other components

N/A

### How are the different components wired up (including port information)

N/A

### Add any other context about the problem that you think might be relevant here

Needs investigation

haslinghuis, 2024-01-12:
Tried Acro, Angle and Horizon on three position switch using RM TX16S with OpenEdge 2.8.0 installed and TBS Micro TX.

Have not been able to reproduce on Foxeer F745 using BF 4.5 master with BlueJay v19.x ESC firmware and TMotor Tracer P1604.

Other components used: TBS CRSF nano RX Pro, Walksnail HD Kit V1 and WS M181GPS/Compass.

haslinghuis, 2024-01-12:
From the [dump / diff](https://github.com/betaflight/betaflight/files/13923948/Hydro_on_BF4.5_broken.txt)

got

[image]

haslinghuis, 2024-01-12:
https://build.betaflight.com/api/support/ef0875d0-2bef-4fc3-bffc-83f9b67372bc
https://build.betaflight.com/api/builds/30f5066ab97b12c52755f63763047320/log

```
make MAMBAF722_I2C EXTRA_FLAGS="-D'BUILD_KEY=30f5066ab97b12c52755f63763047320' -D'RELEASE_NAME=4.5.0-RC2' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
```

0crap, 2024-01-12:
Helped the Discord user create a support ID.
[image]

GSfpv, 2024-01-12:
Switching from 8k Pidloop to 4k Pidloop seems to solve the issue (independently if selecting DSHOT 300 or 600). 
Disarming works as it should. Will run a proper flight test soon.

ctzsnooze, 2024-01-13:
@GSfpv - could you please share the typical output of the 'tasks' CLI command, and note the CPU usage value, when it isn't working, vs when it is?
Also share the output of the 'version' CLI command?

SteveCEvans, 2024-01-13:
Identify the features/attached devices and through a process of elimination determine which one is causing the issue. This could be GPS rescue, LED STRIP, enabling a magnetometer etc. You need to figure out which it is.

GSfpv, 2024-01-13:
Quad parts: Mamba stack F722 I2C 20x20 with a 40a Mamba ESC. GPS BN220 Beitan. Caddx Vista. The FPV Motors. CRSF Nano. Flyhoo Nano Blethooth V3 module on a softserial. Additional mod is Betaflight OSD on Caddx Vista to DJI Goggle G2 described in the video below:
https://www.youtube.com/watch?v=KIis4hB_duQ&list=PLnlllPQhgqG2qne7l9VZfLIy03kwuhuO4&index=6

The following tests done on 8k Pidloop DSHOT 600 (seems that 4k Pidloop solves the issue, although haven't had time to do a test flight yet.

Task and Version: 
[Task and Version.txt](https://github.com/betaflight/betaflight/files/13929757/Task.and.Version.txt)
[image]

CPU load is about 80% when at rest, and 83% when powering the motors (via BF). Although the motor do not respond properly in the Motors tab when operating the sliders. See video:
https://github.com/betaflight/betaflight/assets/156380285/65e8d8a3-b120-4819-b699-bbed99f1cb10

Dump all:
[Dump all - Diff all.txt](https://github.com/betaflight/betaflight/files/13929771/Dump.all.-.Diff.all.txt)

GSfpv, 2024-01-13:
> Identify the features/attached devices and through a process of elimination determine which one is causing the issue. This could be GPS rescue, LED STRIP, enabling a magnetometer etc. You need to figure out which it is.

I deactivated GPS/OSD/Softserial and powered the motors both via the Motor tab and regular arming from Radio. Still on 8k pidloop and DSHOT600. Although it seems that the arming via the Radio currently works (bench testing), it still goes mad if powered via the Motor tab on Betaflight (see video in previous comment).

haslinghuis, 2024-01-13:
If your quad has logging capabilities please enable debug `DEBUG_TIMING_ACCURACY`.

Setup the quad as follow:

```
set pid_process_denom = 1
set motor_pwm_protocol = DSHOT600
set dshot_bidir = ON
set debug_mode = TIMING_ACCURACY
save
```

After testing you should be able to get a blackbox file from blackbox tab and send it here.

GSfpv, 2024-01-14:
Arming Blips LOG with Pidloop 8k and DSHOT 600.
[Blackbox with Timing Accuracy logging (arming blips with Radio).zip](https://github.com/betaflight/betaflight/files/13929906/Blackbox.with.Timing.Accuracy.logging.arming.blips.with.Radio.zip)

ctzsnooze, 2024-01-14:
Can you power down the bluetooth and never that softserial and see what happens?

GSfpv, 2024-01-14:
I did the test by gradually adding the Flyhoo BT features (desoldered the BT module).

4 Logs based on the following:

- BT Module
- Softserial Feature
- Softserial Port
- Softserial Resource Pin Switch

I took everything off and gradually added them like you see in the log titles, switching both Acro and Angle mode.

[Logs.zip](https://github.com/betaflight/betaflight/files/13930899/Logs.zip)

The issue appeared once I switched the Softserial in the Ports Tab on (without having the BT soldered on). Now this was all done arming via RADIO.

I attempted the test with the Betaflight Motor tab sliders, as there the motors do not follow the slider commands. With everything off (BT module, Soft Feature/Port/Resorce pin), it is still giving issues there, although looks fine if arming with the Radio. Therefore there is inconsistency between the test with Radio and BF Motor Sliders, on the same exact settings.

https://github.com/betaflight/betaflight/assets/156380285/c156ff49-597a-4be4-8e99-c8b9c8840c82

haslinghuis, 2024-01-14:
What is connected to the softserial port?

GSfpv, 2024-01-14:
> What is connected to the softserial port?

In the last test, nothing, I de-soldered the FLYWOO BT-NANO V3 Bluetooth Module

haslinghuis, 2024-01-14:
Thanks. It is not recommended to have BT enabled in flight. Using softserial port adds up to mcu computations. But still it should not delay arming procedure.

GSfpv, 2024-01-15:
> Thanks. It is not recommended to have BT enabled in flight. Using softserial port adds up to mcu computations. But still it should not delay arming procedure.

The Flyhoo BT automatically shuts down when arming is activated, therefore it should not be an issue. That said, the disarming issue persists even with BT module completely detached.

GSfpv, 2024-01-16:
Tested new Developer code #13303 and it seems to do the job. It won't allow the Softserial switch the MSP slider. Therefore motors disarm as they should.

closed 2024-01-18 by KarateBrot

doppleganger53, 2025-11-23:
I'm seeing this behavior on 25.12-RC2. Softserial not enabled, pidloop is 8k. FC is a tmotor velox cine F722, status and tasks below.

What steps should I take / info to gather to assist in anlysis?

# status
MCU: STM32F722 CLK=216MHz, Vref=3.29V, Core temp=57degC
STACK: 2048b (0x20010000)
CONFIG: CONFIGURED (3890b / 16384b)

DEVICES DETECTED: SPI=1, I2C=1 (0 errors)
GYRO: (1) ICM42688P enabled locked dma
ACC: ICM42688P
BARO: DPS310
GPS: connected, UART6 57600 (set to 57600), configured, version =  M10
OSD: MSP (53 x 20)
FLASH: JEDEC ID=0x00ef4018 16M

BUILD KEY: 62b59b8d5ac0149b37e8981280ab1bd7 (2025.12.0-RC2)

System Uptime: 232 seconds, Current Time: 2025-11-23T00:20:07.681+00:00
CPU:59%, cycle time: 124, GYRO rate: 8064, RX rate: 0, System rate: 9
Voltage: 2.09V (0S battery - NOT PRESENT)
Arming disable flags: RXLOSS CLI MSP POS_HOLD_SW

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       9       0    0.0%    0.0%         2      1   2519       0
01 - (         SYSTEM)    969      13       0    1.2%    0.0%       153     16 242432       4
02 - (           GYRO)   8091      15       1   12.1%    1.3%      3756      0 2035689       0
03 - (         FILTER)   8091      39      17   31.5%   14.0%     36091      0 2035689       0
04 - (            PID)   8091      89      53   72.0%   42.8%    111786      0 2035689       0
05 - (            ACC)    969      19       5    1.8%    0.5%      1272     21 240332       9
06 - (       ATTITUDE)    481      27      10    1.2%    0.5%      1278     42 120255      14
07 - (             RX)     20      32      15    0.0%    0.0%       252     55  15159      16
08 - (         SERIAL)    245  336039       4 8232.9%    0.1%      1023      0  55368       6
09 - (       DISPATCH)    964      10       0    0.9%    0.0%       166     36 242564       3
10 - (BATTERY_VOLTAGE)     50      12       1    0.0%    0.0%        27     10  12568       4
11 - (BATTERY_CURRENT)     50      16       0    0.0%    0.0%        17      4  12568       1
12 - ( BATTERY_ALERTS)      5       9       2    0.0%    0.0%         3      0   1261       6
13 - (         BEEPER)    100       9       2    0.0%    0.0%        51      3  24659       8
14 - (            GPS)    100      66       3    0.6%    0.0%       171   1937  19978      18
15 - (     GPS_RESCUE)    100      16       4    0.1%    0.0%       122      9  25108       9
16 - (        POSHOLD)    100      12       0    0.1%    0.0%        18     18  24638       0
18 - (           BARO)     10     375      10    0.3%    0.0%       191   2556  15397       8
19 - (       ALTITUDE)    100      17       2    0.1%    0.0%        64     14  24573       5
21 - (      TELEMETRY)    244      18       0    0.4%    0.0%        59     66  55367       1
22 - (            OSD)     12      25       6    0.0%    0.0%       169     39  27131      13
24 - (            CMS)     20      17       2    0.0%    0.0%        10      2   5006       2
25 - (        VTXCTRL)      5      21       0    0.0%    0.0%        14      1   1258      15
27 - (    ADCINTERNAL)      1      11       3    0.0%    0.0%         1      0    253       9
RX Check Function                  18       3                       117
Total (excluding SERIAL)                                59.1%