betaflight/betaflight #13291: Bf 4.5.RC2  SBUS stopped working.

E120103016, 2024-01-12:
### Describe the bug

Fc: OMNIBUSF4SD
Rx: radiomaster R161
Radio: Tx16s 

The radio will connect to the quad  but Bf4.5.0 RC2 doesn't see my stick inputs. Downgrading the firmware back to 4.4 solves the issue

### To Reproduce

Try upgrading OMNIBUSF4SD  to BF4.5.0RC2 with a radiomaster R161 on uart6

### Expected behavior

It should read the incoming stick and button inputs like normal. But it doesn't

### Support ID

```txt
feature RX_SERIAL
set serialrx_provider = SBUS
set serialrx_inverted = OFF
set serialrx_halfduplex = OFF
save
```

### Flight controller

OMNIBUSF4SD/BF4.5.0RC2

haslinghuis, 2024-01-12:
Please provide support ID.

E120103016, 2024-01-13:
board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: V250

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 C09
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 C08
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource LED 1 B05
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
[... 824 lines cut ...]
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set serialmsp_halfduplex = OFF
set timezone_offset_minutes = 0
set rpm_filter_harmonics = 3
set rpm_filter_q = 500
set rpm_filter_min_hz = 100
set rpm_filter_fade_range_hz = 50
set rpm_filter_lpf_hz = 150
set stats_min_armed_time_s = -1
set stats_total_flights = 0
set stats_total_time_s = 0
set stats_total_dist_m = 0
set craft_name = V250
set pilot_name = David Shu
set altitude_source = DEFAULT
set altitude_prefer_baro = 100
set altitude_lpf = 300
set altitude_d_lpf = 100
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

patemil, 2024-01-26:
Also getting issue with SBUS on 4.5.0 coming from 4.2.11.
No channels are working

Downgrading to 4.2.11 and it works again

# 

# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4FW, version: b0bd8e54, date: 2019-10-20T13:05:07Z
# board: manufacturer_id: AIRB, board_name: OMNIBUSF4FW

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4FW, version: b0bd8e54, date: 2019-10-20T13:05:07Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4FW
manufacturer_id AIRB
mcu_id 003b00283536471334383630
signature 

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 10 1500 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 6 2 1300 1700 0 0
aux 4 46 9 1600 2100 0 0
aux 5 13 6 1750 2100 0 0
aux 6 19 3 1900 2100 0 0
aux 7 37 6 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = -22,81,3,1
set serialrx_provider = SBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_switch_mode = STAGE2
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 5
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_use_mag = OFF
set pid_process_denom = 2
set osd_vbat_pos = 2432
set osd_link_quality_pos = 2105
set osd_rssi_dbm_pos = 2136
set osd_tim_1_pos = 2081
set osd_flymode_pos = 2411
set osd_vtx_channel_pos = 2484
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2336
set osd_motor_diag_pos = 2241
set osd_gps_speed_pos = 2200
set osd_gps_lon_pos = 2496
set osd_gps_lat_pos = 2513
set osd_gps_sats_pos = 2508
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2220
set osd_flight_dist_pos = 2168
set osd_compass_bar_pos = 2090
set osd_altitude_pos = 2262
set osd_avg_cell_voltage_pos = 2400
set osd_core_temp_pos = 2295
set osd_efficiency_pos = 2464
set debug_mode = RTH
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5865
set gyro_1_align_yaw = 1800
set gyro_2_align_pitch = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

patemil, 2024-01-26:
up to 4.4.3 seem to have a working SBUS, none of the 4.5 releases work

haslinghuis, 2024-01-26:
Support ID can be found in CLI tab using configurator 10.10

patemil, 2024-01-26:
support id for my issue is Id: "5300f855-3066-42b6-88ce-a9c60d682641"

haslinghuis, 2024-01-26:
https://build.betaflight.com/api/support/5300f855-3066-42b6-88ce-a9c60d682641
https://build.betaflight.com/api/builds/5441e7213f8aee9c50ce2c4bc5144cee/log

haslinghuis, 2024-02-08:
Maybe use discord for further support - as we don't see any reports popping up.

AdrianEddy, 2024-03-21:
I had the same issue and I wanted to have both CRSF and SBUS support
It worked for me when I added `SBUS SERIALRX_SBUS` to the cloud build defines while keeping radio protocol at CRSF

closed 2024-03-21 by haslinghuis