betaflight/betaflight #13295: ID: "a05250a5-6cdf-4850-a4c2-2a33ff975cb2"

PhijFPV, 2024-01-14:
### Describe the bug

Gyro and ACC not detectet

### To Reproduce

Update to Betaflight 4.4 on BETAFPVF4SX1280 with GYRO=MPU6500, ACC=MPU6500
[AIO002.pdf](https://github.com/betaflight/betaflight/files/13929923/AIO002.pdf)

### Expected behavior

Resolve Bug

### Support ID

```txt
a05250a5-6cdf-4850-a4c2-2a33ff975cb2
```

### Flight controller

BETAFPVF4SX1280

haslinghuis, 2024-01-14:
This is DarwinFPV product. Not BetaFPV

Need to add `ACC_SPI_MPU6500 GYRO_SPI_MPU6500` to custom defines before flashing.

ot0tot, 2024-01-18:
You have the wrong target flashed. The correct target is `DARWINF4SX1280HD`. Save a diff, reflash the correct target, and restore your diff.