betaflight/betaflight #13298: 4.5 RC2 - ESC temps available but stuck at 0 on OSD

0crap, 2024-01-14:
### Describe the bug

Only tested with 4.5-RC2.
ESC temp values are visible on the Motors tab in the Configurator.
[image]

However the OSD is stuck at 0 and does not show actual ESC temps in flight.
[image]

### To Reproduce

Use bidir dshot and add the ESC temps element to the OSD.

### Expected behavior

Show ESC temps on OSD when also available on motors tab in the Configurator.

### Support ID

```txt
579b6b59-43f1-4b32-a897-7dfc8a5083ff
```

### Flight controller

F4 2-3S 20A AIO FC V1

### Other components

Walksnail digital VTX

### How are the different components wired up (including port information)

BETAFPV Pavo Pico Walksnail version
https://betafpv.com/products/pavo-pico-brushless-whoop?variant=40164423696518

0crap, 2024-01-14:
Same issue with a totally different 5" quad running bidir dshot. (Also on 4.5 RC2)
So it seems not related to hardware.

haslinghuis, 2024-01-14:
Please test:

[betaflight_4.5.0_STM32F405_BETAFPVF405.zip](https://github.com/betaflight/betaflight/files/13932881/betaflight_4.5.0_STM32F405_BETAFPVF405.zip)

0crap, 2024-01-14:
Tested, but unfortunately the OSD is still all zero.
The Configurator lost the value for Motor 4 with this test firmware.

[image]

0crap, 2024-01-14:
Motor errors on all motors remained at zero %.
Test hover was fine.

haslinghuis, 2024-01-14:
That's weird - the only change made was in osd_elements.c:

```
static void osdElementEscTemperature(osdElementParms_t *element)
{
#if defined(USE_ESC_SENSOR)
    if (featureIsEnabled(FEATURE_ESC_SENSOR)) {
        tfp_sprintf(element->buff, "E%c%3d%c", SYM_TEMPERATURE, osdConvertTemperatureToSelectedUnit(osdEscDataCombined->temperature), osdGetTemperatureSymbolForSelectedUnit());
    } else
#endif
#if defined(USE_DSHOT_TELEMETRY)
    {
        uint8_t escTemperature = 0;

        for (uint8_t k = 0; k < getMotorCount(); k++) {
            if ((dshotTelemetryState.motorState[k].telemetryTypes & DSHOT_EXTENDED_TELEMETRY_MASK) != 0 && (dshotTelemetryState.motorState[k].telemetryTypes & (1 << DSHOT_TELEMETRY_TYPE_TEMPERATURE)) != 0) {
                    escTemperature = MAX(escTemperature, dshotTelemetryState.motorState[k].telemetryData[DSHOT_TELEMETRY_TYPE_TEMPERATURE]);
            }
        }

        tfp_sprintf(element->buff, "E%c%3d%c", SYM_TEMPERATURE, osdConvertTemperatureToSelectedUnit(escTemperature), osdGetTemperatureSymbolForSelectedUnit());
    }
#else
    {}
#endif
}
```

0crap, 2024-01-14:
I'm afraid that is all black magic to me. :-)

haslinghuis, 2024-01-14:
@damosvil any clue?

damosvil, 2024-01-15:
It could happen the following:

- dshot_edt is not enabled. So far the only way to do this is by console (set dshot_edt=on). In motors tab dshot command 13 is sent when the tab opens. There is also a configurator pr that allows to enable EDT at https://github.com/betaflight/betaflight-configurator/pull/3262, and also adds new debug modes.
- edt command 13 in bf 4.4 and 4.5 is sent once, but sometimes dshot frames are lost so edt can potentially not enable in one or more motors and sometimes there are no temperatures in some motors until next arm. There is a pr that fixes this at https://github.com/betaflight/betaflight/pull/12170

0crap, 2024-01-15:
@damosvil Is there a way to test PR #12170 ?
I tried like below with zulu but that didn't solve the issue.

[image]

0crap, 2024-01-15:
I just flashed 4.5_zulu and changed the setting `set dshot_edt=on`
Guess what, it now works!

When the lipo is first time connected, the ESC temps are still zero.
However on arming it all came alive!
When I disarm the last known values are frozen on the OSD. (When you blip the throttle to remove the disarmed stats screen.)
When arming again it also starts to update the ESC temp values again.
This is reproducible, disarmed no ESC temp updates, armed ESC temps are shown correctly.

haslinghuis, 2024-01-15:
It will use EDT V1 afaik. But confused about the version:

http://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware

0crap, 2024-01-15:
Bit confusing indeed.
I have Bluejay v0.19.2 running on the AIO ESC.
The Bluejay table at https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#bluejay says EDTv2.
Or should that table says EDTv1 and EDTv2?

haslinghuis, 2024-01-15:
Maybe 0.20 will be EDT v3 - just guessing

0crap, 2024-01-15:
Might be, but v0.19.2 is the latest stable as of today.

Should we change the label of this issue to something else?
I guess this is not a bug but expected behavior for BF v4.5.0?

haslinghuis, 2024-01-15:
Yes expected behavior. We need EDT for ESC temp to show up. Would be nice to disable the element otherwise as it is also used when ESC_SENSOR is enabled with valid data from ESC wire.

0crap, 2024-01-15:
Good point.
If EDT is disabled (which it is by default) the Configurator should not allow the ESC temp element to be enabled.
Something like that?

haslinghuis, 2024-01-15:
Also make EDT configurable from the motors tab configuration. However we would have to duplicate MSP request on OSD tab to have knowledge about the setting. So a tooltip should be easier to implement.

0crap, 2024-01-15:
Close to a stable release a tooltip sounds fair enough.
Keep it simple for now?

haslinghuis, 2024-01-15:
[image]

0crap, 2024-01-15:
Looks good to me!

closed 2024-01-18 by haslinghuis