betaflight/betaflight #13300: 4.4.0 PROBLEMA GPS

ilpino8, 2024-01-14:
### Describe the bug

Quando attacco la LiPo in OSD non vedo i dati del GPS

### To Reproduce

Appena finisce il fix sento solo che il beeper suona, ma in OSD non vedo nessun dato di volo riguardante il GPS. In più non fa nemmeno armare i motori

### Expected behavior

Normalmente appena attacco la LiPo dovrebbe apparire i dati del GPS e quando arriva ad 8 satelliti il beeper suona.

### Support ID

```txt
cdffb9ec-e1f0-40f5-855a-7a8f15bac69d
```

### Flight controller

FOXEERH743 (STM32H743)

### Other components

RX CRSF, VTX Walksnail, GPS Foxeer M10Q 250, beeper autoalimentanto

aowi7280, 2024-01-16:
Try updating to 4.5. Lots of improvements since 4.4.0

ilpino8, 2024-01-19:
HI! I tried updating to 4.5.0 RC2, but nothing. The GPS still doesn't detect my GPS. I leave the problem ID code below
ID: "ef256ee4-a486-4e5d-9452-de6e5d6d7461"

Pancronos, 2024-02-01:
Did you add in your build gps ?

haslinghuis, 2024-02-01:
`GPS: NOT CONNECTED, UART1 19200 (set to 57600), NOT CONFIGURED unknown`

Recommended to set GPS to 115200 baudrate.

See: https://www.foxeer.com/foxeer-m10q-250-gps-5883-compass-g-508

closed 2024-03-10 by github-actions[bot]