betaflight/betaflight #13301: canvas not ok for HD dji 03

11BEX11, 2024-01-15:
### Describe the bug

I cant change the canvas height an widht to default, it always goes back to some strange nummbers.. so i cant have the correct osd placement.. 4.4 it works fine 4.5zulu rc1 and rc2 not ok...i have dji 03 unit with speedybee f405v4 stack..

### To Reproduce

make it work like in version 4.4

### Expected behavior

when i change the canvas height to 20 it should stay 20 not some random number. it goes the same for width 53..

### Support ID

```txt
.
```

### Flight controller

speedybee f405v4 bought from the speedybee site

### Other components

radiomaster rp3 rx, radiomaster ranger tx, dji 03 unit, all latest firmware..

### How are the different components wired up (including port information)

rp3 module on tx2
o3 unit rx1
gps geprcM10 rx6
rx3 is for lidar but that is for inav,not connected
rx4 is bluetooth

XRTce, 2024-01-16:
I encountered the same. Had setup my OSD for my DJI Goggles 2. Went out to the field and the OSD was gone. Back at home i set it up again (positions and display type (ntsc, pal, hd) were reset). OSD was working fine. Replugged the LiPO and the OSD was gone again

11BEX11, 2024-01-16:
strange...by me when i try to click ntsc or pal it goes back to hd but in the cli the setting for canvas height and widht it cant go to default setting...so I have the big squared in the osd tab not the hd small ones..in the  dji goggles v2 there is nothing but the original dji goggles things on the display..

TelefraggerRick, 2024-01-17:
Also having same issue with o3 air unit / goggles integra. Canvas is sized incorrectly / placed incorrectly resulting in the OSD being half off the screen in the goggles and to the right. Happens in all versions of 4.5.0. Not an issue in 4.4.3. Unable to change from PAL / NTSC / AUTO or HD. Just reverts right back to HD right away but its not set correctly. Ive seen similar issue with older versions but usually pick auto then go back to HD and it was resolved.

SteveCEvans, 2024-01-18:
Ensure that `osd_displayport_device=MSP`.

sugaarK, 2024-01-18:
@11BEX11 can you provide a diff as well?

ctzsnooze, 2024-01-18:
Did you `set vcd_video_system = HD` and `set osd_displayport_device = MSP`?

11BEX11, 2024-01-19:
[BTFL_cli_SPEEDYBEEF405V4 dump all 19.01.2024.txt](https://github.com/betaflight/betaflight/files/13986203/BTFL_cli_SPEEDYBEEF405V4.dump.all.19.01.2024.txt)

if there is any diference for you :)

haslinghuis, 2024-01-22:
Please test with https://github.com/betaflight/betaflight/pull/13320

closed 2024-02-29 by github-actions[bot]