betaflight/betaflight #13307: SoftSerial Range, Serial VTX

T3rorx, 2024-01-17:
### Describe the bug

On FuryF4OSD F405

When i put soft serial CMD

# resources
resource SERIAL_TX 12 C09
resource ESCSERIAL 1 NONE

# feature
feature -AIRMODE
feature SOFTSERIAL
feature GPS

Return me 
# resource SERIAL_TX 12 C09
###ERROR IN resource: INDEX NOT BETWEEN 1 AND 11###

### To Reproduce

Put In CLI :
resource SERIAL_TX 12 C09

### Expected behavior

accept command and can see softserial in ports tab

### Support ID

```txt
ID : "15c9f377-4b6f-4df2-b242-1625e5aeb210"
```

### Flight controller

Mamba f405 mk2.5 mini

haslinghuis, 2024-01-17:
Use resource SOFTSERIAL instead

T3rorx, 2024-01-17:
> Use resource SOFTSERIAL instead
resource SOFTSERIAL_TX 1 C09
Is good ! Thanks

closed 2024-02-25 by github-actions[bot]