betaflight/betaflight #13317: Barometer not detected in MAMBAH743_2022B FC target

popaflorin, 2024-01-21:
### Describe the bug

Barometer is not detected at all using DIATONE MAMBA MK4 H743 V2 FC and BF 4.4.3. In the specs of this FC is stated that is using a SPL06 barometer. 
I am not sure if this is a hardware or software bug but if anyone using the same target please confirm that if is working?

### To Reproduce

I have tried the following versions of BF and baro is not detected in any of them:
4.4.3
4.3.2
4.2.11

### Expected behavior

Baro icon will show in Betaflight

### Support ID

```txt
###

# # Problem description

# # Barometer not detected on MAMBAH743_2022B FC

# ###

# version
# Betaflight / STM32H743 (SH74) 4.4.3 Nov 18 2023 / 09:46:00 (738127e7e) MSP API: 1.45

# config: YES
# board: manufacturer_id: DIAT, board_name: MAMBAH743_2022B

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.32V, Core temp=34degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3712, max available: 131072
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MSP (30 x 16)
BUILD KEY: 42a3a02706dd88b8470c886c5ee5a2b5 (4.4.3)
System Uptime: 54 seconds, Current Time: 2024-01-21T11:02:09.652+00:00
CPU:14%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 28 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 5
FLASH: JEDEC ID=0x00efaa21 128M
Arming disable flags: RXLOSS CLI MSP

# flash_info
Flash sectors=1024, sectorSize=131072, pagesPerSector=64, pageSize=2048, totalSize=134217728 JEDEC ID=0x00efaa21
Partitions:
  0: BBMGMT    1003 1023
  1: FLASHFS   0 1002
FlashFS size=131465216, usedSize=71680

# dump master

# version
# Betaflight / STM32H743 (SH74) 4.4.3 Nov 18 2023 / 09:46:00 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MAMBAH743_2022B
manufacturer_id DIAT

# name: SpeedyBee
[... 1096 lines cut ...]
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

#
```

### Flight controller

DIATONE MAMBA MK4 H743 V2

### Other components

DJI O3                                        --> UART3
Radiomaster RP1 V2 ELRS       --> UART1
iFlight Blitz M10 GPS Module  --> UART2

haslinghuis, 2024-01-21:
Does the baro light up when the battery is connected (needs to be restarted)?

popaflorin, 2024-01-21:
> Does the baro light up when the battery is connected (needs to be restarted)?

Hmmm...yes it does show up when I connect the battery, but only after I restart it. Is this normal? Does that mean that it works when I power it up (not connected to BF) ?

haslinghuis, 2024-01-22:
You need to fly it. See: https://github.com/betaflight/betaflight/issues/13180#issuecomment-1828323324

Add `BARO` debug in blackbox for logging.

More info: http://betaflight.com/docs/wiki/guides/current/Barometer

haslinghuis, 2024-01-23:
Closing issue as the baro is detected now. We see this behavior on other builds where we need battery power for I2C devices with both internal and external sensors used. Must have something to do with [termination] resistance.

closed 2024-01-23 by haslinghuis