betaflight/betaflight #13323: Trouble with Bidirectional Shot

Hasselfpv, 2024-01-23:
### Describe the bug

i got some issues with rpm filtering (Bidirectional D-shot)n in betaflight 4.5 .
Fc is an Iflight Blitz mini atf435 and BLITZ E55S 4in1 ESC with Bluejay  0.19.2 at pwm 24 khz.
when enable Bidirectional dshot with shot 600 , i can spin all motors in Motor tap but got the erro Numbers on motor 2,3,4, when switch to shot 300 only errors are in motor 2 and 4 
[image]
[image]

### To Reproduce

Set bi directional shot on

### Expected behavior

I will fix the error

### Support ID

```txt
Id: "b22ce32c-047e-4334-8940-c2b7206ba4a4
```

### Flight controller

Iflight Blitz Mini ATF435

### Other components

Iflight Blitz mini E55s , Tbs Crossfire nano , Dji air unit o3

haslinghuis, 2024-01-23:
Try `set dshot_bitbang = ON`

Hasselfpv, 2024-01-23:
> Try `set dshot_bitbang = ON`

Tanks , now it still works on bench 👍

ot0tot, 2024-01-23:
> Try `set dshot_bitbang = ON`

Strange that timer based DSHOT is not working properly while bitbanged does, and that different behavior is observed with DSHOT600 and DSHOT300 though.

haslinghuis, 2024-01-23:
@ot0tot two motors are on timer 3 and two on timer 8. Should we enable bitbang in the config or do you want to change timers?

Hasselfpv, 2024-01-25:
> @ot0tot two motors are on timer 3 and two on timer 8. Should we enable bitbang in the config or do you want to change timers?

Thanks , with bitbang on it still works now.

Hasselfpv, 2024-01-25:
> @ot0tot two motors are on timer 3 and two on timer 8. Should we enable bitbang in the config or do you want to change timers?

how ist it possible to change the timers?

closed 2024-02-25 by haslinghuis