betaflight/betaflight #13327: GPS does not work on softserial RX-TX pair

aidaho, 2024-01-25:
### Describe the bug

I have a HappyModel X12 PNP v1.1 FC and I am trying to get GPS unit working on softserial RX-TX pair using RX2 and LED_strip pads.

### To Reproduce

I have mapped resources the following way:

```
resource PPM 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A00
resource SERIAL_RX 2 NONE
resource SERIAL_RX 11 A03
```
Where GPS TX is soldered to RX2 and GPS RX is soldered to LED_strip.

Original defaults for this FC were:
```
# resource
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C14
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 B09
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
```

### Expected behavior

I expect after setting up the ports and GPS settings to get a GPS icon light up in configurator as well as sat counter to show up on OSD.

### Support ID

```txt
2024-01-25 @14:29:06 -- Build server failure: HTTP 400
```

### Flight controller

HappyModel X12 PNP v1.1

### Other components

EP2 TXCO
Beitian BE-122 GPS

### How are the different components wired up (including port information)

EP2 TXCO is wired to TX1-RX1
GPS TX is soldered to RX2 and GPS RX is soldered to LED_strip

### Add any other context about the problem that you think might be relevant here

GPS unit is getting 3D fix as indicated by an LED.

I've tried playing with different port speeds and GPS settings to no avail.
[image]

[image]

For some reason getting Support ID dump failed with:
2024-01-25 @14:29:06 -- Build server failure: HTTP 400

I've tried several times.
Here is the dump:

[support_data.txt](https://github.com/betaflight/betaflight/files/14051771/support_data.txt)

aidaho, 2024-01-25:
Another datapoint: `gpspassthrough` shows data streaming in.  

[image]

haslinghuis, 2024-01-25:
Please check http://betaflight.com/docs/wiki/guides/current/SoftSerial#softserial-with-gps

aidaho, 2024-01-25:
So I need to set up GPS unit for lower baud and refresh rate first.  
I've ordered USB-UART converter and will update the ticket once I have a chance to perform the changes necessary.

aidaho, 2024-01-28:
I've received USB-UART converter, soldered it's rx-tx on top of FC connections, and successfully changed GPS settings to 9600, 2Hz, UBX NAV-PVT only.
I've verified softserial is at fixed 9600 speed and turned off GPS auto-config in BF configurator.

Unfortunately, nothing changed. GPS icon in configurator does not light up and satellite list is empty even though the unit has a 3d fix.

I still see gibberish in configurator console using `gpspassthrough`, but do not see any data when trying to connect PyGPSClient using passthrough.
At the same time, I'm able to connect with PyGPSClient through USB-UART to confirm the unit is working as it should.

Any ideas?

aidaho, 2024-01-28:
I've also updated to FC to Betaflight 4.4.3, this however did not influence the outcome.

aidaho, 2024-01-30:
Additional things I've tried: 

• Do a 2hz 19200 setup.
• Desolder USB-UART in case it causes any interference.
• Do an RX only setup on LED pad, insead of splitting UART.

```
# resource
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 A00
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource FLASH_CS 1 A14
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 30 2 9600 9600 0 115200
```

I tried the last option with both 19200 and 9600 fixed speeds and in both cases it did not work.
Furthermore, upon getting `gpspassthrough` command, FC have not echoed it back and immediately locked up. No gps data stream, no OSD refreshes.

surgeobrb, 2024-02-23:
Any updates on this? I am having the same issue down to the fc freezing up.

aidaho, 2024-03-25:
The issue is still there. While the GPS unit is no longer a part of the build, I've left a bunch of wires attached to the FC in order to test any proposed fixes.

haslinghuis, 2024-04-25:
Try `serial 30 2 115200 19200 0 19200`

aidaho, 2024-04-29:
> Try `serial 30 2 115200 19200 0 19200`

I've soldered GPS back and tried suggested configuration.  
Unfortunately, the symptoms stay the same.

```
# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 30 2 115200 19200 0 19200
```

haslinghuis, 2024-04-29:
Try the unsupported define while flashing from https://github.com/betaflight/betaflight/pull/13540

aidaho, 2024-04-29:
> Try the unsupported define while flashing from #13540

Thank you for the suggestion. I've flashed https://build.betaflight.com/api/builds/996b9517770dddabbf3bec373bcc664c/log

It doesn't seem to behave any differently.  
One thing that bothers me, is the same CPU usage with and without SoftSerial enabled in BF configurator. 50% vs 52% idle usage. I expected to see a larger jump.

I'll try to go over the whole setup again tomorrow to see if I'm missing something.

```
# status
MCU F411 Clock=96MHz (PLLP-HSE), Vref=3.31V, Core temp=48degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3479, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
GYRO=ICM20689, ACC=ICM20689
OSD: MAX7456 (30 x 16)
BUILD KEY: 996b9517770dddabbf3bec373bcc664c (4.6.0-dev)
System Uptime: 278 seconds, Current Time: 2024-04-29T19:32:30.278+00:00
CPU:53%, cycle time: 144, GYRO rate: 6944, RX rate: 15, System rate: 10
Voltage: 380 * 0.01V (1S battery - OK)
I2C Errors: 0
GPS: NOT CONNECTED, UART31 9600 (set to 19200), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP
```
[BTFL_cli_GHOST_20240429_223339_CRAZYBEEF4DX.txt](https://github.com/betaflight/betaflight/files/15153997/BTFL_cli_GHOST_20240429_223339_CRAZYBEEF4DX.txt)

aidaho, 2024-04-30:
I've checked my wiring and SoftSerial configuration.  
Soldered GPS to USB-UART. It was set up for 9600 speed and 2Hz NAV-PVT, so I've soldered it back to FC and configured 9600 for SoftSerial port. GPS auto-config is off.

GPS still is not getting detected. Yet as before I see an unreadable data stream using `gpspassthrough`.
I think SoftSerial is somehow broken on this hardware.

```
MCU F411 Clock=96MHz (PLLP-HSE), Vref=3.31V, Core temp=58degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3479, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
GYRO=ICM20689, ACC=ICM20689
OSD: MAX7456 (30 x 16)
BUILD KEY: 996b9517770dddabbf3bec373bcc664c (4.6.0-dev)
System Uptime: 29 seconds, Current Time: 2024-04-30T07:48:41.984+00:00
CPU:52%, cycle time: 146, GYRO rate: 6849, RX rate: 15, System rate: 9
Voltage: 412 * 0.01V (1S battery - OK)
I2C Errors: 0
GPS: NOT CONNECTED, UART31 38400 (set to 9600), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP
```
[BTFL_cli_GHOST_20240430_104832_CRAZYBEEF4DX.txt](https://github.com/betaflight/betaflight/files/15160080/BTFL_cli_GHOST_20240430_104832_CRAZYBEEF4DX.txt)

closed 2024-06-07 by github-actions[bot]

NitemareReal, 2024-11-08:
Exactly same problem for me using a JHEMCU F4 Noxe (F411) (only one UART free: UART 2, used for VTX). I did (more o less, since I didn't modified GPS baud rate) same things you did, before reading this thread, and I got same results. Now I'm stucked (I'm using last betaflight configurator available)
GPS works: I desoldered VTX on UART2 to test GPS on UART2 and it works.

aidaho, 2024-11-08:
I have not found any solution either, just got tired playing bump game with github-actions bot.

NitemareReal, 2024-11-09:
I found a workaround:
My FC (https://www.jhemcu.com/e_productshow/?57-F411-NOXE-V3-Flight-Controller-5V-10V-BEC-OSD-57.html) has only 2 UARTs, UART 1 and UART 2. In my current configuration, I must use UART1 for iBUS receiver (RX1) so, UART 2 is free. I connect my **analog** VTX on UART 2 using TX2 for smartAudio (only one pin needed)... so, what I did?
I connected my GPS on UART2 (RX2 and TX2 used) and I remapped LED_STRIP (RSSI is valid too, for example) to SOFTSERIAL_TX1 and I connected VTX SmartAudio to that pin.
This configuration works. I'm investigating how VTX uses UART TX pin because my VTX worked perfectly (even SmartAudio) without soldering that pin, but now it is soldered.
There is even other possible setup and is remaping UART_TX 1 (in my configuration it is free, it is not used by iBUS) to SOFTSERIAL_TX1, I haven't tested it, but it should work too. 

Summary: GPS on real UART: UART 2, VTX SmartAudio on a remapped SOFTSERIAL pin (only one pin needed).

aidaho, 2024-11-09:
Yes, I've previously did the exact same setup on the other FC and it worked fine.
This however was not an option for X12 FC, since VTX is integrated into AIO and permanently consumes UART.

NitemareReal, 2024-11-09:
> Yes, I've previously did the exact same setup on the other FC and it worked fine. This however was not an option for X12 FC, since VTX is integrated into AIO and permanently consumes UART.

😞