betaflight/betaflight #13332: LED_STRIP not working or very inconsistent

bisak, 2024-01-28:
### Describe the bug

LEDs are not working in most configurations. Seem to work as expected when all of them are configured to do the same thing. Most consistently what works is just the "color" configuration. Please see reproduction steps.

### To Reproduce

Connect USB and Battery to FC
Open CLI tab, run `defaults`
Open CLI tab, run `diff all`, verify `feature LED_STRIP` is enabled.
Open LED Strip tab
Assign 4 LEDs
Select function `Battery` for all 4
Save
Disconnect USB, Disconnect Battery,
Re-connect - verify LEDs have the ability to work - `Works as expected`

Open LED Strip tab
Change LED 1 to function `GPS`
Change LED 4 to function `Color`, toggle `Blink Always` ON
Save - `Works as expected`
Disconnect USB, Disconnect Battery
Re-connect - `Does NOT Works as expected. LED 4 is blinking, but 1 is NOT showing GPS status and 2,3 are NOT showing Battery`

There are other ways to make this reproduce, but this seems like the most consistent one.

### Expected behavior

Expected that configuration persists between FC reboots and generally to work as configured.

### Support ID

```txt
72646f99-6257-4e5b-96e8-dd7449208d05
```

### Flight controller

IFLIGHT_BLITZ_F722 V1.1 / STM32F7X2

ASDosjani, 2024-01-28:
@SteveCEvans I dropped your PR and it works again, so the problem is there.
https://github.com/betaflight/betaflight/pull/13247

haslinghuis, 2024-02-06:
@ASDosjani are you working on a fix - for instance does it help to for less powerful targets to use

```
#define LED_STRIP_MAX_LENGTH           16
```

SteveCEvans, 2024-02-06:
Have you tried other functions? Does rainbow work? Is this GPS specific for example? What about the modifiers such as throttle position?

SteveCEvans, 2024-02-06:
@ASDosjani have you been able to do a git bisect to find the exact commit which first shows the issue?

ctzsnooze, 2024-02-06:
Need more information about exactly which aspects of the LED functions work, and which does not, whether reducing LED_STRIP_MAX_LENGTH solves the issue, checking TASKS for late task issues, trying out a lower PID loop rate, etc.

Once a specific configuration is reliably identified that does not work, then must go back to 4.4, 4.3, etc, and find out if it ever worked in the past or not.  If it worked in the past, then need to move forward until we identify when it started to fail.

ASDosjani, 2024-02-12:
@SteveCEvans I had a bit of time to test today. I used simple static colod on LED 0 and GPS on LED 1. Target is BETAFPVF4SX1280. What I found:
- #13218 started to make weird stuff. LED 1 should have short blink priodically because no GPS is attached to the FC but instead it sometimes seemed to stuck switched on and insted of turing on the LED it started to do it's reverse, it turned off the led for a short time and it's kind of random when it does this.
- #13247 completely broke this, no lights at all on LED 1.

I tested on 4khz PID loop because on F411 8khz is unstable although it defaults to this...

TASK_LEDSTRIP_RATE_HZ is only used in Rainbow effect, I introduced it wrongly because it's the only effect which frequency is changable. Sorry about that, I missinerpreted how effects work, it shouls have a static 60Hz like Larson Scanner has, I will make a quick PR with the fix.

nerdCopter, 2024-02-16:
is this solved by the merged https://github.com/betaflight/betaflight/pull/13363 or not completely?

ASDosjani, 2024-02-16:
> is this solved by the merged #13363 or not completely?

No, it's not related.

SteveCEvans, 2024-02-16:
@ASDosjani  Have you raised the promised PR?

ASDosjani, 2024-02-16:
Yeah, #13363 . But it's not important. How we solve the freezing effect bugs I tested?

SteveCEvans, 2024-02-16:
Please provide the configs which cause breakage. @ctzsnooze  and I did extensive testing including rainbow mode so I’m curious to see what coding has issues.

ASDosjani, 2024-02-16:
@SteveCEvans check my comment above, for example GPS led strip function or simple Blink effect not working.

SteveCEvans, 2024-02-17:
Are provide LED config fragments for easy reproduction

ASDosjani, 2024-02-19:
@SteveCEvans 4khz PID and
```
led 0 12,10::CB:3
led 1 13,10::G:3
```

ctzsnooze, 2024-03-01:
@SteveCEvans I didn't test 'mixing modes', ie setting one individual LED to GPS, and some other individual LED to Battery.
It is likely that requiring the code to handle multiple modes at once is the cause of the problem.

sugaarK, 2024-03-03:
@MindRiderFPV @bisak @ASDosjani can you guys test https://github.com/betaflight/betaflight/pull/13412