betaflight/betaflight #13341: ESC Passthrough Issue?

afrolou, 2024-02-04:
### Describe the bug

4.5RC2 and 4.5zulu not allowing pass through for blheli configurator. 

### To Reproduce

Multiple flight controllers (Tmotor F7 Pro, BrainFPV Radix2, Diatone Mamba F722APP) seem to keep blheli suite from addressing a 4-in-1 ESC for configuration (tested on Foxeer Reaper 128k 65A and Diatone Mamba 50a escs). Downgrading to 4.4.3 fixes issue. 

### Expected behavior

Esc passthrough configuration should work as expected. 

### Support ID

```txt
N/A
```

### Flight controller

Tmotor F7 Pro, BrainFPV Radix2, Diatone Mamba F722APP

fredakilla, 2024-02-04:
Same issue here.
Unable use BLHeliSuite32 after flash BF 4.5RC2 (Downgrading to 4.4.3 and it's ok)
Tested on 2 differents stack.

haslinghuis, 2024-02-04:
Does it work in RC1 or zulu and what operating system is used?

afrolou, 2024-02-04:
> Does it work in RC1 or zulu and what operating system is used?

Haven't tested RC1, but issue persists on RC2 and zulu.

weeli87, 2024-02-04:
Latest Zulu with latest Nightly Configurator (as of today) ESC-Passthrough works fine with BlHeli32 10.10 on Windows 10 with the following boards:

- HGLRCF722E
- MAMBAF722_I2C

Can't confirm this issue.

haslinghuis, 2024-02-04:
Getting to think this issue seems to be related to OSX ?

Also: https://github.com/stylesuxx/esc-configurator/issues/439

fredakilla, 2024-02-05:
I have issue on windows 11 and latest and previous version of blheli.
tested on stack t-motor and stack hobbywing xrotor.
It works perfectly when downgrading to bf 4.4.x, but each time I flash to bf 4.5rc2, blheli doesn't work.

fredakilla, 2024-02-05:
today tested on 4.5RC1 and its works with this version
no diff in cli dump between 4.5RC1 and 4.5RC2

haslinghuis, 2024-02-05:
If you drop a support ID for 4.5 I can create a hex for testing

fredakilla, 2024-02-06:
*** Support data submitted *** 
Id: "b210f843-9e4b-4ed3-80c0-c4486ee0b15b"

SteveCEvans, 2024-02-06:
Might this be related to https://github.com/betaflight/betaflight/pull/13287 and https://github.com/betaflight/betaflight/pull/13277?

This was OK in all my testing except for F435 based FCs flashing via ESC-configurator under MacOS which was what led to this change to improve behaviour.

haslinghuis, 2024-02-06:
@fredakilla for testing the issue this hex has #13277 removed.

[betaflight_4.5.0_STM32F7X2_HOBBYWING_XROTORF7CONV.zip](https://github.com/betaflight/betaflight/files/14179647/betaflight_4.5.0_STM32F7X2_HOBBYWING_XROTORF7CONV.zip)

fredakilla, 2024-02-06:
yes it works with this hex

SteveCEvans, 2024-02-06:
In which case, should we back out the timeouts or increase them? Double?

haslinghuis, 2024-02-06:
Think removing for now would be best or make the changes only apply to AT32.

nAdibFPV, 2024-02-12:
Same here with the Speedy Bee F7 V2 stack - on 4.5 RC2, Blheli configurator does not recognize ESC can't upgrade esc firmware

SteveCEvans, 2024-02-12:
> Think removing for now would be best or make the changes only apply to AT32. At least we need more investigation on different MCU's on different platforms besides web issues on esc-configurator.
> 
> Have the timeout configurable (default to original behavior) might be another option to aid the investigation.

I’ll raise a PR to add a debug reporting the max time taken, but not to enforce the timeouts. This will effectively revert the behaviour but allow us to better understand what’s needed.

haslinghuis, 2024-02-12:
@afrolou @fredakilla please test #13364 and report back

closed 2024-02-13 by haslinghuis