betaflight/betaflight #13353: Support for Lidar rangefinder

potato1992, 2024-02-06:
### Is your feature request related to a problem? Please describe

I cannot judge the height over fresh snow(smooth surface) over the fpv camera. And the barometer height is way not reliable. Has crashed into snow twice.

### Describe the solution you'd like

Add support for lidar rangefinder, so that we can use it to deternine the distance to smooth surface.
The best option would be auto minimal height maitain(fc wont allow height to ground smaller than a set threshold)

### Describe alternatives you've considered

Display rangefinder distance over osd.

haslinghuis, 2024-02-06:
You can add `RANGEFINDER` in custom defines to enable support for LIDAR.

MisterHW, 2024-03-12:
Is LightWare SF10 / SF11 supported or is support planned?
https://betaflight.com/docs/wiki/guides/current/Supported-Sensors#introduction

> Range Finder
> 
>     HCSR04
>     Benewake LIDAR TF02
>     Benewake LIDAR TFmini

haslinghuis, 2024-03-12:
LightWare SF11 will be discontinued later this year.

[image]

Have used a LIDAR-Lite v3HP with Ardupilot - have not tried with BF.

ctzsnooze, 2024-04-06:
No ongoing support for lidar unless someone takes an interest and manages that code.

closed 2024-04-06 by ctzsnooze

KANENAS1, 2024-06-08:
what type of support do we need to get this working for BF. i'd like to assist or support

rsarwar87, 2024-07-08:
I can get a hardware-level support working for MTF-01. seems straight forward from the code based on this directory: 

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/rangefinder/rangefinder.h

Can I assume all the calc needed can be used as is?

makiisthenes, 2025-10-22:
Any updates on this?

haslinghuis, 2025-10-22:
FYI 

Lidar and optical flow driver support for MTF has been added. No additional flight features or OSD functions were added yet.