betaflight/betaflight #13360: Mag and Baro stop working during flight

trenth00, 2024-02-12:
### Describe the bug

When my Mag is configured (connected via i2c bus) it appears to work as expected on the bench. However when I connect a battery and start flying the Mag spontaneously stops working and seems to take the on-board Baro with it. In the Blackbox logs you can see clearly when both stop working. If I disable the Mag (but leave it physically connected), then the Baro works as expected

### To Reproduce

There is nothing special required to reproduce this issue as it happens, however once it occurs it will not rectify itself without removing power from the FC. I have replaced the FC, re-routed power away from the Mag, replaced the CAP,  replaced the GPS/MAG & relocated the receiver antenna

### Expected behavior

Baro and Mag should continue to operate as expected during flight

### Support ID

```txt
*** Support data submitted *** 
Id: "2f769fa1-0cb3-49ee-a1a1-d90a5ba23c03"
```

### Flight controller

Radix 2 HD

### Other components

O3 Air VTX
TBS Crossfire receiver
Hobbywing XRotor Micro 4-in-1 60A ESC 6S BLHeli_32 

### How are the different components wired up (including port information)

Baro is on-board (i2C)
Mag and GPS are part of "Mateksys GNSS M10Q-5883" (i2c)
GPS is configured on UART 6
Mag is connected via SDA/SCL pads
GPS/MAG is powered from same 5V pad as receiver (powered via usb)

### Add any other context about the problem that you think might be relevant here

Once this issue occurs the only way to resolve it is to recycle power from the FC. I have run iNav with this same setup and never noticed the issue. With Betafglight it happens every time I go out and sometimes even when it's on the bench (powered via USB). I initially thought it may have been interference on the i2C bus so I replaced all of the cables, rerouted all power to the other end of the quad, and relocated my Crossfire antenna . If I "disable" the mag in Betaflight but leave connected, the Baro works as expected

NeariX67, 2024-02-13:
This issue is somewhat related to your setup.
With my Speedybee F7 V3 and the GEPRC M10-DQ, i do not have this issue. Barometer (from FC or GPS doesn't matter) works in combination with mag.

trenth00, 2024-02-13:
Yep its not a hardware thing as it works well under iNav

sugaarK, 2024-02-19:
best to contact Radix we do not support their hardware in Betaflight they manage their own branch of Betaflight.  it sounds like a timer definition issue

Giammi, 2024-03-17:
> best to contact Radix we do not support their hardware in Betaflight they manage their own branch of Betaflight. it sounds like a timer definition issue

Unfortunately I can confirm this is happening also to me on two quads:
1) Kakute H7 (official branch), Foxeer M10 250 5883 + BMP280, both on RC2 and Dev release;
2) Radix2 (non HD), Foxeer M10 250 5883 + BMP280, only RC2 tested.

I haven’t logged the issue, but it is clear from the OSD. The baro stops working (vario stays at 0 and altitude is locked) while the graphic compass still moves, but giving wrong heading. Also the Home arrow and numerical heading are wrong, they’re consistent with the graphic compass.

https://github.com/betaflight/betaflight/assets/66137675/a2e0f1ed-832c-4994-bad1-6ca9b56c8422

trenth00, 2024-03-17:
@haslinghuis can I please ask you to reconsider this issue as I have just found someone else with exactly the same problem, however in his instance it was with a totally different FC. His scenario has been posted into this thread too

mluessi, 2024-03-18:
I suspect this problem is related to the QMC5883 "getting stuck", i.e., it keeps pulling the SDA line low which then causes subsequent I2C transactions to fail. Both ArduPilot and INAV have code to detect and "unstuck" the bus when this problem occurs, but betaflight currently doesn't. See e.g. [discussion of this problem with ArduPilot](https://discuss.ardupilot.org/t/weird-compasses-hangs-and-readings-on-chibios/35286/8). The heading data keeps updating when this occurs as it also uses gyro data for computing the heading.

haslinghuis, 2024-03-18:
Thank you @mluessi was already suspecting something like that.

@SteveCEvans do you have to time to look into this issue ?

ledvinap, 2024-03-18:
Can you try few tests: 

- (ideally): take i2c protocol trace during failure (Saleae etc.). Include analog channel data if available.

when board is stuck:

- measure voltage on SCL/SDA - ideally oscilloscope, but simple multimeter will do
- softreset FC (from cli/configurator)
- (if easy): disconnect/reconnect power to mag (without resetting FC)
- (if you have some experience with electronics): Short SCL line to ground few times

trenth00, 2024-03-18:
I'll give these tests a crack, but I am no expert when it comes to electronics.
So to confirm:
- to measure voltage on SCL & SDA pads, I just connect the positive of multi-meter to pad and negative to GPS ground pad?
- mg mag is connected via a GH connector so its easy to disconnect, however in doing so it will not only disconnect power but tx/rx/sda/scl too. is this ok?
- as the GPS/MAG unit has two GH connectors I can expose the second SCL/SDA connectors whilst the first is still connected to FC. Is it ok to take the voltage measurements from here?. Also can I use this second SCL to short it to the ground pad?

Giammi, 2024-03-19:
I have at home one unsued BN-880 (HMC5883) that I can use as a test bench. I'll put it together with a Radix2 and play with it. BTW, on the Foxeer M10, the Mag is **QMC5883** and not **HMC5883**.

NeariX67, 2024-03-19:
Even though i said it works fine on my setup, i now encounter this issue as well. Can't tell what changed that this did not occur before.

On second 7 of the flight, the baro gets stuck. Homepoint direction is way off, and heading rotates by 60° without significant yaw turn.

https://github.com/betaflight/betaflight/assets/31779313/a8da1cb0-97fc-41e0-bf01-1d71b594b1b5

trenth00, 2024-03-19:
@ledvinap I managed to get some measurements during a failure.
-SDA voltage = 3.27v
-SCL voltage = .03v
-Softreset (from configurator) made it work again.

It's actually quite hard to get it to fail on the bench, so I am trying again and will revert with additional measurements relating to the last two tests you requested. The ones above were taken from a multi-meter

ledvinap, 2024-03-19:
> my mag is connected via a GH connector so its easy to disconnect, however in doing so it will not only disconnect power but tx/rx/sda/scl too. is this ok?

Yes, that will help. But from your voltage measurement, it seems that it won't help.

> as the GPS/MAG unit has two GH connectors I can expose the second SCL/SDA connectors whilst the first is still connected to FC. Is it ok to take the voltage measurements from here?. Also can I use this second SCL to short it to the ground pad?

They seem to be internally connected. It is possible to use this side.

> I managed to get some measurements during a failure. 
-SDA voltage = 3.27v -SCL voltage = .03v 
-Softreset (from configurator) made it work again.

This is interesting. This will indicate FC side problem - it QMC5883L pulled SCL low, there should be no way to unstick it by FC softreset.

SDA low can be caused by SCL desync, unstick should work in this case.

> removed power from MAG (by unplugging the unit and re-plugging but also disconnecting/reconnecting RX, TX, SCA, SCL, GND, 5V in the process. Resulted in the problem remaining, however upon testing SDA & SCL voltages both were .03v

Can you measure voltages after unplugging M10Q-5883?

trenth00, 2024-03-19:
Can you measure voltages after unplugging M10Q-5883?
Now this was extremely tricky. as the connector is soldered onto the board ( and I cannot get the multi-meter probes into the board as its still mounted in my frame, I had to use the multi-meter probes on the GH connector end itself. 
Both SCL and SDA measured 3.27v when gps/mag was disconnected

NeariX67, 2024-03-19:
If desired, i can try to get this reproduced while attached to a digital analyzer. Will be home in about 7 hours.

trenth00, 2024-03-19:
> If desired, i can try to get this reproduced while attached to a digital analyzer. Will be home in about 7 hours.

That might help. My little multi-meter and I are probably at the extent of our ability :-)

ledvinap, 2024-03-19:
@trenth00 : Can you also check I2C errors (before/after)?

trenth00, 2024-03-19:
yep, when the mag and Baro stops working  I see a small handful of i2c errors (only about 6 or 7). When I disconnect the Mag/GPS I see 1000's of errors

NeariX67, 2024-03-19:
Got everything setup now with my logic analyzer. Has anyone found a reliable method to trigger this issue?

NeariX67, 2024-03-19:
[i2c_scl_short.zip](https://github.com/betaflight/betaflight/files/14655371/i2c_scl_short.zip)

I was unable to get this issue to trigger after 16 minutes, so i shorted SCL to GND like trenth00 wrote earlier.

I'll keep testing until eventually it will happen, hopefully.

trenth00, 2024-03-19:
I find that if you have a gps lock the problem shows itself quicker. Also be sure to generate alt and mag movement as would normally happen during flightTrent

NeariX67, 2024-03-19:
[i2c_loss.sal.zip](https://github.com/betaflight/betaflight/files/14656873/i2c_loss.sal.zip)

Got it. Looks different than the one caused by a SCL short.

Giammi, 2024-03-20:
> [i2c_scl_short.zip](https://github.com/betaflight/betaflight/files/14655371/i2c_scl_short.zip) (Logic 2 Captures)
> 
> I was unable to get this issue to trigger after 16 minutes, so i shorted SCL to GND like trenth00 wrote earlier.
> 
> I'll keep testing until eventually it will happen, hopefully.

@NeariX67 in my case I noticed that the issue is more keen to appear when heading is changed rapidly and for a prolonged period of time. If you watch my video above, you can see the baro getting stuck during a 360 orbit.

ledvinap, 2024-03-20:
@NeariX67 : Can you try once more, please?

QMC5883 stops responding at 265.726772s (interpreted as 0xff bytes). Detecting false NAK may cause this.
After this, data are partially read and QMC returns 0x04 as status (Data Skip). DRDY cant be set (data registers are locked until 0x05 is read).

at 265.791742s:
- QMC does not ACK read address byte. 
- IMO this triggers bug in ST HAL drivers. 25ms busy wait is inserted (Blocking gyro processing etc.).
- QMC does not ACK ADDR few times
- QMC ACKs address (265.824 594 833 s), HAL does not handle it well and ends up in clock stretching.

Masking/fixing QMC5883 problem is possible and quite easy. But IMO this only makes second problem more likely to occur, it will happen anyway.
And fixing ST manure is not something I'm going to do for fun ...

ledvinap, 2024-03-20:
https://community.st.com/t5/stm32cubeide-mcus/hal-i2c-mem-read-it-fails-if-a-nak-occurs-on-the-memory-address/td-p/591154

NeariX67, 2024-03-20:
[Archive.zip](https://github.com/betaflight/betaflight/files/14671768/Archive.zip)

Got to capture the bug again, 2 times.
Also captured when i removed the gps module after the issue occured and replugged it.

ledvinap, 2024-03-21:
Disconnect1: Probably SDA low causing Arbitration Loss on master. It may be caused by slave seeing extra pulse and holding SDA during ACK

Disconnect2: First data overrun. Then arbitration loss, possibly by slave missing one SCL pulse.

Can you try with 400Khz I2C clock? `i2c?_clockspeed_khz = 400`

ledvinap, 2024-03-21:
Can you test https://github.com/betaflight/betaflight/pull/13467 ?

It does should one kind of problem - magnetometer not responding, but baro working. 
But bus errors will still cause lockup, but with smaller probability (maybe 5x smaller).

NeariX67, 2024-03-21:
Yes, i can do that after work. Before i flash the PR, i will test with 400 kHz

trenth00, 2024-03-21:
@mluessi could I ask you to include this in a custom build so I can test on the Radix 2 HD please
@ledvinap I did test with an i2c Clockspeed of 400 Hz a few weeks ago. Martin did a custom build as part of our early troubleshooting. The issue persisted though

ledvinap, 2024-03-21:
Can you test varying I2C speed? Even increasing it, it may cause problem sooner.

@NeariX67 : Any chance you can measure I2C using oscilloscope (ideally both x1 and x10 probe to change capacitive loading)?

NeariX67, 2024-03-21:
No, i don't have an oscilloscope, the digital analyzer is all i got.

trenth00, 2024-03-21:
I was of the opinion that the option Martin added only let me reduce to 400KHz, not increase. @mluessi , that custom build you did for me to test a slower clock speed with the command "set i2c1_clockspeed_khz = 400". Does that allow us to set any speed?

ledvinap, 2024-03-21:
The limit is 100kHz to 1300kHz. And it is possible if CLI variable i2c1_clockspeed_khz  exists

trenth00, 2024-03-21:
ok, so I managed to find a really easy way to recreate the problem on demand, and in doing so tried 100Khz, 400Khz, 800KHz, and 1300KHz. regardless of the speed I set the problem re-occured at will.

Steps to recreate issue on demand 
1. Turn on radio (RadioMaster TX16S) Crossfire @ 915 MHz
2. Via Lua Script set telemetry transmit power of TBS Nano RX Pro to Max
3. rub my finger on the quad's receiver antenna (Immortal T antenna)

Within about 2 seconds of rubbing my finger on the antenna the i2c bus locks up with the BF configurator often showing I2C errors.

ledvinap, 2024-03-21:
@NeariX67: This is ST HAL bug, locked data / data overflow did not occur this time. 
The problem is probably caused by noise,  maybe using smaller I2C pullup resistors may help (but it will only mask BF bug). 
What hardware do you use?
Adding 2-5kOhm to 3.3V (both SCL and SDA) may help

NeariX67, 2024-03-21:
On this test setup, i have a Skystars F7 Pro4 with a GEPRC M10-DQ. Nothing else.
On my 5 inch i have a NeutronRC F745 with a Speedbee F7 50A ESC, GEPRC M10-DQ as well.

Did occur on both systems.

Will test again...

ledvinap, 2024-03-21:
@NeariX67 : Looks like signal integrity issue, but hard to tell without oscilloscope.

NeariX67, 2024-03-21:
[4.7k_pullups.sal.zip](https://github.com/betaflight/betaflight/files/14703734/4.7k_pullups.sal.zip)

Now with 4.7k Pull-ups

ledvinap, 2024-03-21:
@betaflight/firmware : What about buggy I2C drivers? ST will eventually fix it, but it can take years .. 

One option is to replace `HAL_I2C_Mem_Read_IT` with HAL_I2C_Master_Seq_Transmit_IT + HAL_I2C_Master_Seq_Receive_IT. It will remove blocking code and will probably fix NAK problem.

But unstick routine is still necessary. Implementation will be a bit complicated in nonblocking mode.

ledvinap, 2024-03-21:
@NeariX67 : It is failing more sooner with 4k7 resistors?

NeariX67, 2024-03-21:
Didn't really notice a difference. As it was pretty random when the issue occurs, i did not see a correlation with or without resistors.
I will keep them attached.

If interested, i can provide a Ubuntu VM attached to the FC and Logic analyzer.

ledvinap, 2024-03-21:
Always NAK in in read address byte. At least it is software bug ..

trenth00, 2024-03-24:
I just flashed 4.5 RC3 + #13467 (thanks @mluessi ), went for a fly and didn't get a single error. After 4 packs and 20 mins of flight the Baro and Mag behaved like best friends. Many thanks @ledvinap and everyone else involved, you have ended months of frustration. I appreciate that this issue may not be 100% eliminated according to your analysis but it is a heck of a lot better.

Giammi, 2024-03-24:
@ledvinap I tested the commit #13467 on both my rigs. Both are equipped with Kakute H7 V1.3 and Foxeer M10 with QMC5883.

Rig 1 the issue of barometer + magnetometer did not appear even stressing the flight a little bit. Whereas in the past it was happening quite quickly. I consider this a success from this point of view. On the other hand unfortunately, I don’t know why, I have now weird vibrations in the quad and I have tons of jello in the FPV feed. 4.5RC3 was flying perfectly. Settings are the same. Just flashed 4.5 dev with selected commit and transferred the diff.

Rig 2 the issue is now even worse. On the bench everything is ok, but as soon as I plug the battery, weird things start to happen. Sometimes the magnetometer is like disabled (and I can confirm it is disabled by connecting the quad to BetaFlight, the MAG icon is grey). Sometimes it works, but just for few moments. For instance I can see the compass in the OSD working fine and then, all of a sudden, it starts to give wrong heading. It starts to do that even before I arm the quad.

trenth00, 2024-03-24:
I Know I’m being a bit cheeky here but giving a plug for feature request #13448 . One way to fix the problem is for MSP support for the external mag’s (no i2c issues then). :-)

ledvinap, 2024-03-25:
@Giammi: 

Rig1: Can you try disabling magnetometer? It is possible (but unlikely) that PID timing may be affected by I2C timeouts.

Rig2: How does it behave with 4.5RC3 ?

Giammi, 2024-03-25:
> @Giammi:
> 
> Rig1: Can you try disabling magnetometer? It is possible (but unlikely) that PID timing may be affected by I2C timeouts.
> 
> Rig2: How does it behave with 4.5RC3 ?

@ledvinap Ok, I will try Rig1 without MAG and let you know. About Rig2, with RC2 and RC3 I was at least able to take off with working magnetometer and barometer (video in my comment above is on RC2). They were stopping working mid flight, but never happened the MAG and Baro got stuck before taking off. It may be a coincidence, but it's hard to tell. From the flight performance point of view, no issue to highlight.

One thing is sure, on the bench, powered up from USB, both rigs work great. From the sensor tab I can get consistent X, Y, Z reading with magnetic fields.

Oh, another thing to mention about Rig2 is that the battery is bottom mounted. This means that when I plug the battery, the quad is upside-down. I don't know if this can cause any trouble to FC initialization or something and can lead to bad MAG initialization. Apart from that, the two quads share the same FC/ESC/GPS/MAG configuration.

ledvinap, 2024-03-25:
@Giammi : Can you post photo of the rig 2 (wiring is important)? It may be caused by ground loops etc ..

Giammi, 2024-03-25:
@ledvinap as of now I only have pictures with the canopy on. What are you looking for in particular? So I can take more detailed pictures.

Nevertheless, the GPS unit is placed on a dedicated TPU mount that sits on top of the canopy, right on top of the FC to keep the wires as short as possible. Wires come down from the GPS unit and they're soldered straight on the FC. Moreover, the FC GPS pads are on the upper side of the FC, so the GPS wires are routed directly without passing close to any other wires, especially motor or lipo wires that can cause noise. On top of that, the build has a spike absorber soldere onto the ESC pads along with a 1000uf capacitor, and even the FC has a 470uf capacitor soldered to a VBAT pad to clean the low frequencies noise as much as possible. It's a 10" and it's flying great, so I assume it is clean from the electrical point of view.

[image]

trenth00, 2024-03-29:
Hi guys, I had another 100% success with this fix today (over 5 packs). Didn't miss a beat!.
What's the process with these bug's. Do I leave it open, let it auto close, or do I close it?

haslinghuis, 2024-03-29:
Fixed in #13467

Giammi, 2024-03-29:
@ledvinap here is the picture you asked for. 5V and GND pads of the GPS are next to each other. Other peripherals connected to the FC are just the RX, the buzzer and the O3 Air Unit (just RX and TX, VTX is powered directly from lipo pads). I don't know what a ground loop is, but to me looks like a clean build.

The dirt you see on the pads is just conformal coating, nothing more. All electronic is fresh.

Any suggestions?

Thank you!

[image]

nerdCopter, 2024-03-29:
@Giammi ,
Github for bug-reports.
**Discord for support and misc communications.**

Giammi, 2024-03-29:
@nerdCopter I am actually reporting a bug. I tested the commit, but worked only on one of my build. The other one still behaves as reported by @trenth00 in the initial post. On the bench, all good. In the field, weird things happen. I don't think I need support as I was able to fly this quad with the magnetometer fully working. The guide about mag is great. If you think I should continue on Discord, fair enough, I will move there.

ledvinap, 2024-03-29:
@haslinghuis https://github.com/betaflight/betaflight/pull/13467 fixes only part of this problem.

trenth00, 2024-04-02:
Not sure if this is related at all but I went for a fly on the weekend and on 1 flight I had a number of occasions where the system flashed up “Rescue N/A” for a few seconds. On review of Dvr footage I can see that the gps went from 17-18 sats to 0 during this warning and then instantly back to 17-18. Looks like baro and mag were unaffected. Do we think this is related?… or maybe something else entirely?

ctzsnooze, 2024-04-03:
If GPS is over serial, losing sats abruptly is likely a different problem.  It sometimes happen if the auto-config fails because the module doesn't permit it, or if the baud rate/data rate are inappropriate, or if the quad is running 8k8k.

ctzsnooze, 2024-04-03:
I've got a PR to update the Mag documentation, and one to revise and improve the Mag calibration procedures.  These don't relate to the hardware issues here, but should help resolve calibration and orientation issues a bit more easily.

trenth00, 2024-04-03:
> If GPS is over serial, losing sats abruptly is likely a different problem. It sometimes happen if the auto-config fails because the module doesn't permit it, or if the baud rate/data rate are inappropriate, or if the quad is running 8k8k.

Ok, thanks . I’ll drop the speed back to 57600 and keep an eye on it.

[44 more replies]

closed 2026-09-26 by blckmn